cocotb/cocotb #1704: caonstrained random and coverage

Divyanirankari, 2020-04-24:
```
hi, 
  i have to randomise  some bits of 32 bit instruction and some bits will be constant like below format 
  
+---------+-------+-------+-----+-------+---------+
| 0100000 | XXXXX | XXXXX | 111 | XXXXX | 0110011 |
+---------+-------+-------+-----+-------+---------+

through direct random function we can randomise cant find coverage using cocotb-coverage library.
    for i in range(n):
            instruction = random.randint(0,0xFFFFFFFF)
            instruction = instruction & 0x01FF8F80
            
            instruction = instruction | 0x40007033
            
            
            
how we can find constrained random and coverage using cocotb-coverage?

```

themperek, 2020-04-24:
@Divyanirankari  You have to ask here: https://github.com/mciepluc/cocotb-coverage

closed 2020-04-24 by themperek

Divyanirankari, 2020-04-24:
thanks @themperek