cocotb/cocotb #1707: Simulator makefiles do not support VHDL-on-top

cmarqu, 2020-04-25:
Our simulator makefiles get the list of Verilog and VHDL sources in separate variables, and in ``Makefile.questa``, ``Makefile.aldec``, ``Makefile.activehdl``, there are separate compiler calls to compile Verilog and VHDL.
Right now in those makefiles, VHDL is compiled first, followed by Verilog. This means that a VHDL entity is in the library when the Verilog is compiled, but not the other way round.
Thus, designs where a VHDL toplevel instantiates Verilog modules cannot be compiled with these makefiles.
(The problem is also present with more complicated sandwiched instantiations.)

See also #1686.

themperek, 2020-04-25:
Depending on how you look at this/. It is a test definition issue, not a simulator. One has to specify just one top,

cmarqu, 2020-04-25:
I don't see how. I do only have one top per setting of ``TOPLEVEL_LANG``.
I'd be happy to be wrong of course :)

themperek, 2020-04-26:
I may be wrong but something like this:
```make
ifeq ($(TOPLEVEL_LANG),verilog)
  VERILOG_SOURCES += top.v
else ifeq ($(TOPLEVEL_LANG),vhdl)
  VHDL_SOURCES += top.vhdl
endif
```

garmin-mjames, 2020-10-21:
Active-HDL allows compiling Verilog/SystemVerilog modules in any order.

You do get a compilation warning:
`Warning: VCP2515 /cocotb/examples/mixed_language/hdl/toplevel.sv : (100, 3): Undefined module: endian_swapper_vhdl was used. Port connection rules will not be checked at such instantiations.`

So potentially the VHDL file compilation could be moved after Verilog. I'm not sure about Riviera-PRO.

cmarqu, 2020-10-22:
I have found that there is already an old comment talking about the same thing:

https://github.com/cocotb/cocotb/blob/dc6ad51ebb387d36ed78389d7971717ab9935ec7/tests/designs/vhdl_configurations/Makefile#L5-L10

cmarqu, 2022-03-13:
Questa in recent versions has a `qrun` command which probably could be used to hand in all HDL source files at once and let the command figure out the correct order on its own (assuming it has that capability).

LoganBrist, 2022-04-27:
It appears the current workaround is https://github.com/cocotb/cocotb/issues/2941#issuecomment-1101613774.  for versions of Questa that do not have `qrun` (like mine),  you can copy the Questa makefile and manually reorder the compilation steps, roughly solving the problem.