cocotb/cocotb #1709: Verilator: Offer option to use FST Tracer

markusdd, 2020-04-25:
Hi,

very short proposal: Verilator includes FST-Wavedump support for some time now, enabling it would only require an option variabe and instantiating a `VerilatedFstC` instead of the vcd tracer.

FSTs are much more space efficient compared to vcd's .

cmarqu, 2020-04-25:
The main location that needs a change is probably this:
https://github.com/cocotb/cocotb/blob/66d1cad02c003d3e010993d186439394114041cb/cocotb/share/lib/verilator/verilator.cpp#L37-L43

markusdd, 2020-04-25:
Yes. For FST, it should something like this (without the switch yet):

```
#if VM_TRACE
    Verilated::traceEverOn(true);

    std::unique_ptr<VerilatedFstC> tfp(new VerilatedFstC);
    top->trace(tfp.get(), 99);
    tfp->open("dump.fst");
#endif
```

bluecmd, 2020-09-07:
+1! Another really cool feature is the support for enum values.

[image]

I tested the above patch and it worked (just needed an extra include as well, and needed to install libz-dev).

bluecmd, 2020-09-08:
@cmarqu Would you be open for a first time code contribution for this? It doesn't look that hard and it would benefit me so I can give it a shot.

themperek, 2020-09-08:
One could reuse `VM_TRACE` and add recognistion for `VM_TRACE=fst` or `VM_TRACE=dump.fst` ....

cmarqu, 2020-09-08:
@bluecmd Yes, sure!

markusdd, 2020-10-03:
Also, the makefile would need to change as well:

```
╰─ grep -ir Vcd .                                                                                                                                                                                           ─╯
./share/makefiles/simulators/Makefile.verilator:	@rm -f dump.vcd
```

markusdd, 2020-10-22:
Closing here as my PR was just accepted.

closed 2020-10-22 by markusdd