cocotb/cocotb #1710: README makefile missing export setting of coco share dir

markusdd, 2020-04-25:
Hi,

I was trying out the minimal example in the README to get cocotb also going on mac, but there are 2 obstacles:

a) you basically must use virtualenv, otherwise, even with homebrew version of python3, it will pull in python2 crap during compile
b) in the Makefile I need to do this:

`export COCOTB_SHARE_DIR=$(shell cocotb-config --share)`

otherwise I end up with this error:

```
make SIM=icarus                                                                                                                                                                 
/Users/xxxx/projects/cocotb_test/lib/python3.7/site-packages/cocotb/share/makefiles/Makefile.sim:52: *** "Couldn't find makefile for simulator: "icarus"! Available simulators: ".  Stop.
```

especially the export is important, just setting it does not suffice.

I might have overlooked this, but it doesn't say anywhere COCOTB_SHARE_DIR needs to be set manually.

themperek, 2020-04-26:
You use diffrent version of cocotb than examples. If you install one version of cocotb you have to use examples and documentation from same version.

markusdd, 2020-04-26:
I'm using 1.3.1, so latest greatest stable.
I was not able to find the place where it is mentioned that this needs to be set.
Just noticed when selecting 1.3.1 that this README example was not yet there, yet I hope you will agree it should be simple enough to get going anyway.

So what is the recommended env setup before using cocotb?
The Makefile in the quickstart guide of 1.3.1 does not specify setting it.

markusdd, 2020-04-26:
ok, found the problem.
1.3.1 wants the Makefile.inc included explicitly, on master in the README that is not the case anymore.

That can be closed then.

closed 2020-04-26 by markusdd

cmarqu, 2020-04-26:
Since #1712 we are explicit about the version dependency.