cocotb/cocotb #1715: pip install, can't run examples

alowell-ssl, 2020-04-26:
I'm using virtualenv with python 3.6.6 in Ubuntu 18.04 , just did pip install to get cocotb 1.3.1.  Running through the endian_swapper example, I type "make" and get:

/home/alowell/python-environments/env3/lib/python3.6/site-packages/cocotb/share/makefiles/Makefile.sim:53: *** "Couldn't find makefile for simulator: "icarus"! Available simulators: ".  Stop.

I think the issue is in Makefile.sim:
HAVE_SIMULATOR = $(shell if [ -f $(COCOTB_SHARE_DIR)/makefiles/simulators/Makefile.$(SIM_LOWERCASE) ]; then echo 1; else echo 0; fi;)

COCOTB_SHARE_DIR is not defined... if I then add the line:
COCOTB_SHARE_DIR := $(shell cocotb-config --share)

Things still don't work.  Running make, I then get:
make results.xml
make[1]: Entering directory '/home/alowell/Downloads/test/cocotb/examples/endian_swapper/tests'
cd ../cosim && make
make[2]: Entering directory '/home/alowell/Downloads/test/cocotb/examples/endian_swapper/cosim'
usage: cocotb-config [-h] [--prefix] [--share] [--makefiles] [-v]
cocotb-config: error: unrecognized arguments: --python-bin
make[2]: -c: Command not found
usage: cocotb-config [-h] [--prefix] [--share] [--makefiles] [-v]
cocotb-config: error: unrecognized arguments: --python-bin
make[2]: -c: Command not found
cc  -fPIC -pthread -fno-strict-aliasing -O2 -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 \
-fexceptions --param=ssp-buffer-size=4 -D_GNU_SOURCE \
-fwrapv -I -c io.c -o io_module.o
io.c:36:10: fatal error: Python.h: No such file or directory
 #include <Python.h>
          ^~~~~~~~~~
compilation terminated.
Makefile:23: recipe for target 'io_module.o' failed
make[2]: *** [io_module.o] Error 1
make[2]: Leaving directory '/home/alowell/Downloads/test/cocotb/examples/endian_swapper/cosim'
Makefile:65: recipe for target 'hal' failed
make[1]: *** [hal] Error 2
make[1]: Leaving directory '/home/alowell/Downloads/test/cocotb/examples/endian_swapper/tests'
/home/alowell/python-environments/env3/lib/python3.6/site-packages/cocotb/share/makefiles/Makefile.sim:80: recipe for target 'sim' failed
make: *** [sim] Error 2

cmarqu, 2020-04-26:
Just minutes ago, we improved the instructions in https://docs.cocotb.org/en/latest/quickstart.html#running-your-first-example (see #1712) - in short, you need to use the example files from the 1.3.1 release, not the development version of cocotb.

Please close the issue if this helped, thanks!

alowell-ssl, 2020-04-26:
haha, I noticed a change in the docs.  I'll give it a shot.

FYI all my problems went away when I switched to 1.3.0 version.

themperek, 2020-04-26:
Looks like problem solved.

closed 2020-04-26 by themperek

alowell-ssl, 2020-04-26:
Ok I was able to run the examples after switching to tag 1.3.1.

alowell-ssl, 2020-04-26:
BTW I'm trying to make my own makefile, following the directions in the quickstart guide, and I still get the "couldn't find makefile for simulator" error.  Is that simple example in the docs truly a minimal working example of a makefile?

themperek, 2020-04-26:
Do you look at proper version of documentation?

alowell-ssl, 2020-04-26:
Nope, missed the line to include Makefile.inc in the 1.3.1 documentation.  Thanks!