cocotb/cocotb #1716: Move utility scripts out of the cocotb package

ktbarrett, 2020-04-26:
We should introduce another namespace to put python libraries/scripts that don't depend on the cocotb runtime. For example, the build script.

ktbarrett, 2020-04-26:
In the case of the `build_ext` it probably should be merged into setup.py, as it is done in the [main python distribution](https://github.com/python/cpython/blob/master/setup.py).

eric-wieser, 2020-04-27:
But note that that setup.py imports `distutils`, so its still the case that not all the code is in one place. In my opinion the split aids clarity for now, so we should keep it as long as we do so in a way that doesn't cause other issues.

ktbarrett, 2020-04-27:
I tried just moving build libs into the top level, but the import fails when using tox. Perhaps there is some option to get tox to recognize it as a build dependency? CPython puts their `build_ext` overload in setup.py; it's a big file, I feel if they knew how to split it up, they would.

eric-wieser, 2020-04-27:
Did you remember to add it to `MANIFEST.in`?

PR #1719 (2020-04-27): Move build_libs.py out of cocotb package
Prevents the running of the cocotb package when importing. Need for unconditioned import of the simulator module. Closes #1716.
closed 2020-04-27 by eric-wieser