cocotb/cocotb #1722: Document behavior of Verilog system tasks

garmin-mjames, 2020-04-27:
Cocotb registers its own function calls for Verilog system tasks:

- `$info`
- `$warning`
- `$error`
- `$fatal`

Document behavior of these system tasks:
- Calls `gpi_log` with message
- `$fatal` fails the current test

eric-wieser, 2020-04-28:
Worth mentioning #130 too

gtaylormb, 2020-10-01:
Why doesn't $error fail the current test?

Calling $fatal in cocotb under Questa seems to crash the simulator.

ktbarrett, 2020-10-01:
@gtaylormb Yeah, attempting to overload `$fatal` was a bad idea. Most simulators explicitly state you can't overload builtin system tasks, which is why Questa isn't happy. I think the idea behind `$error` not failing tests was that those commands were meant only for logging purposes. `$error` logs with the Python logger at the `error` level. Calling `cocotb.log.error` in Python does not cause a test failure, so it shouldn't from HDL either.

We do need system tasks to inform the cocotb regression manager of test failure or passage to better support cocotb as a general purpose cosimulation framework. @eine and @umarcor are also interested in this.

garmin-mjames, 2020-10-01:
The Aldec simulators ignore the cocotb overloading and use the built-in ones.

umarcor, 2020-10-08:
ccing @LarsAsplund, as he is the one familiar with how SV's builtin tasks are combined with VUnit's HDL runner.

ktbarrett, 2020-10-08:
@umarcor You may be interested in #2094 specifically.

garmin-mjames, 2020-11-05:
These were removed in #2133. See #2094 for further discussion of cocotb system tasks.

closed 2020-11-05 by garmin-mjames