cocotb/cocotb #1724: Document cocotb module variables

ktbarrett, 2020-04-28:
The `cocotb` module has some important variables which aren't documented:

 * SIM_NAME
 * SIM_VERSION
 * RANDOM_SEED
 * log
 * scheduler
 * regression_manager
 * plusargs
 * argv/argc
 * LANGUAGE

cmarqu, 2020-04-28:
Some related notes:

https://docs.cocotb.org/en/latest/library_reference.html?highlight=plusargs#cocotb.plusargs should get a backreference to https://docs.cocotb.org/en/latest/building.html?highlight=plusargs#var-PLUSARGS

Same for ``RANDOM_SEED`` once we add a docstring (https://docs.cocotb.org/en/latest/building.html#envvar-RANDOM_SEED).

Also, ``plusargs`` should be moved out of the "Utilities" section.

What's up with the different cases? Should they all be lowercased, or ``plusargs`` uppercased since it also comes from an envvar? (But then, ``SIM_*``  do not come from envvars.)

ktbarrett, 2020-04-28:
Yeah, it's not very apparent that they are there, so I guess people forget about them. We could provide aliases with a consistent casing? Not sure that's worth it though. I wonder if there is some better way to make that information available.

PR #1848 (2020-05-23): Add the RegressionManager and cocotb module vars to documentation
`Scheduler` and `cocotb.scheduler` are documented, but `RegressionManager` and `cocotb.regression_manager` were not. None of the cocotb module level variables were either. This fixes #1724. This PR also contains a small bugfix in the documentation of `RANDOM_SEED`.

Note: the link to `TOPLEVEL_LANG` in the docs for `LANGUAGE` are broken because of #1847.
closed 2020-05-23 by ktbarrett