cocotb/cocotb #1746: Makefile.ghdl doesn't write to sim.log

cmarqu, 2020-05-01:
Our ``Makefile.ghdl`` doesn't have the usual `` | tee $(SIM_BUILD)/sim.log`` at the end of the simulation command line nor any other option to that effect, so it doesn't produce a ``sim.log`` file at all.

cmarqu, 2020-05-13:
Actually, this file ``sim.log`` is only used in about 50% of our Makefiles, the other ones are letting the simulator write whatever file it likes.
Since the ``| tee`` is breaking the ``isatty`` detection for color output, we should rather try to remove this.

cmarqu, 2020-10-08:
The ``| tee`` is our way to collect all log output, even the lines before and after the actual simulator run.

garmin-mjames, 2020-10-08:
There's some issue with **activehdl**, `tee`, and `set -o pipefail`. On some tests and examples when running regression suite it fails even if the tests pass and the simulator exits normally.

I haven't been able to track down the exact details, but it looked like somehow `tee` is exiting with a non-zero status.

cmarqu, 2020-10-08:
@garmin-mjames Huh. Probably worth its own issue?

garmin-mjames, 2020-10-08:
Yeah I'll try to track it down more precisely and make an issue

PR #3927 (2024-06-10): Rename `SIM_OUTPUT_REDIRECT` to `SIM_CMD_SUFFIX` and add to all Makefiles
Closes #1746. Adds `SIM_OUTPUT_REDIRECT` to all Makefiles, and then defaults it back to empty to prevent the redirection. 

From that issue:

> Since the | tee is breaking the isatty detection for color output, we should rather try to remove this.

This does it in a non-destructive way by making it opt-in.
closed 2024-06-10 by ktbarrett