cocotb/cocotb #1749: Publish a gitpod docker image to allow downstream projects to use gitpod

eric-wieser, 2020-05-02:
> It might be nice to put the `DockerFile` in its own repo, and find a way to publish it to hub.docker.com.
>
> That way people using cocotb in downstream repos can easily rely on the docker image to support gitpod themselves.

_Originally posted by @eric-wieser in https://github.com/cocotb/cocotb/pull/1683#issuecomment-617253914_

themperek, 2020-05-02:
Can also enable prebuilds for PRs: https://www.gitpod.io/docs/prebuilds/

themperek, 2020-05-02:
> Can also enable prebuilds for PRs: https://www.gitpod.io/docs/prebuilds/

@imphil Can you enable this if you think usefull?

cmarqu, 2020-12-22:
@imphil Could you create a new repo for this please, e.g. `cocotb/docker-gitpod` ?

I would put our `.gitpod.Dockerfile` into it (like in https://hub.docker.com/r/cmarqu/cocotb-docker-gitpod/dockerfile)
and then create a PR for https://github.com/cmarqu/uvm-python/tree/add-gitpod, and potentially most of the other `cocotbext-*` packages.

ktbarrett, 2026-04-08:
Gitpod has come and gone. Long live Gitpod.

closed 2026-04-08 by ktbarrett