cocotb/cocotb #1758: Unable to run mean example with questa.

m-kru, 2020-05-04:
I try to run mean example with questa simulator, however:
```
# Unable to open lib libcocotbfli_modelsim.so (libcocotbfli_modelsim.so: cannot open shared object file: No such file or directory)
# cocotb: Error loading shared library libcocotbfli_modelsim.so
```
`locate libcocotbfli_modelsim.so` returns nothing.

eric-wieser, 2020-05-04:
#1725 added a better error message for this.

You maybe need to reinstall cocotb, if you installed modelsim after it. Otherwise, your questa version may just not support FLI.

m-kru, 2020-05-04:
This library should be provided by questa, right?

themperek, 2020-05-04:
@m-kru See https://docs.cocotb.org/en/latest/simulator_support.html#mentor-modelsim

m-kru, 2020-05-04:
Thanks, closing `vdbg` was not in the PATH during installation.

m-kru, 2020-05-04:
Ok, maybe I have closed this issue to quickly.

```
# Loading /tmp/mkru@arch_dpi_20587/linux_x86_64_gcc-9.3.0/STUB_SYMS_OF_libcocotbvpi_modelsim.so
# Loading /home/mkru/.local/lib/python3.8/site-packages/cocotb-1.4.0.dev0-py3.8-linux-x86_64.egg/cocotb/libs/libcocotbvpi_modelsim.so
#      -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:71   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
# ** Fatal: (vsim-12005) Undefined function 'Py_Initialize' introduced from '/home/mkru/.local/lib/python3.8/site-packages/cocotb-1.4.0.dev0-py3.8-linux-x86_64.egg/cocotb/libs/libcocotb.so' is being called. Exiting ... 
# End time: 16:33:21 on May 04,2020, Elapsed time: 0:00:01
# Errors: 1, Warnings: 0
```

m-kru, 2020-05-11:
@eric-wieser @themperek do you know what might be wrong, why it reports that `Py_Initialize` is undefined?

themperek, 2020-05-11:
@m-kru I do not know. I had such in the past do not remember at the moment why. Does Icarus work?

m-kru, 2020-05-11:
It has mixed sources VHDL/Verilog. Icarus does not support this.

themperek, 2020-05-11:
Not mixed language examples are ok?

m-kru, 2020-05-11:
No matter what example I run with questa simulator, it always ends up with `Undefined function 'Py_Initialize'`

elgorwi, 2020-05-11:
The mean example has just been updated #1778 . It has been tested with modelsim. The non-mixedlang systemverilog version can now be run with Icarus as well. 
I recommend pulling the latest from master and reinstalling from scratch. Then try running the adder example.
I’ve noticed some issues with the libs not rebuilding properly on my machine, so now I always delete the cocotb/build directory before reinstalling cocotb.

m-kru, 2020-05-12:
I always `git clean -xdf` before building. It looks like this issue is generally related to `questa`, I have tried all examples, with mixed or SystemVerilog sources only.

elgorwi, 2020-05-12:
Can you run the examples with Icarus?

m-kru, 2020-05-12:
Yes, running examples with Icarus is fine, so is running examples with GHDL (in case of full VHDL examples).

elgorwi, 2020-05-12:
Can we have some more details such as:
- os version
- questa version
- how you installed cocotb

m-kru, 2020-05-12:
Linux arch 5.6.11-arch1-1
Questa Sim-64 vsim 10.6b Simulator 2017.05 May 25 2017
`python setup.py install --user`
vsim is in the `$PATH` during installation process.

elgorwi, 2020-05-12:
I don’t know much about the installation process, bit afaik pip is needed for installation. @themperek should know more about this.

themperek, 2020-05-13:
`python setup.py install --user` should work

m-kru, 2020-05-21:
The issue is probably related to linking to python embed library during compilation. Both `python3-embed` and `python-3.8-embed` are present on my system, so it should be possible to call `Py_Initialize`. Anyone fluent in cocotb library build process?

themperek, 2020-06-01:
Can you get paste log file when `pip install -v` and when running the test after `make clean`?

jrri, 2020-09-13:
Hello, I would like to ask for help in this issue as well. My system settings:

Ubuntu 18.04
Questasim 64-bit 10.5b_1
cocotb 1.4.0
Python 3.6

* cocotb was installed using: pip3 install cocotb
* Installation of cocotb was done after the installation of questa

I get the exact same error when trying to run the adder example (make SIM=questa):

```
# Loading sv_std.std
# Loading work.adder(fast)
# Loading /tmp/unknown@b5cd0c757113_dpi_2237/linux_x86_64_gcc-4.7.4/STUB_SYMS_OF_libcocotbvpi_modelsim.so
# Loading /usr/local/lib/python3.6/dist-packages/cocotb/libs/libcocotbvpi_modelsim.so
#      -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:74   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
# ** Fatal: (vsim-12005) Undefined function 'Py_Initialize' introduced from '/usr/local/lib/python3.6/dist-packages/cocotb/libs/libcocotb.so' is being called. Exiting ... 
# End time: 21:42:33 on Sep 13,2020, Elapsed time: 0:00:01
# Errors: 1, Warnings: 0
```

ktbarrett, 2020-09-13:
@jrri I opened #2071 for your issue. Don't hijack other issues, you don't know if they are related (and I suspect it isn't).

ktbarrett, 2025-10-26:
This issue is probably resolved by now. There have been numerous improvements to library (include libpython) discovery in the last 5 years.

closed 2025-10-26 by ktbarrett