cocotb/cocotb #1759: Update vhpi_user.h to follow standard's implementation

ktbarrett, 2020-05-04:
From https://github.com/cocotb/cocotb/pull/1742#issuecomment-622599691

The standards committee provides an implementation of vhpi_user.h, we should be using their definition if possible. I'm not sure from where the original definition was sourced, or why it was changed along the way. Perhaps these definitions are necessary to make certain simulators work? I think we should investigate.

eric-wieser, 2020-05-04:
Github has the history a little fragmented, but you can see most of it here: https://github.com/cocotb/cocotb/commits/9ba81fe3f7a129b4888a519df8f9645d811a90a7/include/vhpi_user.h

themperek, 2020-05-04:
VCS usees this one: https://github.com/jamiekang/embedded-sw-dev-tools/blob/master/binsim/vhpi_user.h (if we ever would like to do VHPI in VCS). So we may finish with few versions,

eric-wieser, 2020-05-04:
Perhaps we should go the other way and not ship one at all, instead building against the one we find

ktbarrett, 2020-05-04:
In an ideal world yes we would build against every simulator's vhpi_user.h, but that means we have to write an implementation of VHPI that works with all support's simulator's header impl and then we have to add support for finding that header file for each supported simulator to the build script. That's a lot of work...

I have had the idea floated to me by @leftink that it might actually be easier to have simulator-specific libraries rather than interface-specific libraries. That way we can contain in one object all supported interfaces of a simulator and handle all the weird quirks of that simulator (like not implementing certain callbacks, using special vhpi_users.h, etc.). It might also make supporting new simulators easier since we don't have to try to write an implementation that works for *everyone* and end up breaking someone else in the proces.