cocotb/cocotb #1761: Link on Github Front not working

silvan-hdl, 2020-05-04:
[image]

The link on the first page is not working.

silvan-hdl, 2020-05-04:
https://cocotb.org

eric-wieser, 2020-05-04:
Used to work. Thanks for the report.

imphil, 2020-05-04:
Either this got automatically switched to https, or I did it and didn't double-check. Either way, http works, http doesn't. (It's only a redirect, and we don't have a full cert setup for that.)

closed 2020-05-04 by imphil

eric-wieser, 2020-05-04:
> (It's only a redirect, and we don't have a full cert setup for that.)

Perhaps it would be better to put the docs link there until we have a cert setup.

An insecure redirect is as bad as an insecure site. While it's not a big deal if someone MITMs our docs, if the page is just a redirect anyway there's likely little point linking there.

imphil, 2020-05-04:
done.