cocotb/cocotb #1768: Is yield planned to be deprecated ?

Martoni, 2020-05-05:
With async framework we now can «block» on events with two methods.
Will these two methods continue to coexist?
Or is it planned to abandon the yield method?

eric-wieser, 2020-05-05:
> Will these two methods continue to coexist?

At least for the next release, yes

> Or is it planned to abandon the yield method?

Doing so would break a very large number of existing users. At the very least we'd want to wait for cocotb 2.0 before doing that, or possibly even longer.

Having said that, if you have the ability to choose which style to use today, you should choose `await`!

Martoni, 2020-05-05:
> Having said that, if you have the ability to choose which style to use today, you should choose await!

Ok, that's what I wanted to know.

Thanks

closed 2020-05-05 by Martoni