cocotb/cocotb #1775: Makefile for icarus not found

crusader2000, 2020-05-07:
I am trying to test a testbench. when I run the make command , I get the following error. I tried changing the location of Makefile.icarus but it didnt work out. 

```/home/ansh/.local/lib/python3.7/site-packages/cocotb/share/makefiles/Makefile.sim:52: *** "Couldn't find makefile for simulator: "icarus"! Available simulators: ".  Stop. ```

Can someone help with this error?

themperek, 2020-05-07:
Your installed version is diffren tthan examples see:

themperek, 2020-05-07:
Your installed version is different then examples you use see: https://docs.cocotb.org/en/latest/quickstart.html#running-your-first-example

Please close if this solves your issue.

crusader2000, 2020-05-07:
Hello,
This doesnt clear my issue. I am trying to run a custom testbench file.

themperek, 2020-05-07:
1. What is cocotb version you have installed `cocotb-config --version`?
2. How does your control makefile looks like?

crusader2000, 2020-05-07:
1) 1.3.0

2) 
Makefile -
```
SIM = icarus
TOPLEVEL_LANG = verilog

VERILOG_SOURCES = $(PWD)/prga_fifo.v
# TOPLEVEL is the name of the toplevel module in the Verilog or VHDL file_
TOPLEVEL=prga_fifo
# MODULE is the name of the python testfile
MODULE=prga_fifo_cocotb

include $(shell cocotb-config --makefiles)/Makefile.sim
include $(shell cocotb-config --makefiles)/Makefile.inc
```

themperek, 2020-05-07:
A wrong Makefile includes order?
```make
include $(shell cocotb-config --makefiles)/Makefile.inc
include $(shell cocotb-config --makefiles)/Makefile.sim
```

crusader2000, 2020-05-07:
That worked

closed 2020-05-07 by crusader2000