cocotb/cocotb #1785: Array value assignment and access

crusader2000, 2020-05-10:
I am trying to assign value to an input array. 
```
input [7:0] src [0:1023];
```
I tried using this syntax 
```

# Assign a value to a memory deep in the hierarchy
dut.src[0] <= 190
``` 
But I get the following error. 
```
IndexError: src(GPI_REGISTER)  contains no object at index 0
```

How to resolve this?

themperek, 2020-05-10:
If you want increase chances that someone will know I would suggest to provide:
- cocotb version 
- simulator version
- os
- minimum reproducable example

crusader2000, 2020-05-11:
Cocotb version -1.3.0
Simulator - iverilog - 10.3
OS - Fedora 29
Python file - https://github.com/crusader2000/prga/blob/dev/cocotb/fifo/prga_fifo_cocotb.py
Top Module - https://github.com/crusader2000/prga/blob/dev/cocotb/fifo/prga_fifo_tb_wrapper.v

elgorwi, 2020-05-11:
Have a look at the mean example. It has just been updated to support Icarus and does assign values to an array. I recommend pulling the latest from master and reinstalling. Then try running the mean example.

ktbarrett, 2020-11-17:
This could be resolved in #2079. Try [installing the development version](https://docs.cocotb.org/en/stable/install_devel.html#install-devel) and report back if you are still experiencing this problem.

closed 2020-11-25 by github-actions[bot]