cocotb/cocotb #1786: Using icarus verilog master on Windows throws error

tpambor, 2020-05-11:
Using latest icarus verilog from master fails with:
```
/mingw64/bin/vvp -M /mingw64/lib/python3.8/site-packages/cocotb/libs -m libcocotbvpi_icarus   sim_build/sim.vvp
libcocotbvpi_icarus:`C:/msys64/mingw64/lib/python3.8/site-packages/cocotb/libs\libcocotbvpi_icarus.vpl' failed to open using dlopen() because:
    %1 is not a valid Win32 application.
```
Starting with [a59b183b](https://github.com/steveicarus/iverilog/commit/a59b183bd1890eff63cc59d38084dde15de281ff) icarus verilog on windows no longer exports `vpi_*` in vvp.exe. As a result, the current libcocotbvpi_icarus.vpl is unable to load because it no longer can import these symbols.

eric-wieser, 2020-05-11:
> Starting with a59b183b icarus verilog on windows no longer exports vpi_* in vvp.exe.

Does icarus supply any documentation for what is expected from VPI applications now?

eric-wieser, 2020-05-11:
Also, we should probably add a CI run against icarus master to verify this is broken. We can mark it an expected failure like we do with GHDL.

tpambor, 2020-05-15:
CI reproduces the error, [https://travis-ci.org/github/cocotb/cocotb/jobs/687371303#L1911](https://travis-ci.org/github/cocotb/cocotb/jobs/687371303#L1911).

garmin-mjames, 2020-11-27:
This was fixed in https://github.com/steveicarus/iverilog/issues/395

closed 2020-11-27 by garmin-mjames