cocotb/cocotb #1814: crv.py module

tsengr0916, 2020-05-13:
We used  to have crv.py module in cocotb. It is not in there any more. Is there a reason for that? What constraint random module would you recommend to use now? And where can we download it from? 

Thanks,
Richard

ktbarrett, 2020-05-13:
If that file was in this repo, it should be in the history. Otherwise you could take a look at [cocotb-coverage](https://github.com/mciepluc/cocotb-coverage).

garmin-mjames, 2020-05-13:
See also https://github.com/cocotb/cocotb/wiki/Further-Resources for other cool stuff.

tsengr0916, 2020-05-14:
Thank you all very much. This is helpful!!

There is another python-constraint at this link seems to be a global wised effort for constraint solver: https://github.com/python-constraint/python-constraint

Could we integrate it into COCOTB constraint solver engine?

Richard

themperek, 2020-05-14:
>There is another python-constraint at this link seems to be a global wised effort for constraint solver: https://github.com/python-constraint/python-constraint

[cocotb-coverage](https://github.com/mciepluc/cocotb-coverage) is using it.

>Could we integrate it into COCOTB constraint solver engine?

The consensus is at the moment this should be an external module.

themperek, 2020-05-14:
@tsengr0916 I think this answers your question so I close. Fell free to reopen if it is not the case.

closed 2020-05-14 by themperek