cocotb/cocotb #1819: Control clock based on signal

MansurHu, 2020-05-14:
I want to control my VHDL clock based on a output change.
currently I am doing:
    while dut.trigger_out !=1:
        yield RisingEdge(dut.clk) #execute read or write in VHDL  

The problem with this is that there is constant communication from my Cocotb test and the simulator, which affects the performance.
I read that there is a solution with the fork() function. But the fork function must be used in combination with the Timer() function and for the Timer function a time value must be specified.
But I dont know how long the process in VHDL takes before the value of trigger_out changes.
Any ideas how I can control the clock based on a output? (with minimal communication between Cocotb and the simulator)

gretzteam, 2020-05-14:
What about waiting directly on a Rising or Falling edge of dut.trigger?

while True:
  await RisingEdge(dut.trigger_out)
  do stuff

themperek, 2020-05-14:
Maybe look at #1676  ?

MansurHu, 2020-05-14:
> What about waiting directly on a Rising or Falling edge of dut.trigger?
> 
> while True:
> await RisingEdge(dut.trigger_out)
> do stuff

And fork the clock at the same time i geuss. Because I need to keep passing simulation time

gretzteam, 2020-05-14:
> And fork the clock at the same time I guess. Because I need to keep passing simulation time
If you're generating the clock from cocotb (forked prior to the code you're showing), then there isn't much you can do to improve performance as the clock will dominate.

I discussed this here and got help to improve performance by quite a bit in subsequent post. 
https://gitter.im/cocotb/Lobby?at=5eac387e7975db7ebfda7ded

Now generating the clock on the HDL side, and keeping cocotb monitor only your dut.trigger_out will give you a solid boost. I saw 20x in my example but that probably depends on the clk/trigger_out ratio.