cocotb/cocotb #1821: Style guide for HDL examples

imphil, 2020-05-14:
It would be nice to give our HDL coding examples a uniform style. Ideally, we can find two style guides which are similar between VHDL and Verilog.

For Verilog I'd go with https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md (mostly because I'm quite familiar with it, and it is the most complete Verilog style guide I know with linter and formatter support)

For VHDL I found https://github.com/jeremiah-c-leary/vhdl-style-guide quite useful, but I'm not writing VHDL day-to-day so I'm open to suggestions.

gretzteam, 2020-05-14:
For Verilog, where is the formatter (automatic?)

ktbarrett, 2020-05-14:
We really should be using `default_nettype none` as well. I don't see that mentioned in the style guide, but agree with everything else it says.

imphil, 2020-05-14:
@gretzteam one formatter compatible with that style guide is at https://github.com/google/verible.

ktbarrett, 2020-05-14:
Should this stick to code style only, or consider behavioral style, like prefer synchronous resets, no latches, examples (not test deisgns) should be synthesizable, etc?

As for the VHDL, I'd like a written out style guide like what was provided for Verilog, a linter is nice but doesn't suggest "best practices". Also I really disagree with some of the decisions the posted VHDL linter makes (I think you can turn them off though).

It's probably not necessary to run a linter in CI, just when changing HDL source files.