cocotb/cocotb #1829: ImportError: from cocotb import simulator

elgorwi, 2020-05-15:
Just pulled the latest from master and build is broken on windows mingw 32-bit.
All is working fine using version b08bf0ce74144886c17e6f824438890d08503a71
Could be related to this change:
https://github.com/cocotb/cocotb/commit/3a0bf9460515ae0c2269890b75ddbeaa80085cff

Here's a log when running the adder example: 
``` make SIM=modelsim
Traceback (most recent call last):
  File "C:\msys32\mingw32\bin\cocotb-config-script.py", line 11, in <module>
    load_entry_point('cocotb', 'console_scripts', 'cocotb-config')()
  File "C:/msys32/mingw32/lib/python3.8/site-packages/pkg_resources/__init__.py", line 490, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "C:/msys32/mingw32/lib/python3.8/site-packages/pkg_resources/__init__.py", line 2859, in load_entry_point
    return ep.load()
  File "C:/msys32/mingw32/lib/python3.8/site-packages/pkg_resources/__init__.py", line 2450, in load
    return self.resolve()
  File "C:/msys32/mingw32/lib/python3.8/site-packages/pkg_resources/__init__.py", line 2456, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "C:/work/cocotb/cocotb/__init__.py", line 42, in <module>
    import cocotb.handle
  File "C:/work/cocotb/cocotb/handle.py", line 36, in <module>
    from cocotb import simulator
ImportError: DLL load failed while importing simulator: The specified module could not be found.
Makefile:47: /Makefile.sim: No such file or directory
make: *** No rule to make target '/Makefile.sim'.  Stop.
```
Any suggestions?

eric-wieser, 2020-05-15:
Can you try uninstalling and reinstalling cocotb, in case you've ended up with a mixture of installations?

@ktbarrett, any ideas?

themperek, 2020-05-15:
I can reproduce.

It looks this has to be removed (for 3.8 it does not do what we expect):
https://github.com/cocotb/cocotb/blob/bd4f5c4ef7827758c99a54ae8ad41f0f035c70a4/cocotb/_os_compat.py#L17-L19

eric-wieser, 2020-05-15:
Why does this work in CI? Is our windows CI not 3.8?

themperek, 2020-05-15:
I think conda is still 3.7 by default 

Maybe we can test more but would move to  azure/GitHub for this maybe?

ktbarrett, 2020-05-15:
I guess I never checked, and I guess it worked by accident, the directory is wrong...

https://github.com/cocotb/cocotb/blob/bd4f5c4ef7827758c99a54ae8ad41f0f035c70a4/cocotb/_os_compat.py#L14

That should be `extra_dll_dir = os.path.dirname(__file__)`. The simulator module is not in 'libs'. Could you confirm @themperek?

themperek, 2020-05-15:
The problem is not `simulator` module but dependencies from 'libs' folder.

eric-wieser, 2020-05-15:
Is there any way to find out exactly which dll is failing to load?

tpambor, 2020-05-15:
On Windows, probably something like rpath can be done using manifest files. See [https://bugs.python.org/issue35688#msg335667](https://bugs.python.org/issue35688#msg335667) for a solution for a similar issue.

eric-wieser, 2020-05-15:
Perhaps the thing to do for the imminent 1.4 release is to just set `PATH` on all python versions, and come back to this later.

tpambor, 2020-05-15:
[image]
For the records here is a trace. It fails to load any dependencies in the libs folder

eric-wieser, 2020-05-15:
Can you get the `AddDllDirectory` call in that trace too?

tpambor, 2020-05-15:
Nope, these are only the syscalls. `AddDllDirectory` probably is handled in userspace.

eric-wieser, 2020-05-15:
Which simulator are you using, @elgorwi?

eric-wieser, 2020-05-15:
It's possible we need something like https://github.com/numpy/numpy/pull/13019

elgorwi, 2020-05-16:
> 
> 
> I can reproduce.
> 
> It looks this has to be removed (for 3.8 it does not do what we expect):
> 
> https://github.com/cocotb/cocotb/blob/bd4f5c4ef7827758c99a54ae8ad41f0f035c70a4/cocotb/_os_compat.py#L17-L19

Thanks @themperek, commenting out these lines resolves the issue.

elgorwi, 2020-05-16:
> 
> 
> Which simulator are you using, @elgorwi?

I'm using Modelsim SE 10.2c

themperek, 2020-05-16:
@elgorwi 
>I can open a PR for the proposed fix, or do we want to proceed with #1836 ?

Not sure if we manage with #1836 for 1.4 release (next days) so probably fast "fix" would be good to have.

elgorwi, 2020-05-17:
> 
> 
> @elgorwi
> 
> > I can open a PR for the proposed fix, or do we want to proceed with #1836 ?
> 
> Not sure if we manage with #1836 for 1.4 release (next days) so probably fast "fix" would be good to have.

OK, opened PR #1841

eric-wieser, 2020-05-28:
Removing the 1.4 milestone since #1829 stops the bleeding

PR #1836 (2020-08-14): Support rpath-like "<probing privatePath='' />" on Windows
This PR implements support for the [probing element](https://docs.microsoft.com/en-us/dotnet/framework/configure-apps/file-schema/runtime/probing-element) which can be used to specify a rpath-like search path for a assembly which for us equals to a shared library (.dll).

A `Manifest` is embedded into the shared libraries on Windows. On load, these libraries will then be handled via the [SxS](https://en.wikipedia.org/wiki/Side-by-side_assembly) loader. The loader reads the `*.2.config` configuration file. In the configuration file, usi [... cut]
closed 2020-08-14 by ktbarrett