cocotb/cocotb #1833: Update CoCoTB to allow test to access constants defined in packages

jwprice100, 2020-05-15:
When working with VHDL designs, it's common to have constants defined in packages for use across many source files. At the moment, the only way to access those constants is to create copies in the dut so that cocotb tests can access them. While this work around is functional, it's a bit inelegant. It would be nice to able to get a handle to other objects such as packages. I understand this is possible, but might require modifying the C++ back-end.

garmin-mjames, 2020-05-16:
A `packages` attribute could be added to `simulator` with handles to the package instantiations.

For VHPI, here's an example snippet from the language ref manual showing how to get the packages:
```c
/* if not found in the design hierarchy, search for the signal
in the instantiated packages
*/
itr = vhpi_iterator(vhpiPackInsts, NULL);
if (itr) {
  while (hdl = vhpi_scan(itr)) {
    sigHdl = vhpi_handle_by_name(sigName, hdl);
    if (sigHdl) {
      return sigHdl;
    }
  }
}
```

andrewandrepowell, 2020-08-02:
Hello,

I plan to work on this issue if no one else is doing so already. I propose the following interface:

```python

@test()
def some_test(dut, packages) -> None:
    SOME_CONSTANT = packages["package_name"].SOME_CONSTANT.value

```

The packages parameter can be an optional argument the user can add to their test to retrieve a dictionary with all the packages inside it.

eric-wieser, 2020-08-02:
I think making it a function or attribute on the simulator module is probably easier

andrewandrepowell, 2020-08-02:
I was thinking maybe during the `_initialize_testbench` the packages can all be discovered and stored, that way the user wouldn't have to directly use anything in the simulator module. Are you suggesting only have an interface to the packages from the simulator module only?

ktbarrett, 2020-08-03:
@andrewandrepowell I agree with Eric. The discovery will have to go through the simulator module anyways. As for pythonland abstraction around that, I'm not a fan of adding it to the test as an optional function. In fact I'm for the opposite (#1827). If we want to support alternative test frameworks putting less requirements on it is better.

garmin-mjames, 2020-08-03:
One more vote for a method on `simulator`. There is already `simulator.get_root_handle`, so probably `simulator.get_packages`.

I think it makes sense to have each package be a `HierarchyObject`, and we should check each GPI `registered_impls` for the mixed-language case.

I'm not sure what `get_packages` should return, though:
1. tuple of package objects
2. "fake" `HierarchyObject` named `packages`, which would allow `packages.pkgName` named attribute access vs iterating over tuple and checking the name

eric-wieser, 2020-08-03:
@garmin-mjames, 2 sounds good to me. I don't think we need a `HeirarchyObject`, something simple like `types.SimpleNamespace` would work fine.

If we really wanted to go off the rails here, we could expose this as `import cocotb.vhdl_packages.my_package_name`, and put something in `sys.meta_path` which makes the vhdl modules look like python modules.

ktbarrett, 2020-08-03:
Retuning a `SimpleNamespace` or even just a dictionary of packages sounds good to me.

ktbarrett, 2020-08-03:
There is also the problem of GHDL only supporting VPI which does not have a notion of packages. AFAICT, there is no way to inspect packages in GHDL.

garmin-mjames, 2020-08-03:
@ktbarrett Yeah it looks like they only support **IEEE 1364-2005** VPI (partially) at the moment.

ktbarrett, 2020-08-03:
The SV VPI extension supports `vpiPackage`, but I don't think GHDL supports SV VPI extensions.

garmin-mjames, 2020-08-03:
GHDL has `Get_Package_Inst`:
https://github.com/ghdl/ghdl/blob/657fcfde5fb93c1311fe5fd2d28146c89852614d/src/grt/grt-avhpi.ads#L427

It might not be too much work to support `vpiPackage` only.

marlonjames, 2022-03-21:
It has not, and I'm not aware of anyone currently working on it.

mrv96, 2022-08-26:
I'm interested to this feature too

slagernate, 2022-10-05:
Also interested.

ktbarrett, 2022-10-12:
One way to accomplish this right now is to use the following.

```python
my_pack = cocotb.handle.SimHandle(cocotb.simulator.get_root_handle("my_pack"))
```

marlonjames, 2023-02-16:
@ktbarrett Does that work with VHPI? It separates root instantiation and packages into `vhpi_handle(vhpiRootInst, NULL)` and `vhpi_iterator(vhpiPackInsts, NULL)`

darrenxonaspace, 2023-10-24:
Also desire this feature

nickg, 2024-05-18:
> @ktbarrett Does that work with VHPI? It separates root instantiation and packages into `vhpi_handle(vhpiRootInst, NULL)` and `vhpi_iterator(vhpiPackInsts, NULL)`

I think it works because `VhpiImpl::get_root_handle()` falls back to calling `vhpi_handle_by_name()` if `vhpi_handle(vhpiRootInst, NULL)` doesn't return anything. Anyway I had a go at adding proper support in #3899.