cocotb/cocotb #1839: Abort (Core dumped) when running unittests with uvm-python

tpoikela, 2020-05-16:
Running `make test` (or any unittest file) for https://github.com/tpoikela/uvm-python results in the following error:

```
/usr/include/c++/9/bits/stl_vector.h:1042: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = GpiImplInterface*; _
Alloc = std::allocator<GpiImplInterface*>; std::vector<_Tp, _Alloc>::reference = GpiImplInterface*&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
```

I bisected the issue to commit 3a0bf9460515ae0c2269890b75ddbeaa80085cff. It seems some imports have been changed to unconditional ones. More specifically, changing cocotb/utils.py back to conditional import seems to fix this issue. Any ideas on the source of error?

Background: uvm-python unit tests are run without simulator, but files under test are still importing stuff from cocotb.

To reproduce the issue:

```
mkdir temp
cd temp
python -m env .venv
source .venv/bin/activate
git clone https://github.com/cocotb/cocotb.git
cd cocotb
pip install -e .
cd ..
git clone https://github.com/tpoikela/uvm-python.git
cd uvm-python
ln -s src/uvm ./uvm
pip install cocotb_coverage pip install regex
python -m unittest unit/test_sv.py  # Or make test-unit
```

eric-wieser, 2020-05-16:
Right, I think the key difference is that `cocotb.utils._LOG_SIM_PRECISION` used to default to `1e-15` when not in a simulator, but now it's replacement assumes that anyone requestion something in simsteps must be connected to a simulator.

The crash looks like asking for item 0 from an empty list of simulators.

eric-wieser, 2020-05-16:
Can you enable the PYTHONFAULTHANDER mechanism to get a python stack trace?

tpoikela, 2020-05-16:
```
Current thread 0x00007fbec61676c0 (most recent call first):
  File "/home/tpoikela/temp/cocotb/cocotb/utils.py", line 43 in _get_simulator_precision
  File "/home/tpoikela/temp/cocotb/cocotb/utils.py", line 100 in get_time_from_sim_steps
  File "/home/tpoikela/temp/cocotb/cocotb/utils.py", line 73 in get_sim_time
  File "/home/tpoikela/temp/uvm-python/uvm/base/uvm_globals.py", line 437 in uvm_sim_time
  File "/home/tpoikela/temp/uvm-python/uvm/base/uvm_report_server.py", line 487 in compose_report_message
  File "/home/tpoikela/temp/uvm-python/uvm/base/uvm_report_server.py", line 358 in process_report_message
  File "/home/tpoikela/temp/uvm-python/uvm/base/uvm_report_handler.py", line 319 in process_report_message
  File "/home/tpoikela/temp/uvm-python/uvm/base/uvm_report_object.py", line 224 in uvm_process_report_message
  File "/home/tpoikela/temp/uvm-python/uvm/base/uvm_report_object.py", line 147 in uvm_report
  File "/home/tpoikela/temp/uvm-python/uvm/base/uvm_report_object.py", line 163 in uvm_report_info
  File "/home/tpoikela/temp/uvm-python/uvm/base/uvm_root.py", line 897 in report_header
  File "/home/tpoikela/temp/uvm-python/uvm/base/uvm_root.py", line 152 in __init__
  File "/home/tpoikela/temp/uvm-python/uvm/base/uvm_root.py", line 184 in m_uvm_get_root
  File "/home/tpoikela/temp/uvm-python/uvm/base/uvm_coreservice.py", line 69 in get_root
  File "/home/tpoikela/temp/uvm-python/uvm/base/uvm_root.py", line 133 in get
  File "/home/tpoikela/temp/uvm-python/uvm/base/uvm_root.py", line 946 in <module>
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 728 in exec_module
  File "<frozen importlib._bootstrap>", line 677 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 967 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 983 in _find_and_load
  File "/home/tpoikela/temp/uvm-python/uvm/base/__init__.py", line 34 in <module>
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 728 in exec_module
  File "<frozen importlib._bootstrap>", line 677 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 967 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 983 in _find_and_load
  File "/home/tpoikela/temp/uvm-python/uvm/__init__.py", line 2 in <module>
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 728 in exec_module
  File "<frozen importlib._bootstrap>", line 677 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 967 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 983 in _find_and_load
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 953 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 983 in _find_and_load
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 953 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 983 in _find_and_load
  File "/home/tpoikela/temp/uvm-python/unit/test_sv.py", line 3 in <module>
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 728 in exec_module
  File "<frozen importlib._bootstrap>", line 677 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 967 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 983 in _find_and_load
  File "/usr/lib64/python3.7/unittest/loader.py", line 154 in loadTestsFromName
  File "/usr/lib64/python3.7/unittest/loader.py", line 220 in <listcomp>
  File "/usr/lib64/python3.7/unittest/loader.py", line 220 in loadTestsFromNames
  File "/usr/lib64/python3.7/unittest/main.py", line 159 in createTests
  File "/usr/lib64/python3.7/unittest/main.py", line 147 in parseArgs
  File "/usr/lib64/python3.7/unittest/main.py", line 100 in __init__
  File "/usr/lib64/python3.7/unittest/__main__.py", line 18 in <module>
  File "/usr/lib64/python3.7/runpy.py", line 85 in _run_code
  File "/usr/lib64/python3.7/runpy.py", line 193 in _run_module_as_main
[3]    9006 abort (core dumped)  PYTHONFAULTHANDLER=1 python -m unittest unit/test_sv.py
```
I have this statement in uvm_globals.py:
```
from cocotb.utils import get_sim_time, simulator
...
# Further down in the file:
def uvm_sim_time(units='NS'):
    if simulator is not None:
        return get_sim_time(units=units) # Line 437
````

Apparently, the imported simulator variable used to be None when running without a simulator. Is there a method is `simulator.is_active()` or something similar?

eric-wieser, 2020-05-16:
Right, that is indeed your problem. I don't remember whether we provided a way to tell if you're in the simulator. @ktbarrett had some ideas here, but I don't remember whether we finished implementing them.

Milestoning as 1.4, since we at least need to provide you a way to detect this.

themperek, 2020-05-16:
Maybe that can do that `get_simulator_product`  (all other functions?) return `None` and not do `Segmentation fault` if not in a simulation.

ktbarrett, 2020-05-16:
I also need a way to determine if I am running in a cocotb context or not. I think everything in the simulator module should throw an exception if no simulator is active. The issue is detecting no simulator running without invoking seg faults.

eric-wieser, 2020-05-16:
Unfortunately the gpi c interface doesn't really have a design for error propagation.

eric-wieser, 2020-05-16:
One way we could go is to move all `cocotb.simulator.func` members to `cocotb.simulator.find().func`, where the `find ()` function performs the currently failing indexing. That would mean a corresponding change to the gpi header.

ktbarrett, 2020-05-16:
I think it is sufficient to add a static flag to the GPI that is set when an interface is loaded and add a `gpi_simulator_is_active` function. All the functions in the simulator module could would check that and throw if not active, except for `get_sim_precision` whose old behavior was to return `-15` when there was no simulator.

themperek, 2020-05-16:
Maybe can use `cocotb.SIM_NAME` for now ?

tpoikela, 2020-05-16:
@themperek  Thanks for suggestion. `hasattr(cocotb, 'SIM_NAME')` seems to provide a workaround. Any potential pitfalls?

Nevertheless, it is better not to leak segfaults, as these cannot be caught inside Python code.

themperek, 2020-05-16:
We maybe should default `cocotb.SIM_NAME` to `None` in cocotb.

ktbarrett, 2020-05-18:
@themperek That is a good idea that will also help us with #1724.

commit 8e5a0e0d (2020-05-26): Fix #1839 Prevent segfault when using cocotb when no simulator is running

When a simulator is not running, meaning cocotb was not started by a
simulator, simulatormodule methods throw an exception.
`get_sim_precision` is an exception, it's previous behavior was to
return -15 when no simulator was present. This old behavior was
returned, it was previously broken.
closed 2020-05-26 by eric-wieser