cocotb/cocotb #1844: Handling errors in the GPI

ktbarrett, 2020-05-18:
The GPI has two customers, the simulator and the user (cocotb). It is necessary to provide information to the end user (you and me) and the other customer when important events occur while one customer is calling into the GPI.

When the simulator is calling into the GPI we should use the `LOG_*` macros and use `embed_sim_init` to feed critical information to cocotb. This is already done. 

However, when the user (cocotb) is calling into the GPI (e.g. `gpi_set_signal_value_int`) the most appropriate way should be to use `LOG_*` for warnings and below, while errors should feed back to the user like an exception (and would become an exception in the simulator module). I think the best way forward is to change these functions to return an optional string, `NULL` represents no error, and not `NULL` is a string with the error information.

For example,
https://github.com/cocotb/cocotb/blob/b54acce7e1879c49d18dd10ea4499b24a4a3b653/cocotb/share/lib/fli/FliObjHdl.cpp#L228-L247
Would become:
```c++
char const * FliEnumObjHdl::set_signal_value(const long value, const gpi_set_action_t action)
{
    if (action != GPI_DEPOSIT) {
       return "Force or release action not supported for FLI.";
    }

    if (value > m_num_enum || value < 0) {
       return format_str("Attempted to set a enum with range [0,%d] with invalid value %d!\n", m_num_enum, value).c_str();
    }

    if (m_is_var) {
        mti_SetVarValue(get_handle<mtiVariableIdT>(), value);
    } else {
        mti_SetSignalValue(get_handle<mtiSignalIdT>(), value);
    }

    return NULL;
}
```

eric-wieser, 2020-05-18:
Code of the form `return std::string(....).c_str()` results in a use-after-free, so that's a no go.

I'd recommend we make the C++ API throw exceptions, and then deal with converting to error codes at the C boundary.

ktbarrett, 2020-05-19:
At a minimum we should address all instances of `LOG_CRITICAL` used in functions called by cocotb.

ktbarrett, 2020-05-25:
From https://github.com/cocotb/cocotb/pull/1863#discussion_r429747349.

Logging is not a great way to signal an error to either cocotb or the user via the log. For one, the error might be able to be handled, it might end the test if unhandled, or it might be grounds to end the regression; there currently is no way to decide in cocotb. According to Eric, `log.error` messages are supposed to indicate test failure, and as mentioned, if you can handle them, a error level message in the log might not mean that. There is also the distinction between `LOG_ERROR` and `LOG_CRITICAL`. According to Eric, critical level logs imply regression end. The GPI is now making that decision about the severity for cocotb. Error logging has no place in user-called GPI code.

Logging is also inappropriate in simulator called code for much the same reasons as above. We have `gpi_embed_event` as a way to pass information to cocotb from simulator controlled GPI code, and we should leverage that. Depending on the error cocotb could kill the test, or the whole regression. The GPI shouldn't make this decision for cocotb. I'm not entirely sure what this should look like. So error logging has no place in simulator-called GPI code either.

I don't think we should get rid of either `LOG_ERROR` or `LOG_CRITICAL` on second thought, they might be useful to another GPI user. But they should be absent from the interface and GPI common code.

ktbarrett, 2020-06-05:
`LOG_CRITICAL` is no more, as of #1793. The last usages in simulator-called code have been changed to usages of `gpi_sim_end` which calls `gpi_embed_event` to signal sim end to users.