cocotb/cocotb #1847: TOPLEVEL_LANG is not documented

ktbarrett, 2020-05-20:
I can imagine this frustrates a lot of people.

cmarqu, 2020-05-23:
Wow, indeed. Strange that it took so long until someone noticed.

regis-c, 2020-06-04:
Minor comment.
Just notice TOPLEVEL_LANG=verilog (or something else ?) is needed playing with SIM=questa  

By the way if SIM=verilator is used, TOPLEVEL_LANG=verilog seems to be optional/not used.
(make sense as verilog is only possible target).

ktbarrett, 2020-06-04:
>Just notice TOPLEVEL_LANG=verilog (or something else ?) is needed playing with SIM=questa

What do you mean? Unless something changed recently it should default to verilog.

regis-c, 2020-06-04:
Sorry, I was not clear in my comment.

When lauching "make SIM=questa" without defining TOPLEVEL_LANG=verilog, I get following error message:
> *** "A valid value (verilog or vhdl) was not provided for TOPLEVEL_LANG=".  Stop.
Which sounds perfectly fine for me.

I was just a bit surprised that my other usecase with SIM=verilator of the same makefile is running fine without defining TOPLEVEL_LANG.
In this latest verilator usecase, the definition of TOPLEVEL_LANG looks optional/unused.

ktbarrett, 2020-06-05:
@regis-c I have no issues with that. If I leave it unspecified, it defaults to verilog. Can you run the regression with just `make SIM=questa`?

imphil, 2020-06-05:
@regis-c @ktbarrett see https://github.com/cocotb/cocotb/issues/1893. Simulators which support both VPI and VHPI typically need toplevel lang to be set.

regis-c, 2020-06-05:
@ktbarrett  
Hi give a try to "mean" example with a fresh install.
> anaconda pip install https://github.com/cocotb/cocotb/archive/master.zip --user
> git clone https://github.com/cocotb/cocotb.git cocotb

Then in mean/tests, running "make SIM=questa" pass the regression.  
Only issue is with the "$dumpvars(0, i_data[idx]);" from mean.sv that reports error: see below 
Looks like the array of port/signal are not well handled to dumpvar ?.  
I get a vcd but without i_data[] signals in it.

I am using this version of questa  
```
# //  Questa Sim-64
# //  Version 10.6c linux_x86_64 Jul 25 2017

```

```
#      0.00ns INFO     cocotb.mean                                test_mean.py:50   in value_test                      Detected DATA_WIDTH = 6, BUS_WIDTH = 4
# ** Error: (vsim-PLI-3537) $dumpvars() : Argument 2 is invalid.    : /users/rcz/workspace/cocotb_testing/cocotb/examples/mean/tests/../hdl/mean.sv(68)
#    Time: 0 ps  Iteration: 0  Instance: /mean
# ** Error: (vsim-PLI-3537) $dumpvars() : Argument 2 is invalid.    : /users/rcz/workspace/cocotb_testing/cocotb/examples/mean/tests/../hdl/mean.sv(68)
#    Time: 0 ps  Iteration: 0  Instance: /mean
# ** Error: (vsim-PLI-3537) $dumpvars() : Argument 2 is invalid.    : /users/rcz/workspace/cocotb_testing/cocotb/examples/mean/tests/../hdl/mean.sv(68)
#    Time: 0 ps  Iteration: 0  Instance: /mean
# ** Error: (vsim-PLI-3537) $dumpvars() : Argument 2 is invalid.    : /users/rcz/workspace/cocotb_testing/cocotb/examples/mean/tests/../hdl/mean.sv(68)
#    Time: 0 ps  Iteration: 0  Instance: /mean
#    350.00ns INFO     cocotb.regression                         regression.py:348  in _score_test                     Test Passed: mean_basic_test

```

Let me know if you want more checks from my side.

PR #1924 (2020-06-12): Document TOPLEVEL_LANG
Fixes #1847. Document `TOPLEVEL_LANG` in the makefile variable reference section of the documentation.
closed 2020-06-12 by eric-wieser