cocotb/cocotb #1851: Verilator issues

elgorwi, 2020-05-21:
I’ve stumbled upon some issues while trying to get the mean example running with verilator. 
One of the issues is also described in #1313 and happens when accessing arrays.
Here’s a patch that I’ve been working on to make the array writes work for verilator:
https://github.com/cocotb/cocotb/compare/master...elgorwi:mean_verilator

It looks like there could be multiple problems:

- I’ve noticed that the Verilator VPI error ```info.code``` can be null. Not sure if this could cause issues with the logging mechanism. 
https://github.com/elgorwi/cocotb/blob/017ba6d4d58d6b3399ffe28f958571664afca906/cocotb/share/lib/vpi/VpiImpl.h#L90

- The recursionError mentioned in #1313 appears when verilator raises a VPI error. Commenting out the ```check_vpi_error()``` call in ```VpiImpl::get_sim_time()``` gets rid of the recursionError. 
https://github.com/elgorwi/cocotb/blob/017ba6d4d58d6b3399ffe28f958571664afca906/cocotb/share/lib/vpi/VpiImpl.cpp#L65
Perhaps the VPI error does not get reset by Verilator after calling vpi_get_time()? Still need to look into this in more detail.  

- Verilator exposes the ‘i_data’ array as VPI type ```vpiMemory```, where Icarus uses ```vpiNetArray```. In addition, there seems to be a problem with following: https://github.com/elgorwi/cocotb/blob/017ba6d4d58d6b3399ffe28f958571664afca906/cocotb/share/lib/vpi/VpiCbHdl.cpp#L163 Not sure why we need this exactly, but it is also mentioned in #1834, where it causes a segfault with Active-HDL. I’ve added a special case for vpiMemory as a workaround, but I don’t think this is a permanent solution.

- There are still some other issues with accessing parameters/localparams, but this is due to Verilator not exposing them. The ‘mean_discovery_test’ fails for that reason. 
Another issue is that moving the discovery test up before the other tests in test_mean.py and running everything will cause subsequent tests to fail. So something is not getting cleaned up between tests.

eric-wieser, 2020-05-21:
> I’ve noticed that the Verilator VPI error info.code can be null. Not sure if this could cause issues with the logging mechanism.

Isn't that addressed here?

https://github.com/cocotb/cocotb/blob/d5d3f78e8cd39886719ec59c0d28f76a3e88bf63/cocotb/share/lib/vpi/VpiImpl.h#L47-L48

eric-wieser, 2020-05-21:
> The recursionError mentioned in #1313 appears when verilator raises a VPI error. a

The code has changed since that `RecursionError` was reported. Can you file a new issue with the full traceback?

elgorwi, 2020-06-13:
All of the relevant issues above have been fixed on the Verilator side.
So best to upgrade to the latest Verilator version from git (v4.037 devel) and compile/install manually or alternatively wait until the next v4.038) Verilator package release.

closed 2020-06-26 by elgorwi