cocotb/cocotb #1854: [Riviera-PRO] test_iteration_vhdl fails with "Not able to map type ((null)) 0 to object"

imphil, 2020-05-23:
Riviera-Pro 2019.10 crashes when running the `test_iteration_vhdl` test.

```
# COUT:      0.00ns INFO     Starting test: "recursive_discovery"
# COUT:                      Description:
# COUT:                          Recursively discover every single object in the design
# COUT:
# COUT:      0.10ns WARNING  VHPI: Implementation does not know how to iterate over vhpiProcessStmtK(1082)
# COUT:      0.10ns WARNING  VHPI: Implementation does not know how to iterate over vhpiProcessStmtK(1082)
# COUT:      0.10ns ERROR    Not able to map type ((null)) 0 to object
# COUT:      0.10ns WARNING  Unable to create gen_generic_sp_ram via any registered implementation
# COUT: Stack pointer: 0x7f2e714a4b30
# COUT: Local variable address: 0x7f2e714a4f4f
# KERNEL: Fatal Error: System exception occurred: SIGSEGV. Cannot continue.
```

[Full log](https://github.com/cocotb/cocotb/files/4671974/log.txt)

I didn't debug this further yet.

garmin-mjames, 2020-05-28:
Looks like this is my fault in #1164 (https://github.com/cocotb/cocotb/commit/c2c808a9dec0ee7339a35935cec644e08382bbe7):

https://github.com/cocotb/cocotb/blob/924f35a3b7d39543118b7bfaed77dd4808e6612b/cocotb/share/lib/vhpi/VhpiCbHdl.cpp#L49

What I didn't understand at the time is that pseudo-regions are constructs that don't represent an actual object, they are just useful for encapsulating things like `for generate` statements.

The handle that is released is a handle to an actual region object, and so there's a use-after-release error. I'll submit a PR to fix it.

It will be nice to have a (mostly) working VHDL simulator in the CI pipeline for things like this.

imphil, 2020-05-28:
> It will be nice to have a (mostly) working VHDL simulator in the CI pipeline for things like this.

@garmin-mjames the Private CI we have under new PRs is running Riviera-Pro 2019.10.

If you have a (proposed) patch but no access to a VHDL simulator please open a PR. I'll be able to help you out getting access to potential error messages.

garmin-mjames, 2020-05-28:
I have access to Aldec Active-HDL. I saw the same problem and verified the fix.

closed 2020-06-18 by ktbarrett