cocotb/cocotb #1855: Implement Force and Release for the FLI

ktbarrett, 2020-05-23:
This was skipped in #1403 because the author did not have access to Questa. This would use the `mti_ForceSignal` function for all signal sets because it supports DEPOSIT and FORCE. The `mti_ReleaseSignal` function would be used to support release. Interestingly, `mti_ForceSignal` also supports DRIVE, so it may be possible in the future to support *driving* signals rather than using force/release. AFAICT the other interfaces can not support this.

garmin-mjames, 2020-05-26:
Regarding driving signals, *VHPI* allows for creating drivers for signals with `vhpi_create`.

ktbarrett, 2020-05-26:
Driving should probably be a separate issue.

ktbarrett, 2022-02-26:
Closed by #2775.

closed 2022-02-26 by ktbarrett

Terrathaw, 2022-07-15:
I'm using the latest version (1.6.2) with questa and receive the error "200.00ns ERROR    Force or release action not supported for FLI." when trying to force or freeze a signal.
As far as I understand this feature should be available for questa since the pull request #2775 so it seems that I'm overlooking something else. Does anyone have an idea what I might be doing wrong?

ktbarrett, 2022-07-15:
It's been merged, but not released. It will be a part of the 1.7 release. In the mean time you can install the development version of the library, instructions for which are listed in the installation documentation.