cocotb/cocotb #1856: Running `make clean` without Icarus installed fails

ktbarrett, 2020-05-23:
For some reason the clean target depends on a simulator being set and if you invoke `make clean` and don't have icarus installed (`SIM` defaults to icarus) you get:

```
> make clean
.../site-packages/cocotb/share/makefiles/simulators/Makefile.icarus:51: *** "Unable to locate command >iverilog<".  Stop.
```

ktbarrett, 2020-05-25:
Looks like cleanup behavior depends on the type of simulator. I should probably just be using `git clean -xdf` instead.

imphil, 2020-05-31:
I just got annoyed by the same issue when repeatedly switching between simulators. We should be able to provide a `make clean` which calls a `clean`-like target on all available simulators.

An easy implementation is probably using recursive Makefile calls, but there might be room for a better implementation.

ktbarrett, 2020-06-01:
Should we be trying to ensure that all sim artifacts end up in `SIM_BUILD`? Then we can just rm -rf that.

themperek, 2020-06-01:
> Should we be trying to ensure that all sim artifacts end up in `SIM_BUILD`? Then we can just rm -rf that. No need for simulator specific cleanup.

Here somehow was decided to run in current directory https://github.com/cocotb/cocotb/pull/1063 although running from `sim_build` would avoid artifacts.

Raghupal123, 2022-09-03:
Hi ,

I am very new to cocotb and trying to run simple example in conda prompt and seeing following error when running make
simulators/Makefile.icarus:51: *** "Unable to locate command >iverilog<".  Stop

could you please let me know how to resolve this

imphil, 2022-09-03:
@Raghupal123 thanks for your interest in cocotb. You asked the same question in the Gitter channel, let's address it there.

In general (no harm done!), please avoid "re-using" an existing issue unless you're very sure you have the same problem than originally described. It's easier to merge issues than to split them apart :-)

Raghupal123, 2022-09-03:
> @Raghupal123 thanks for your interest in cocotb. You asked the same question in the Gitter channel, let's address it there.
>
> In general (no harm done!), please avoid "re-using" an existing issue [...]

@imphil thank you for suggestion. Sure in future will not split. Struggling with this for more than a day, out of desperation looking for answer everywhere :) sorry about that

imphil, 2022-09-03:
Feel free to open an issue at any time when you feel something is wrong in cocotb, or start a [discussion](https://github.com/cocotb/cocotb/discussions) for any kind of questions. We're happy to help!

Shangkorong, 2022-11-16:
@imphil @Raghupal123 can you please share the link of the GItter channel, I cant find it. I am also having the same problem, "Makefile.icarus:53: *** Unable to locate command >iverilog<.  Stop." when running make in anaconda prompt. BTW is the problem solved?

imphil, 2022-11-17:
@Shangkorong your issue is unrelated to the feature request here. Your problem is a non-working installation of Icarus Verilog, or alternatively you need to specify `SIM=...` to use another simulator. The cocotb gitter channel is at https://gitter.im/cocotb/Lobby, please ask around there or file a new issue -- but let's keep this issue focused on the actual feature request.

PR #4505 (2025-02-23): Makefile cleanup
Closes #2361. Closes #1856.

- [x] newfragments
closed 2025-02-23 by ktbarrett