cocotb/cocotb #1859: Riviera: Test issue_1279 fails with  "Cleanup() called outside of the main thread"

imphil, 2020-05-23:
Riviera: Test issue_1279 fails with  "Cleanup() called outside of the main thread"

Riviera-PRO 2019.10

```
philipp@philipp-VirtualBox:~/src/cocotb$ make SIM=riviera COCOTB_REDUCED_LOG_FMT=1 -C tests/test_cases/issue_1279/ clean all
...
# COUT:      0.00ns INFO     Running on Riviera-PRO version 2019.10.138.7537
# COUT:      0.00ns INFO     Running tests with cocotb v1.4.0.dev0 from /home/philipp/.local/lib/python3.6/site-packages/cocotb
# COUT:      0.00ns INFO     Seeding Python random module with 1590270879
# COUT:      0.00ns INFO     Found test issue_1279.test_sim_failure_a
# COUT:      0.00ns INFO     Found test issue_1279.test_sim_failure_b
# COUT:      0.00ns INFO     Running test 1/2: test_sim_failure_a
# COUT:      0.00ns INFO     Starting test: "test_sim_failure_a"
# COUT:                      Description: None
# KERNEL: Simulation has finished. There are no more test vectors to simulate.
endsim
# COUT:      0.00ns ERROR    Failing test at simulator request before test run completion: Simulator shutdown prematurely
# COUT: Traceback (most recent call last):
# COUT:   File "/home/philipp/.local/lib/python3.6/site-packages/cocotb/__init__.py", line 236, in _sim_event
# COUT:     scheduler.finish_scheduler(SimFailure(msg))
# COUT:   File "/home/philipp/.local/lib/python3.6/site-packages/cocotb/scheduler.py", line 826, in finish_scheduler
# COUT:     self._test.abort(exc)
# COUT:   File "/home/philipp/.local/lib/python3.6/site-packages/cocotb/decorators.py", line 281, in abort
# COUT:     cocotb.scheduler.unschedule(self)
# COUT:   File "/home/philipp/.local/lib/python3.6/site-packages/cocotb/scheduler.py", line 493, in unschedule
# COUT:     self.cleanup()
# COUT:   File "/home/philipp/.local/lib/python3.6/site-packages/cocotb/scheduler.py", line 846, in cleanup
# COUT:     raise Exception("Cleanup() called outside of the main thread")
# COUT: Exception: Cleanup() called outside of the main thread
# COUT:      0.00ns ERROR    Passing event to upper layer failed
# VSIM: Simulation has finished.
ERROR: results.xml was not written by the simulation!
/home/philipp/.local/lib/python3.6/site-packages/cocotb/share/makefiles/simulators/Makefile.riviera:146: recipe for target 'results.xml' failed
make[1]: *** [results.xml] Error 1
make[1]: Leaving directory '/home/philipp/src/cocotb/tests/test_cases/issue_1279'
/home/philipp/.local/lib/python3.6/site-packages/cocotb/share/makefiles/Makefile.inc:39: recipe for target 'sim' failed
make: *** [sim] Error 2
make: Leaving directory '/home/philipp/src/cocotb/tests/test_cases/issue_1279'
```

[full log](https://github.com/cocotb/cocotb/files/4672619/log.txt) (ignore the apport warning, it's Ubuntu 18.04-injected)

ktbarrett, 2020-05-23:
Likely related to #1793.

imphil, 2020-05-25:
Unfortunately, simply downgrading the Exception to an error won't do the trick quite yet. With a workaround in https://github.com/imphil/cocotb/tree/workaround-riviera-exit

I get a failing test `test_closedown.test_failure_from_system_task`:

```
# COUT:      0.00ns INFO     Starting test: "test_failure_from_system_task"
# COUT:                      Description: 
# COUT:                          Allow the dut to call system tasks from verilog.
# COUT:                          $fatal() will fail the test, and scheduler will cleanup forked coroutines.
# COUT:                          
# KERNEL: Info: /home/philipp/src/cocotb/tests/designs/close_module/close_module.v (58): 
# KERNEL: Info: /home/philipp/src/cocotb/tests/designs/close_module/close_module.v (59): This is an info message
# KERNEL: Warning: /home/philipp/src/cocotb/tests/designs/close_module/close_module.v (60): 
# KERNEL: Warning: /home/philipp/src/cocotb/tests/designs/close_module/close_module.v (61): This is a  warning message
# KERNEL: Error: /home/philipp/src/cocotb/tests/designs/close_module/close_module.v (62): 
# KERNEL: Error: /home/philipp/src/cocotb/tests/designs/close_module/close_module.v (63): This is an error message
# KERNEL: Fatal Error: /home/philipp/src/cocotb/tests/designs/close_module/close_module.v (64): This is a a fatal message that fails the test
# RUNTIME: Info: RUNTIME_0068 close_module.v (64): $finish called.
# KERNEL: Time: 1060 ps,  Iteration: 0,  Instance: /close_module,  Process: @INITIAL#55_3@.
# KERNEL: stopped at time: 1060 ps
# VSIM: Simulation has finished. There are no more test vectors to simulate.
endsim
# COUT:      1.06ns ERROR    Failing test at simulator request before test run completion: Simulator shutdown prematurely
# COUT:      1.06ns ERROR    cleanup() called from thread Dummy-1 (140585757644544) instead of the main thread (140585675855616). Ignoring for Aldec simulators, see issue #1859
# COUT:      1.06ns ERROR    Test error has lead to simulator shutting us down
# COUT:                      cocotb.result.SimFailure: Failing test at simulator request before test run completion: Simulator shutdown prematurely
# COUT:      1.06ns ERROR    Test error has lead to simulator shutting us down
# COUT:                      cocotb.result.SimFailure
# COUT:      1.06ns ERROR    Failed 2 out of 2 tests (0 skipped)
# COUT:      1.06ns INFO     ******************************************************************************************************
# COUT:                      ** TEST                                          PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
# COUT:                      ******************************************************************************************************
# COUT:                      ** test_closedown.test_failure_from_system_task    FAIL            1.06          0.42         2.53  **
# COUT:                      ** test_closedown.test_after_system_task_fail      FAIL            0.00          0.00          nan  **
# COUT:                      ******************************************************************************************************
```

I guess we'll need to have a closer look at the `test_failure_from_system_task` test as well, since we already ignore that one for Questa, and double-check if it is really testing the right thing.

garmin-mjames, 2020-05-26:
Could the check in `cleanup` be reversed, and instead of checking that the current thread **is** `_main_thread`, check that it **is not** one of `_pending_threads`?

I think the check is about making sure only the main cocotb thread is shutting things down, not one of the threads spun off by a test.

garmin-mjames, 2020-05-26:
I noticed the same issue with Active-HDL. If you look at your output from `test_closedown`, it's printing the error messages from the simulator runtime, not cocotb.

I have a feeling that overloading the system tasks is not working in VSimSA. The built-in `$fatal` system task calls `$finish` (from Active-HDL 11.1 docs):

[image]

ktbarrett, 2020-07-28:
`issue_1279` test also fails on verilator which interestingly enters an infinite loop and hangs forever.