cocotb/cocotb #1865: Constant string handle __str__ returns garbage value

ktbarrett, 2020-05-24:
If a constant is a string, it gets the value with `get_signal_value_str` which returns a `bytes` object. The `__str__` implementation simply casts the underlying value using `str()` and returns that. Casting a bytes object to a string results in a garbage value: `"b'handle value'"`. This needs to be better thought out. Perhaps constant objects need to be split out into different type classes like `ModifiableObjects` are.

eric-wieser, 2020-05-24:
Implementing `__bytes__` instead of `__str__` would fix this, right?

ktbarrett, 2020-05-24:
Yeah, but that still leaves the existing surprising behavior in the `str` cast. It's easy enough to just use `handle.value` and deal with type conversion yourself. This kind of ties into #1222 where using the built in casts operators directly on handles results in surprising behavior.

eric-wieser, 2020-05-24:
I agree with the sentiment of #1222, but I think we need to add `__bytes__` and remove `__str__` to be consistent with the general change from str to bytes.

eric-wieser, 2020-05-24:
That is, I'd consider this a regression we need to fix for 1.4, _in addition_ to a design flaw we need to address at some point.

ktbarrett, 2020-05-25:
Note this also affects `StringObject`s.

ktbarrett, 2020-05-25:
We also can't remove `__str__` and supplant it with `__bytes__` due to the poor design. That would affect existing string casts on constant logic vectors (`BinaryValue` doesn't implement `__bytes__`) and constant integers and reals (`__str__` is well defined, `__bytes__` is not).

eric-wieser, 2020-05-31:
Isn't this equally a problem for _non_-constant objects?

ktbarrett, 2020-05-31:
Yes, I mentioned it affect `StringObject`s as well, but they can simply have `__str__` -> `__bytes__` without problem, but `ConstantObject` is hard because it services all const types.

I'd really like to see the whole thing refactored so that `SimHandle` can have custom type registered to use a specific `ModifiableObject` subtype, then we could register provided subtypes for strings, reals, logic vector, etc by default. Constness would only be checked/raise when setting the handle. This would require we add the ability to obtain the type of a signal as a string to the GPI so we could support types other than the basics. But that's 2.0 work.

PR #1890 (2020-06-18): Deprecate implicit conversion of handle values to strings.
Note that most visibly, this will make code like the following emit warnings:
```python
log.info("The value of X is %s", dut.X)
```
unless users change it to
```python
log.info("The value of X is %s", dut.X.value)
```
Users who don't make this change will eventually see `the value of X is dut.X` in their logs.

This is a candidate fix for #1865. If we don't want to break user code doing the above, then perhaps we should close #1865 as "working as intended".
closed 2020-06-18 by eric-wieser