cocotb/cocotb #1871: RFC: Tracking integration issues

imphil, 2020-05-27:
I'm proposing to create a more structured way to track issues which are out of our control, but are affecting cocotb. That's mostly simulator issues, but could also be operating system or Python version issues. Read on for the full proposal.

cocotb supports a fair number of simulators in different versions. Even though those simulators are abstracted through VPI/VHPI/FLI, we often run into issues where simulators implement the specifications differently, where we need to work around bugs or unimplemented features, etc.

We see such things in our code base in various ways:
- Adjusted or skipped tests.
- Workarounds in various parts of the code.

Our users do not have a simple way to know which features are known to be broken or unsupported in various simulators. They can search through issues here, sometimes they get a warning output from our code (e.g. "feature unsupported"), or they simply need to try it out.

I'm proposing to create a tracking database for such issues. It should track the following pieces of information in a reasonably structured way:

- Issue number
- Summary
- How the problem affects our users
- Analysis of the problem
- Workaround (if available)
- Affected cocotb version(s)
- Affected tool
- Affected version(s)
- Affected operating systems/distributions
- Upstream status (reported, fixed, etc.)
- Upstream ticket/issue number (if possible)
- Upstream/vendor response (if available)

This information can then be used in all places where we currently need to work around typically simulator issues, especially in tests. And it should be used wherever we inform the user about such problems, either in messages output from the program, or in the documentation.

On implementation:
The probably easiest implementation is a issue template for Github which contains the necessary information. Through the use of tags we can make these issues easier to query. Even though that doesn't give us "fully structured" information, it should be good enough for our purposes. The GitHub API allows us to easily include such information with a dynamic query into HTML pages, etc.

I guess the main question on this approach would be a rather bikeshedding-style one: should we add this information to this repository (cocotb/cocotb), or create a new one especially for this purpose? If we use this repository, we will be creating a reasonable amount of issues which won't be closed any time soon, but are also not actionable. I don't think that's necessarily a bad thing, it's probably a matter of taste.

cmarqu, 2020-05-28:
I haven't read a license agreement in the last years, but it used to be that commercial simulator vendors didn't like to be tracked/compared publicly.

imphil, 2020-05-28:
I'm not looking at a comparison or any data which wouldn't be entered into this issue tracker or used as a comment in a source code anyways. This is not blaming anyone: software has bugs, missing features, or just plain different design points, and users need to be aware of it when using cocotb.

We also don't need to be concerned about the licensing terms for end users. We (as cocotb project) have not signed them, it's our users who report integration issues and need to take care to only share things they are allowed to share.

garmin-mjames, 2020-06-18:
I like the idea of having a cleaner separation by putting them in another repo, but it may cause maintenance headaches as well as make it harder to discover for someone not familiar with the project.

ktbarrett, 2020-06-18:
I agree that it should be outside of this repo. It can be on us to diagnose the bug, and if it applies we would transfer the issue to the other repo. If users would be interested in seeing this issue tracker, then we should mention it's existence in the README and the documentation.

I also agree that Github issues and labels should be sufficient to capture these issues.

eric-wieser, 2020-06-18:
Note that it's possible for the new issue button to present a splash page with links to other pages - which could include the other repo.

garmin-mjames, 2020-12-10:
We could start this by creating issue labels for this repo, then later [transferring](https://docs.github.com/en/free-pro-team@latest/github/managing-your-work-on-github/transferring-an-issue-to-another-repository) the issues to the new repo.