cocotb/cocotb #1875: Syntax error when trying out example "simple_dff"

phdbreak99, 2020-05-28:
The log is like this
```
jw@Mac-mini: /tmp/cocotb/examples/simple_dff (master) ✔                                                                                                                               [2/54511]
➜  make
/Library/Developer/CommandLineTools/usr/bin/make results.xml
PYTHONPATH=/private/tmp/cocotb/examples/simple_dff/build/libs/x86_64:/tmp/cocotb/examples/simple_dff:/Library/Python/3.7/site-packages: LD_LIBRARY_PATH=/private/tmp/cocotb/examples/simple_dff
/build/libs/x86_64::/System/Library/Frameworks/Python.framework/Versions/2.7/lib:/System/Library/Frameworks/Python.framework/Versions/2.7/lib MODULE=test_dff \
        TESTCASE= TOPLEVEL=dff TOPLEVEL_LANG=verilog COCOTB_SIM=1 \
        /usr/local/bin/vvp -M /private/tmp/cocotb/examples/simple_dff/build/libs/x86_64 -m gpivpi sim_build/sim.vvp
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interp
reter
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:104  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:340  in embed_sim_init                  Running on Icarus Verilog version 10.3 (stable)
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:341  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:138  in _initialise_testbench           Running tests with cocotb v1.3.1 from /Library/Python/3.7/site-packages
     0.00ns INFO     cocotb                                      __init__.py:155  in _initialise_testbench           Seeding Python random module with 1590644493
     0.00ns CRITICAL cocotb.regression                         regression.py:137  in initialise                      Failed to import module test_dff: invalid syntax (test_dff.py, line 11)
     0.00ns INFO     cocotb.regression                         regression.py:138  in initialise                      MODULE variable was "test_dff"
     0.00ns INFO     cocotb.regression                         regression.py:139  in initialise                      Traceback:
     0.00ns INFO     cocotb.regression                         regression.py:140  in initialise                      Traceback (most recent call last):
                                                                                                                       File "/Library/Python/3.7/site-packages/cocotb/regression.py", line 135,
 in initialise
                                                                                                                         module = _my_import(module_name)
                                                                                                                       File "/Library/Python/3.7/site-packages/cocotb/regression.py", line 63,
in _my_import
                                                                                                                         mod = __import__(name)
                                                                                                                       File "/tmp/cocotb/examples/simple_dff/test_dff.py", line 11
                                                                                                                         async def test_dff_simple(dut):
                                                                                                                             ^
                                                                                                                     SyntaxError: invalid syntax

Traceback (most recent call last):
  File "/Library/Python/3.7/site-packages/cocotb/__init__.py", line 175, in _initialise_testbench
    regression_manager.initialise()
  File "/Library/Python/3.7/site-packages/cocotb/regression.py", line 135, in initialise
    module = _my_import(module_name)
  File "/Library/Python/3.7/site-packages/cocotb/regression.py", line 63, in _my_import
    mod = __import__(name)
  File "/tmp/cocotb/examples/simple_dff/test_dff.py", line 11
    async def test_dff_simple(dut):
        ^
SyntaxError: invalid syntax
     0.00ns ERROR    cocotb.gpi                                  gpi_embed.c:425  in embed_sim_init                  cocotb initialization failed - exiting
     0.00ns ERROR    cocotb.scheduler                            __init__.py:198  in _sim_event                      Failing test at simulator request before test run completion: Simulator sh
utdown prematurely
```

My environment:
OSX
Python = 3.7.1
Cocotb = 1.3.1
icarus-verilog = 10.3'

The example "adder" works fine

Thank you!
Jim

themperek, 2020-05-28:
The problem may be that this example is for cocotb version 1.4 (not released yet).

eric-wieser, 2020-05-28:
Your issue is that you're (somehow) using python 2.7, which is no longer supported in master.

phdbreak99, 2020-05-28:
Thanks @eric-wieser and @themperek 

I think the reason that `/System/Library/Frameworks/Python.framework/Versions/2.7/lib` is used instead of 3.7 is because my current OSX (10.15.4 Catalina) still has python (python2) and python3. And in the makefile `cocotb/share/makefiles/Makefile.pylib.Darwin` is using `PYTHON_BIN?=python`

So I added this line `PYTHON_BIN = python3` in the makefile to force using `python3`

The error is the same, but the 3.7 lib from Xcode.app is used in `LD_LIBRARY_PATH`

```
/Library/Developer/CommandLineTools/usr/bin/make results.xml
PYTHONPATH=/private/tmp/cocotb/examples/simple_dff/build/libs/x86_64:/tmp/cocotb/examples/simple_dff:/Library/Python/3.7/site-packages: LD_LIBRARY_PATH=/private/tmp/cocotb/examples/simple_dff
/build/libs/x86_64::/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib:/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.fram
ework/Versions/3.7/lib MODULE=test_dff \
        TESTCASE= TOPLEVEL=dff TOPLEVEL_LANG=verilog COCOTB_SIM=1 \
        /usr/local/bin/vvp -M /private/tmp/cocotb/examples/simple_dff/build/libs/x86_64 -m gpivpi sim_build/sim.vvp
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interp
reter
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:104  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:340  in embed_sim_init                  Running on Icarus Verilog version 10.3 (stable)
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:341  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:138  in _initialise_testbench           Running tests with cocotb v1.3.1 from /Library/Python/3.7/site-packages
     0.00ns INFO     cocotb                                      __init__.py:155  in _initialise_testbench           Seeding Python random module with 1590690310
     0.00ns CRITICAL cocotb.regression                         regression.py:137  in initialise                      Failed to import module test_dff: invalid syntax (test_dff.py, line 11)
     0.00ns INFO     cocotb.regression                         regression.py:138  in initialise                      MODULE variable was "test_dff"
     0.00ns INFO     cocotb.regression                         regression.py:139  in initialise                      Traceback:
     0.00ns INFO     cocotb.regression                         regression.py:140  in initialise                      Traceback (most recent call last):
                                                                                                                       File "/Library/Python/3.7/site-packages/cocotb/regression.py", line 135,
 in initialise
                                                                                                                         module = _my_import(module_name)
                                                                                                                       File "/Library/Python/3.7/site-packages/cocotb/regression.py", line 63,
in _my_import
                                                                                                                         mod = __import__(name)
                                                                                                                       File "/tmp/cocotb/examples/simple_dff/test_dff.py", line 11
                                                                                                                         async def test_dff_simple(dut):
                                                                                                                             ^
                                                                                                                     SyntaxError: invalid syntax

Traceback (most recent call last):
  File "/Library/Python/3.7/site-packages/cocotb/__init__.py", line 175, in _initialise_testbench
    regression_manager.initialise()
  File "/Library/Python/3.7/site-packages/cocotb/regression.py", line 135, in initialise
    module = _my_import(module_name)
  File "/Library/Python/3.7/site-packages/cocotb/regression.py", line 63, in _my_import
    mod = __import__(name)
  File "/tmp/cocotb/examples/simple_dff/test_dff.py", line 11
    async def test_dff_simple(dut):
        ^
SyntaxError: invalid syntax
```

Do you have any other suggestions?

themperek, 2020-05-28:
Can you try development version or you have need for 1.3.1?

phdbreak99, 2020-05-29:
No, I don't have to use 1.3.1. Just started to try out cocotb, i think it's a very promising project. So instead of giving up 1.3.1, I posted the issue I ran into so you can improve it. Thanks.

themperek, 2020-05-29:
The installation process has significantly changed since 1.3.1 because of many issues like this and would be good to know if it is ok now (1.4).

phdbreak99, 2020-06-01:
Thanks @themperek for your patience. I've tried to install 1.4 by following the doc
```
➜  cocotb-config --version
1.4.0.dev0
```
However, I continue to got the following error.
```
➜  make SIM=verilator PYTHON_BIN=python3
/Library/Developer/CommandLineTools/usr/bin/make results.xml
MODULE=test_dff TESTCASE= TOPLEVEL=dff TOPLEVEL_LANG=verilog \
        sim_build/dff
dyld: Library not loaded: @rpath/Python3.framework/Versions/3.7/Python3
  Referenced from: /Library/Python/3.7/site-packages/cocotb/libs/libcocotb.so
  Reason: image not found
make[1]: *** [results.xml] Abort trap: 6
make: *** [sim] Error 2
```
Is it because the compiled version directly reference the `xcode` version of Python3? Then I installed XCode (took a long time and space), the error persist.

PS. I've successfully brought up Cocotb with Ubuntu 16.04 in docker.

themperek, 2020-06-01:
Maybe multiple version of python is an issue. I have never used mac. Maybe @mtikekar has some idea?

mtikekar, 2020-06-01:
It looks like the `xcode` version of Python3 has this funny `install_name` for the library: `@rpath/Python3.framework/Versions/3.7/Python3`. 

I have an older MacOS which only comes with python 2.7. I'm using conda python to get python3. For now, I will recommend using conda instead of the system/xcode/homebrew versions of python. We should try to fix this though.

phdbreak99, 2020-06-02:
Thanks @mtikekar , do you know which env var is used to choose Python location? Because when I don't have XCode, it's still trying to point to XCode location like `@rpath/Python3.framework/Versions/3.7/Python3`. That's the reason I installed XCode.

BTW, myself directly use Python3 come with Homebrew.

I've tried `LD_RUNPATH_SEARCH_PATH` which doesn't work. Thank you!

mtikekar, 2020-06-02:
Interesting, Homebrew python should work - cocotb's CI setup uses it. The following works for me:

```console
$ which python3
/usr/local/bin/python3
$ which pip3
/usr/local/bin/pip3
$ git clone https://github.com/cocotb/cocotb
$ cd cocotb
$ pip3 install .
$ cocotb-config --prefix
/usr/local/lib/python3.7/site-packages
$ cd examples/simple_dff
$ make SIM=verilator
```

Before installing, I checked that python and pip are from Homebrew (in `/usr/local/bin`) and after installing, I checked that it indeed got installed in Homebrew.

In your case, the path `/Library/Python/3.7/site-packages/cocotb/libs/libcocotb.so` suggests that it was installed in the Xcode python environment.

phdbreak99, 2020-06-03:
Thanks! @mtikekar I've solved it.
Previously I was using either the Python3 come with Homebrew or the one in XCode.app. So i tried to `brew install python3` and use that one, I've successfully compiled the example on OSX.

closed 2020-06-03 by phdbreak99