cocotb/cocotb #1880: Riviera-PRO fails dff example in VHDL

themperek, 2020-05-28:
Log:

```
2020-05-28T14:44:48.8229145Z # COUT:      -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:97   in set_program_name_in_venv        Using Python virtual environment interpreter at /home/az_devops_agent/agent/_work/1/s/.tox/py3/bin/python
2020-05-28T14:44:48.8556067Z # COUT:      -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:105  in gpi_print_registered_impl       VHPI registered
2020-05-28T14:44:49.1362339Z # COUT:      -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:245  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
2020-05-28T14:44:49.1371028Z # COUT:      0.00ns INFO     cocotb                                      __init__.py:189  in _initialise_testbench           Running on Riviera-PRO version 2019.10.138.7537
2020-05-28T14:44:49.1375745Z # COUT:      0.00ns INFO     cocotb                                      __init__.py:196  in _initialise_testbench           Running tests with cocotb v1.4.0.dev0 from /home/az_devops_agent/agent/_work/1/s/.tox/py3/lib/python3.6/site-packages/cocotb
2020-05-28T14:44:49.1379518Z # COUT:      0.00ns INFO     cocotb                                      __init__.py:216  in _initialise_testbench           Seeding Python random module with 1590677089
2020-05-28T14:44:49.1633126Z # COUT:      0.00ns INFO     cocotb.regression                         regression.py:125  in __init__                        Found test dff_cocotb.run_test_001
2020-05-28T14:44:49.1638835Z # COUT:      0.00ns INFO     cocotb.regression                         regression.py:457  in _start_test                     Running test 1/1: run_test_001
2020-05-28T14:44:49.1644720Z # COUT:      0.00ns INFO     cocotb.test._my_test.0x7f6bc8f40898       decorators.py:256  in _advance                        Starting test: "run_test_001"
2020-05-28T14:44:49.1647439Z # COUT:                                                                                                                      Description: Automatically generated test
2020-05-28T14:44:49.1650512Z # COUT:                                                                                                                      
2020-05-28T14:44:49.1652929Z # COUT:                                                                                                                      
2020-05-28T14:44:49.1663327Z # COUT:      0.00ns INFO     cocotb.scoreboard.dff                     scoreboard.py:216  in add_interface                   Created with reorder_depth 0
2020-05-28T14:44:49.8362647Z # COUT:  55000.00ns ERROR    cocotb.scoreboard.dff.output              scoreboard.py:141  in compare                         Received transaction differed from expected output
2020-05-28T14:44:49.8368040Z # COUT:  55000.00ns INFO     cocotb.scoreboard.dff.output              scoreboard.py:145  in compare                         Expected:
2020-05-28T14:44:49.8371452Z # COUT:                                                                                                                      1
2020-05-28T14:44:49.8373504Z # COUT:  55000.00ns INFO     cocotb.scoreboard.dff.output              scoreboard.py:149  in compare                         1
2020-05-28T14:44:49.8376601Z # COUT:  55000.00ns INFO     cocotb.scoreboard.dff.output              scoreboard.py:155  in compare                         Received:
2020-05-28T14:44:49.8379249Z # COUT:                                                                                                                      0
2020-05-28T14:44:49.8382978Z # COUT:  55000.00ns INFO     cocotb.scoreboard.dff.output              scoreboard.py:159  in compare                         0
2020-05-28T14:44:49.8386382Z # COUT: /home/az_devops_agent/agent/_work/1/s/.tox/py3/lib/python3.6/site-packages/cocotb/scoreboard.py:162: DeprecationWarning: Passing strings to hexdiffs is deprecated, pass bytes instead
2020-05-28T14:44:49.8389748Z # COUT:   log.warning("Difference:\n%s" % hexdiffs(strexp, strgot))
2020-05-28T14:44:49.8393346Z # COUT:  55000.00ns WARNING  cocotb.scoreboard.dff.output              scoreboard.py:162  in compare                         Difference:
2020-05-28T14:44:49.8395238Z # COUT:                                                                                                                      0000      31                                               1
2020-05-28T14:44:49.8397152Z # COUT:                                                                                                                           0000 30                                               0
2020-05-28T14:44:49.8398811Z # COUT:                                                                                                                      
2020-05-28T14:44:49.8401003Z # COUT:  55000.00ns INFO     ..itor._monitor_recv.0x7f6bc8f519b0        scheduler.py:503  in unschedule                      Test stopped by this forked coroutine
2020-05-28T14:44:49.8589418Z # COUT:  55000.00ns ERROR    cocotb.regression                         regression.py:390  in _score_test                     Test Failed: run_test_001 (result was TestFailure)
2020-05-28T14:44:49.8597353Z # COUT:                                                                                                                      Traceback (most recent call last):
2020-05-28T14:44:49.8599924Z # COUT:                                                                                                                        File "/home/az_devops_agent/agent/_work/1/s/examples/dff/tests/dff_cocotb.py", line 64, in _monitor_recv
2020-05-28T14:44:49.8602331Z # COUT:                                                                                                                          self._recv(vec)
2020-05-28T14:44:49.8606123Z # COUT:                                                                                                                        File "/home/az_devops_agent/agent/_work/1/s/.tox/py3/lib/python3.6/site-packages/cocotb/monitors/__init__.py", line 154, in _recv
2020-05-28T14:44:49.8608375Z # COUT:                                                                                                                          callback(transaction)
2020-05-28T14:44:49.8611681Z # COUT:                                                                                                                        File "/home/az_devops_agent/agent/_work/1/s/.tox/py3/lib/python3.6/site-packages/cocotb/scoreboard.py", line 249, in check_received_transaction
2020-05-28T14:44:49.8614300Z # COUT:                                                                                                                          self.compare(transaction, exp, log, strict_type=strict_type)
2020-05-28T14:44:49.8618111Z # COUT:                                                                                                                        File "/home/az_devops_agent/agent/_work/1/s/.tox/py3/lib/python3.6/site-packages/cocotb/scoreboard.py", line 164, in compare
2020-05-28T14:44:49.8620771Z # COUT:                                                                                                                          raise TestFailure("Received transaction differed from expected"
2020-05-28T14:44:49.8623603Z # COUT:                                                                                                                      cocotb.result.TestFailure: Received transaction differed from expectedtransaction
2020-05-28T14:44:49.8626879Z # COUT:  55000.00ns ERROR    cocotb.regression                         regression.py:470  in _log_test_summary               Failed 1 out of 1 tests (0 skipped)
2020-05-28T14:44:49.8629043Z # COUT:  55000.00ns INFO     cocotb.regression                         regression.py:542  in _log_test_summary               *********************************************************************************
2020-05-28T14:44:49.8631278Z # COUT:                                                                                                                      ** TEST                     PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
2020-05-28T14:44:49.8635286Z # COUT:                                                                                                                      *********************************************************************************
[... cut ...]
2020-05-28T14:44:49.8668249Z # COUT:  55000.00ns INFO     cocotb.regression                         regression.py:253  in tear_down                       Shutting down...
2020-05-28T14:44:49.8732824Z # RUNTIME: Info: RUNTIME_0142 finish called.
2020-05-28T14:44:49.8734200Z # KERNEL: Time: 55000001 ps,  Iteration: 0,  TOP instance.
2020-05-28T14:44:49.8892520Z # KERNEL: stopped at time: 55000001 ps
2020-05-28T14:44:49.8948544Z # VSIM: Simulation has finished. There are no more test vectors to simulate.
2020-05-28T14:44:49.8949853Z endsim
2020-05-28T14:44:49.9389119Z # VSIM: Simulation has finished.
```

garmin-mjames, 2020-05-28:
My guess is this involves the VHPI callbacks. Try it with https://github.com/cocotb/cocotb/pull/1498.

garmin-mjames, 2020-05-28:
The `BitMonitor` should probably also have `await ReadOnly()` before getting the signal value:

https://github.com/cocotb/cocotb/blob/924f35a3b7d39543118b7bfaed77dd4808e6612b/examples/dff/tests/dff_cocotb.py#L62-L63

PR #1498 (2020-07-09): Update VHPI callbacks
**PR for testing simulators that support VHPI**
Closes #1061
Closes #1880 

The current use of the `vhpiCbRepEndOfProcesses` callback for `ReadWrite`
triggers can cause subtle issues in simulation. Signal values can take
multiple delta cycles to propogate through designs, and
`vhpiCbRepEndOfProcesses` is called at the end of the current delta cycle.
This means that depending on when signal values are cached when using
the `<=` operator, signal values may be set at the wrong time relative
to other signals.

Changing `ReadWrite` to `vhpiCbRepLastKnownDeltaCycle` and `ReadOn [... cut]
closed 2020-07-09 by ktbarrett