cocotb/cocotb #1884: No access to signals below 'generate' hierarchy for Verilator

stu-e, 2020-05-29:
Signals below a point in the HDL hierarchy instantiated with a generate-if statement are 'invisible' to the Python test bench.
I placed the endian swapper module within a generate block within a wrapper module and this is the error:

`Traceback (most recent call last):
  File "/home/stuart/cocotb/examples/endian_swapper/tests/test_endian_swapper.py", line 59, in stream_out_config_setter
    if dut.MY_GEN_BLOCK.i_endian_swapper.byteswapping.value:
  File "/home/stuart/.local/lib/python3.8/site-packages/cocotb/handle.py", line 295, in __getattr__
    raise AttributeError("%s contains no object named %s" % (self._name, name))
AttributeError: endian_swapper_wrapper contains no object named MY_GEN_BLOCK`

Looking at the Verilator Vtop.h file, the path is correct

themperek, 2020-05-29:
Most lily ventilator issue.

garmin-mjames, 2020-06-01:
Could you post a log with environment variable `COCOTB_LOG_LEVEL=DEBUG`? It should show debug log messages related to traversing the hierarchy.

stu-e, 2020-06-01:
[endian_swapper_debug.txt](https://github.com/cocotb/cocotb/files/4713067/endian_swapper_debug.txt)

garmin-mjames, 2020-06-01:
Related to #1432. The problem appears to be that Verilator gives a type of `vpiScope` for `MY_GEN_BLOCK`, which cocotb does not currently support.

stu-e, 2020-06-01:
Ok thanks for checking

imphil, 2020-06-01:
@stu-e can you come up with a minimal Verilog and cocotb example that we can add to our test bench to keep track of this issue? You can either paste it here, or (if you feel comfortable) open an pull request.

I guess we'll then need to have a closer look what the next steps could be; I wouldn't be surprised if this is something that needs to be fixed in Verilator.

garmin-mjames, 2020-11-24:
I think `test_discovery.pseudo_region_access` covers this.

AndrewNolte, 2024-01-22:
I put a vpi dump test case in Verilator here https://github.com/verilator/verilator/pull/4838/files#diff-5c0aca2db885444f49208c5bbbf322d83b90b08ffb0c320426f493974f59057e The vpiNames look correct in that .out file, but they still don't show up on cocotb.

ktbarrett, 2025-01-07:
@AndrewNolte Is this issue still present?

stu-e, 2025-01-09:
I just checked using CocoTB 1.9.2 and Verilator 5.032. From Python test bench code I can now access signals inside a generate loop at least one level below the top of the hierarchy of the DUT. So I assume this is no longer a problem.

AndrewNolte, 2025-01-09:
Yeah I think this was fixed from either https://github.com/verilator/verilator/pull/4918 or https://github.com/verilator/verilator/pull/4965

stu-e, 2025-01-09:
I'm going to double check this because in both #1884 and #4344 there is a module instanced within a generate block. The test I did today only checked a signal instanced in a generate block and not a signal within a module in a generate block.

davekeeshan, 2025-01-09:
I have checked the test I wrote for #4344 against verilator@5.032 and cocotb 2.0 and it is still failure as recorded in that test.

stu-e, 2025-01-10:
There is still a problem. If I create a generate block that contains only an instance of a module, the generate block is invisible to CocoTB. If I view a waveform dump (dump.fst) I can browse the generate block no problem.
If I instantiate a signal within that block then the generate block magically reappears and can be accessed by CocoTB.
This problem appears to be limited to Verilator. If I use Questa then CocoTB can access the module instanced in the generate block.