cocotb/cocotb #1893: Guess or better document the need to use TOPLEVEL_LANG

imphil, 2020-05-31:
TOPLEVEL_LANG is a requirement for using some, but not all simulators. To make our "recommended" Makefile portable across simulators, we should *either*

- (easy option) Add it to our recommended makefile at https://docs.cocotb.org/en/latest/quickstart.html#creating-a-makefile
- or (better option) Guess TOPLEVEL_LANG if possible. A reasonably safe guess would be: if only VERILOG_SOURCES is set, use verilog. If only VHDL_SOURCES is set, use vhdl. Otherwise, do nothing.

ktbarrett, 2020-05-31:
This leads to an interesting question. In the root `cocotb` module we have the `LANGUAGE` attribute and we use it to 'turn off' certain tests. Is this really a good idea? Should we be agnostic of language in cocotb and rely on the makefiles to 'turn off' tests that would fail if the `TOPLEVEL_LANG` wasn't what we expected?

That decision will affect whether `TOPLEVEL_LANG` be documented as a make variable or a cocotb environment variable.

ktbarrett, 2020-07-13:
Additionally, `TOPLEVEL_LANG` can only be `verilog` on Icarus and Verilator, and `vhdl` on GHDL.

PR #1982 (2020-09-05): Detection of TOPLEVEL_LANG
Adds support for the automatic detection of an appropriate `TOPLEVEL_LANG` value based on the value of `VERILOG_SOURCES` and `VHDL_SOURCES`. This PR also adds detection of invalid `TOPLEVEL_LANG` values in simulators that only support one language like Verilator, Icarus, and GHDL. Closes #1893.
closed 2020-09-05 by ktbarrett