cocotb/cocotb #1894: Docs: Preferred way to set dut/test-only log level?

imphil, 2020-05-31:
`COCOTB_LOG_LEVEL` sets the log level for testbench code and cocotb internals at the same time. (I'm not sure that was always the case, but it is certainly now.) This typically leads to a lot of GPI-related messages at the DEBUG level, which hide user-generated DEBUG messages.

What can we recommend to users to set the log level for only their testbench code?

- Use `dut._log.setLevel()` if a `dut` variable is available. Doesn't scale well with multiple tests.
- Use `logging.getLogger('cocotb.TOPLEVELNAME').setLevel(logging.DEBUG)`, which requires the toplevel name to be known. 

I guess there are a couple other options, but I'm not really convinced by any of them. Am I missing the obviously nice one?

eric-wieser, 2020-05-31:
We could change this line:

https://github.com/cocotb/cocotb/blob/3b80022b13bd1b5876b45158cfb8c7d064cf51a2/cocotb/handle.py#L78

To use `cocotb.handles.%s` instead, although that makes the name longer.

eric-wieser, 2020-05-31:
Do we even have a recommended log instance for users to log to?

ktbarrett, 2020-05-31:
I almost always use `cocotb.log`. The other logger handles are hard to access where I'd want to use them, or don't really offer any value. Task-specific loggers would be useful, but there is no easy way to access them while executing a task.

eric-wieser, 2020-05-31:
Is `cocotb.log` `logging.getLogger('cocotb')`? If so, there's no easy way to set its level without everything else inheriting it.

ktbarrett, 2020-05-31:
I find that beneficial, if I mention debug logs, why wouldn't I want everything to be debug? At worst I'm combing through slightly longer logs. I'm just stating what I and everyone I know do.

imphil, 2020-05-31:
> Do we even have a recommended log instance for users to log to?

Kind of, our documentation recommends using `dut._log`, which produces logs in `cocotb.TOPLEVEL` (e.g. at https://docs.cocotb.org/en/latest/quickstart.html#creating-a-test).

> I mention debug logs, why wouldn't I want everything to be debug?

I  know that there are different preferences regarding logging, and different use cases (e.g. unattended batch runs vs. development of a testbench). This issue is under the premise that users want to set a different log level for their own code than for the cocotb internals.

eric-wieser, 2020-06-09:
> This issue is under the premise that users want to set a different log level for their own code than for the cocotb internals.

Is the answer simply "create your own logger"? That can be done with something simple at the top of their test python module like
```python
import logging
log = logging.getLogger(__name__)
```

imphil, 2020-06-09:
> Is the answer simply "create your own logger"? 

This would indicate we not recommend using `dut._log` any more (as we do in the documentation today), but a custom logger instead?

eric-wieser, 2020-06-11:
That might be one way to go, yes. The main example I can think of for when you do still want to use `dut._log` directly is:
```python
class MyModuleDriver:
    def __init__(self, hdl):
        self.hdl = hdl
    def foo(self):
        self.hdl._log.info("doing the foo thing")
        self.hdl.signal <= 1
```
which you might use in one test as `MyModuleDriver(dut)`, but in another as `MyModuleDriver(dut.left), MyModuleDriver(dut.right)`.