cocotb/cocotb #1899: Produce steps to create cocotb test coverage

imphil, 2020-06-02:
Produce a framework/steps to collect test coverage for cocotb, i.e. how much coverage our in-tree cocotb tests give us on the cocotb code base.

This is different from the built-in coverage support in cocotb, which is meant to collect coverage for user testbench code running with cocotb.

In scope for this issue:
- Provide commands to collect coverage for our runs probably through a configuration for python-coverage.

Going beyond this issue, we can then use and report the data in public form.

- Add these commands to our CI infrastructure.
- Report it to cloud solutions like codecov.io. 

Note that coverage numbers depend on various environment factors: the simulator, the operating system, etc. Typically, these coverage numbers are merged when reporting them (e.g. codecov.io does that by default).

ktbarrett, 2020-06-05:
I've done some initial work on this. I put it right next to the existing coverage collected in the regression manager, but coverage is pretty low. A lot of that is because it runs *after* all the modules are loaded and everything in `__init__.py` has run

To get realistic coverage info, we need to start the coverage before we ever import the cocotb module. Once we have the custom entry point in place (I have a few large PRs before then), we can start the coverage tracer in there, and kill the coverage during shutdown in `_sim_event` (or a wrapper that works with the custom entry point.

The chicken-egg problem of getting the location of cocotb libraries and covering the initial importof cocotb can be solved using `importlib.reload`.

eric-wieser, 2020-06-23:
I think we should either consider this closed by @ktbarrett in #1896, or adjust the title to make this a documentation issue

ktbarrett, 2020-06-23:
We did do what the OP said, and it's an internal process, what needs documentation?

We probably do need to work on internal documentation...

closed 2020-07-10 by ktbarrett