cocotb/cocotb #1900: examples/mean/hdl/mean.sv not compiling with verilator

regis-c, 2020-06-02:
In lastest master branch, trying to compile mean example in  verilator fails.

Getting 3 warning reported by verilator, this stop cocotb makefile flow as Error.

warning are:
%Warning-WIDTH: /home/regisc/cocotb/examples/mean/tests/../hdl/mean.sv:42:21: Operator ADD expects 8 bits on the RHS, but RHS's ARRAYSEL generates 6 bits.
                                                                            : ... In instance mean
   42 |       v_sum = v_sum + i_data[i];
      |                     ^
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
%Warning-WIDTH: /home/regisc/cocotb/examples/mean/tests/../hdl/mean.sv:60:17: Operator ASSIGNW expects 6 bits on the Assign RHS, but Assign RHS's SHIFTR generates 8 bits.
                                                                            : ... In instance mean
   60 |   assign o_data = s_sum >> $clog2(BUS_WIDTH);
      |                 ^
%Warning-STMTDLY: /home/regisc/cocotb/examples/mean/tests/../hdl/mean.sv:69:6: Unsupported: Ignoring delay on this delayed statement.
                                                                             : ... In instance mean
   69 |     #1;
      |      ^

By default, verilator is quite strict on width mismatch (not usual with other compiler I am used to).
Cab be fixed with 
	EXTRA_ARGS += -Wno-WIDTH
in Makefile.

For third warning, STMTDLY can be fixed by usign a "#1ps" instead of #1 used.
(more system verilog style).
Or adding a 
	EXTRA_ARGS += -Wno-STMTDLY
in makefile.

Not sure what the best fix : makefile or .sv change.

I can quickly provide a PR for this one. (almost ready on my side)

regis-c, 2020-06-02:
Using Verilator 4.034

cmarqu, 2020-06-02:
Could you check whether ``#1ps`` is accepted by Icarus Verilog? If so, I'd make the change in SV, otherwise, switch off the error.

cmarqu, 2020-06-02:
I'd probably make Verilator ignore the width warnings -  fixing this "properly" would make the SV code harder to read, distracting from its purpose being a vehicle for teaching cocotb.

regis-c, 2020-06-02:
I tried to run with SIM=icarus.
My old 10.1 version didn't like code for other reason, but after upgrading to 10.3, the #1ps seems to be well accepted. So mean.sv fix may be OK.  
But I still can investigate latest version of verilator.

Agreed adding bitwidth casting in verilog is not clean fix for cocotb tutorial. I prefer makefile warning removal as well.  

Will try to prepare a PR tomorrow.

elgorwi, 2020-06-03:
Just pushed some fixes (PR #1901 ) that I had been working on to address #1851 .

regis-c, 2020-06-03:
OK, thanks for the feedback. Let's wait for these other PR to see what's next from there.

regis-c, 2020-07-23:
I checked back after upgrading to latest cocotb and verilator.
cocotb v1.5.0.dev0
Verilator version 4.039 devel
Now, Test compiled and passed.

Issue can be closed.

closed 2020-07-23 by regis-c