cocotb/cocotb #1904: XGMII Driver initialization incomplete

LeChuck42, 2020-06-04:
The XGMII Driver initialization is incomplete resulting in a corrupted word on the first transfer.

The `_XGMII_Bus` helper class sets it's value to XGMII IDLE during the `__init__`:
https://github.com/cocotb/cocotb/blob/97e28803308d09d2bd885e4279d5036ef9861d42/cocotb/drivers/xgmii.py#L61-L83

But this value is never read during initialization of `XGMII(Driver)` ...
https://github.com/cocotb/cocotb/blob/97e28803308d09d2bd885e4279d5036ef9861d42/cocotb/drivers/xgmii.py#L122-L140

... and as reading the value clears the internal `_integer` buffer of the helper class ...
https://github.com/cocotb/cocotb/blob/97e28803308d09d2bd885e4279d5036ef9861d42/cocotb/drivers/xgmii.py#L104-L113
 
... the first word of the first transaction (actually the PREAMBLE and START byte) is OR'ed with the IDLE value corrupting the transmission.

One way of fixing would be to simply read the value of the bus during `__init__` of `XGMII(Driver)` and thus clearing the buffer.
```
self.signal <= self.bus.value
```

Alternatively (and maybe cleaner) we could remove the default value of the helper class and use the `idle` helper of `XGMII(Driver)`.

I'll submit a pull for the latter, as I feel it's cleaner. Thoughts?

ktbarrett, 2020-06-04:
Do you have a design and tests cocotb could have to test this interface going forward?

ktbarrett, 2020-06-18:
This was resolved by #1905, right?

closed 2020-06-18 by eric-wieser

ecg173-bi, 2020-10-27:
Hi!, 
I think that I have this error, but I am new at this, and I am not sure about the solution. Could someone give me some more indications? 
My xgmii file is as indicated before in lines 61-83, 104-113 and 122-140 and the error is the following one: 

------ Error: ------- 
# COUT:      0.00ns ERROR    cocotb.regression                         regression.py:400  in _score_test                     Test Failed: write_address_0 (result was TypeError)
# COUT:                                                                                                                      Traceback (most recent call last):
# COUT:                                                                                                                        File "/home/.../test_gmii_rx.py", line 35, in write_address_0
# COUT:                                                                                                                          axim = XGMII(dut, dut.gmii_clk)
# COUT:                                                                                                                        File "/home/user/.local/lib/python3.7/site-packages/cocotb/drivers/xgmii.py", line 136, in __init__
# COUT:                                                                                                                          self.idle()
# COUT:                                                                                                                        File "/home/user/.local/lib/python3.7/site-packages/cocotb/drivers/xgmii.py", line 160, in idle
# COUT:                                                                                                                          self.signal <= self.bus.value
# COUT:                                                                                                                      TypeError: '<=' not supported between instances of 'HierarchyObject' and 'BinaryValue'
--------
I have tried to add self.signal <= self.bus.value to init, but I must have done something wrong, because nothing changed.  
Thanks in advance.

garmin-mjames, 2020-10-27:
@ecg173-bi `self.signal` is a `HierarchyObject`, which means it's not a modifiable object that you can assign a value to.

This is a culprit: `axim = XGMII(dut, dut.gmii_clk)`. The first argument should be the bus, but you are passing in the top-level `dut` handle.

ecg173-bi, 2020-10-28:
It is true :(. And some indications to create the bus? I have tried using BusDriver function and also using a BusMonitor similar to that one in matrix multiplication example but, again, I have errors and I am a bit lost. Which would be the right way?
Thanks again.

ecg173-bi, 2020-10-28:
Last tried defining:

    stream_in_bus = Bus(dut, "stream_in", ["gmii_clk", "gmii_rst","rx_er", "rx_dv", "rxd"])
  
    # Reset
    dut.stream_in_gmii_rst <= 1
    axim = XGMII(stream_in_bus, dut.stream_in_gmii_clk, interleaved = True)

gives the next error: 

 Traceback (most recent call last):
# COUT:                                                                                                                        File "/home/user/repos/.../test_gmii_rx.py", line 109, in write_address_0
# COUT:                                                                                                                          axim = XGMII(stream_in_bus, dut.stream_in_gmii_clk, interleaved = True)
# COUT:                                                                                                                        File "/home/user/.local/lib/python3.7/site-packages/cocotb/drivers/xgmii.py", line 131, in __init__
# COUT:                                                                                                                          self.log = signal._log
# COUT:                                                                                                                      AttributeError: 'Bus' object has no attribute '_log'

ktbarrett, 2020-10-28:
@ecg173-bi Have a look [here](https://github.com/cocotb/cocotb/blob/e661dc6dd9bae33e8b5dcc4d170917a3b6cf328a/cocotb/drivers/xgmii.py#L117) at the type annotation for the XGMIIDriver, the `signal` argument is supposed to be a sim handle not a `Bus`. I'd suggest the following, assuming `stream_in` is your XGMII bus.

```python
XGMII(dut.stream_in, dut.stream_in_gmii_clk, interleaved = True)
```