cocotb/cocotb #1907: Compile VHDL source with +acc by default in Questa

ktbarrett, 2020-06-05:
With `TOPLEVEL_LANG=vhdl` `+acc` is not added by default like it is with Verilog and it causes a dozen tests to fail.

eric-wieser, 2020-06-05:
> . Likewise tests assume a 1 ps precision,

Can you point to an example of such a test?

ktbarrett, 2020-06-05:
One was recently fixed by @cmarqu, but I can't remember what it was. [test_timing_triggers.test_timer_with_units](
https://github.com/cocotb/cocotb/blob/8bb292145bd7bc5f1b94abf5ff4b287eba346f40/tests/test_cases/test_cocotb/test_timing_triggers.py#L58) is a test the fails the Questa regression because of this. I'd probably have to look harder for more. You can see in the history their assumption of picosecond since everything references picosecond. `Timer` did until recently.

I'm not saying it's right, it's just what they did in the past. We should fix those tests.

cmarqu, 2020-06-05:
One example was https://github.com/cocotb/cocotb/commit/e6e98a85f49de3d5f50410392f02c06ea1c4f11c