cocotb/cocotb #1913: issues with cocotb on macOS Catalina with homebrew Python3

denniscote, 2020-06-06:
I have installed cocotb on MacOS Catalina with a homebrew Python installation. I also have icarus verilog and verilator installed by homebrew.

OS: Catalina 10.15.5
Python 3.7.7
icarus-verilog 10.3
verilator 4.034

When I tried to build the cocotb dff sample project shown on the GitHub page at https://github.com/cocotb/cocotb I ran into several errors.

First the simple Makefile on that page includes the file Makefile.sim which uses a variable COCOTB_SHARE_DIR which has not been set. I added the following lines to the simple Makefile to get past the first few errors.

```
COCOTB_SHARE_DIR = $(shell cocotb-config --share)
export COCOTB_SHARE_DIR
```

Next I ran into errors due to Makefile.pylib.Darwin using the system python which is still python 2.7.16. Since these command are hard coded using the python command rather than python3 I didn't see a way to overide this other than changing the python commands in this Makefile.

```
ifneq ($(COCOTB_PYTHON_DEBUG),)
	DBG_PYTHON_VERSION=$(shell python3 -c 'from __future__ import print_function; import distutils.sysconfig; print(distutils.sysconfig.get_python_version())')
	PYTHON_BIN?=python$(DBG_PYTHON_VERSION)-dbg
else
	PYTHON_BIN?=python3
endif  
```

At this point cocotb simulations run with either icarus or verilator, but I am still getting some errors regarding the python shared library. I copied some of the output below for reference. 

```
PYTHONPATH=/Users/dennis/projects/verilator/test_dff/build/libs/x86_64:/Users/dennis/projects/verilator/test_dff:/usr/local/lib/python3.7/site-packages: LD_LIBRARY_PATH=/Users/dennis/projects/verilator/test_dff/build/libs/x86_64 LD_LIBRARY_PATH=/Users/dennis/projects/verilator/test_dff/build/libs/x86_64::/usr/local/opt/python/Frameworks/Python.framework/Versions/3.7/lib:/usr/local/opt/python/Frameworks/Python.framework/Versions/3.7/lib MODULE=test_dff \
        TESTCASE= TOPLEVEL=dff TOPLEVEL_LANG=verilog COCOTB_SIM=1 \
        sim_build/dff 
Unable to open lib Python.framework/Versions/3.7/Python (dlopen(Python.framework/Versions/3.7/Python, 9): image not found)
     -.--ns ERROR    cocotb.gpi                                  gpi_embed.c:139  in embed_init_python               Failed to find Python shared library
```

If I break down the first long line into individual paths I see the following:

```
PYTHONPATH=
/Users/dennis/projects/verilator/test_dff/build/libs/x86_64:
/Users/dennis/projects/verilator/test_dff:/usr/local/lib/python3.7/site-packages: 

LD_LIBRARY_PATH=
/Users/dennis/projects/verilator/test_dff/build/libs/x86_64 

LD_LIBRARY_PATH=
/Users/dennis/projects/verilator/test_dff/build/libs/x86_64:
:
/usr/local/opt/python/Frameworks/Python.framework/Versions/3.7/lib:
/usr/local/opt/python/Frameworks/Python.framework/Versions/3.7/lib 

MODULE=test_dff
```

I'm not sure if it matters but there are two LD_LIBRARY_PATH values being set, and the first path in the second is the same as the only path in the first. Then there appears to be an empty path in the second, and finally, the same path repeated twice at the end of the second value. 

The next issue is the error where it reports it failed to find the Python shared library. It looks like it is trying to open Python.framework/Versions/3.7/Python with a relative file path. This file exists but the location is /usr/local/opt/python/Frameworks/Python.framework/Versions/3.7/Python. The incorrect path seems to be determined in Makefile.pylib.Darwin where it sets PYTHON_DYN_LIB using

`sysconfig.get_config_var("LDLIBRARY")`

The missing prefix can be obtained from python with: 

`sysconfig.get_config_var("PYTHONFRAMEWORKPREFIX")`

I tried changing Makefile.pylib.Darwin to add this prefix as shown below, but I get the same error messages about not finding the shared library image, so I believe the wrong path is being generated somewhere else. 

`PYTHON_DYN_LIB:=$(shell $(PYTHON_BIN) -c 'from __future__ import print_function; from distutils import sysconfig; print(sysconfig.get_config_var("PYTHONFRAMEWORKPREFIX")+"/"+sysconfig.get_config_var("LDLIBRARY").replace(".a", ".so"))')`

Hopefully, this can help you streamline the process for new users and maybe you can eliminate the messages about the Python shared library.

themperek, 2020-06-07:
Can use use development version (1.4) or there is some reason you need 1.3?

denniscote, 2020-06-07:
I am using the version I got by running

pip3 install cocotb

as suggested by the installation instructions at https://github.com/cocotb/cocotb <https://github.com/cocotb/cocotb>.

How would I install the development version? 

I went with the basic installation just to try out cocotb. I wanted to kick the tires, so to speak, before I dig into it too far.  Then I ran into a bunch of errors which I’m sure would discourage many potential new users.

eric-wieser, 2020-06-07:
Are you using the examples from master, or from the release you downloaded?

denniscote, 2020-06-07:
I am using the example I copied and pasted from the Usage section of the Github page at https://github.com/cocotb/cocotb <https://github.com/cocotb/cocotb>

eric-wieser, 2020-06-07:
Those examples are in the development readme, so will only work with a development install.

@themperek, do we need to add a warning here telling users to follow the instructions at https://docs.cocotb.org/en/latest/quickstart.html#running-your-first-example instead?

denniscote, 2020-06-07:
This makes no sense!

The instructions on the webpage you link (https://docs.cocotb.org/en/latest/quickstart.html#running-your-first-example <https://docs.cocotb.org/en/latest/quickstart.html#running-your-first-example>), say the same thing as those onthe Github page.

Make sure you have the prerequisites <https://docs.cocotb.org/en/latest/install.html#install-prerequisites> (Python with development packages, a C++ compiler with development packages, GNU Make, a supported simulator <https://docs.cocotb.org/en/latest/simulator_support.html#simulator-support>) and cocotb itself (pip install cocotb) available.

I have the prerequisites and I installed cocotb using “pip3 install cocotb” (pip3 instead of pip to install into my Python3.7 site packages) exactly as stated. This installs the current release version of cocotb, version 1.3. 

Then I created a sample Verilog file and Makefile. The sample Makefile I used ends with the exact same line as the one on the webpage you linked.

include $(shell cocotb-config --makefiles)/Makefile.sim

This pulls in Makefiles from the cocotb 1.3 release which have missing and incorrect definitions. Once I tracked down those issues, my sample verilog and Python test files work correctly. There is not a problem with the example files. 

I’m sure that if I used the example files on the page you linked using the instructions on that page, I would get a cocotb version 1.3 installation and I would have a Makefile that included the Makefile.sim from cocotb 1.3 without a valid definition of COCOTB_SHARE_DIR, and as a result the example would produce the same errors I saw.

Again, as long as "pip install cocotb" installs version 1.3, your examples should work with version 1.3. Doing anything else is just going to lead users astray. 

There isn’t a problem with the sample files. There are problems with the version 1.3 Makefiles that are installed by a normal installation of cocotb. When the Makefile issues are corrected the sample files run just fine. They still produce some errors (unable to load shared library), but the tests run and produce output wave files that I can view in gtkwave. 

If you have a development version you should have a separate web page for advanced users that explains what the differences are and how to install it.

eric-wieser, 2020-06-07:
You're right, the webpage I linked is not helpful to you either. The one you want is

https://docs.cocotb.org/en/v1.3.1/quickstart.html#running-your-first-example

The v1.3.1 docs contain quickstart info that will work with the v1.3.1 release from pypi.

It sounds like we need to stick a big disclaimer at the start of the quickstart page, recommending users look at the _released_ docs.

eric-wieser, 2020-06-07:
> Again, as long as "pip install cocotb" installs version 1.3, your examples should work with version 1.3. Doing anything else is just going to lead users astray.

I feel like we're stuck in a catch 22 here - our examples double as tests, so we want to update them in our repo to make use of new features as soon as we add them. This means we end up with examples on GitHub that do not work against the pypi release.

Do you know how other projects handle this?

denniscote, 2020-06-07:
Hi Eric,

Well this page is better! At least the example now runs. 

The example Makefile here includes the file Makefile.inc before including Makefile.sim. When I add that to my sample Makefile I can remove the definition of COCOTB_SHARE_DIR I added earlier.

The other issues I brought up remain though. MacOS includes an installation of Python 2.7 out of the box. The command python refers to this installation. Since cocotb requires Python 3.5 or better, the user must do a Python 3 installation. Since it is not advisable to remove or replace the factory Python 2 installation since Apple's code may use it, we normally get a python3 command. This is what you get if you use homebrew "brew install python” to install python today.

dennis@Denniss-iMac makefiles % which python
/usr/bin/python
dennis@Denniss-iMac makefiles % which python3
/usr/local/bin/python3

/usr/bin/python is Apples Python 2.7 and /usr/local/bin/python3 is the Python3.7 installed by homebrew.

As a result your Makefile.pylib.Darwin needs to use python3 instead of python to refer to the required Python3.5+ installation instead of Apples factory installed Python2.7 installation. I changed the file in two locations where “python" was hard coded. I’m not sure the debug section works since I am doing DEBUG testing, but “shell python” will run python 2.7. 

ifneq ($(COCOTB_PYTHON_DEBUG),)
	DBG_PYTHON_VERSION=$(shell python3 -c 'from __future__ import print_function; import distutils.sysconfig; print(distutils.sysconfig.get_python_version())')
	PYTHON_BIN?=python$(DBG_PYTHON_VERSION)-dbg
else
	PYTHON_BIN?=python3
endif

I am also still getting an error reported when it tries to load the Python shared library. It is trying to load from an incomplete relative path. 

dennis@Denniss-iMac test_dff % make SIM=verilator
/Applications/Xcode.app/Contents/Developer/usr/bin/make results.xml
PYTHONPATH=/Users/dennis/projects/verilator/test_dff/build/libs/x86_64:/Users/dennis/projects/verilator/test_dff:/usr/local/lib/python3.7/site-packages: LD_LIBRARY_PATH=/Users/dennis/projects/verilator/test_dff/build/libs/x86_64 LD_LIBRARY_PATH=/Users/dennis/projects/verilator/test_dff/build/libs/x86_64::/usr/local/opt/python/Frameworks/Python.framework/Versions/3.7/lib:/usr/local/opt/python/Frameworks/Python.framework/Versions/3.7/lib:/usr/local/opt/python/Frameworks/Python.framework/Versions/3.7/lib:/usr/local/opt/python/Frameworks/Python.framework/Versions/3.7/lib MODULE=test_dff \
        TESTCASE= TOPLEVEL=dff TOPLEVEL_LANG=verilog COCOTB_SIM=1 \
        sim_build/dff 
Unable to open lib Python.framework/Versions/3.7/Python (dlopen(Python.framework/Versions/3.7/Python, 9): image not found)
     -.--ns ERROR    cocotb.gpi                                  gpi_embed.c:139  in embed_init_python               Failed to find Python shared library

     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:104  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:340  in embed_sim_init                  Running on Verilator version 4.034 2020-05-03
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:341  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:138  in _initialise_testbench           Running tests with cocotb v1.3.1 from /usr/local/lib/python3.7/site-packages
     0.00ns INFO     cocotb                                      __init__.py:155  in _initialise_testbench           Seeding Python random module with 1591569096
     0.00ns INFO     cocotb.regression                         regression.py:190  in initialise                      Found test test_dff.test_dff_simple
     0.00ns INFO     cocotb.regression                         regression.py:348  in execute                         Running test 1/1: test_dff_simple
     0.00ns INFO     ...test.test_dff_simple.0x10b751cd0       decorators.py:253  in _advance                        Starting test: "test_dff_simple"
                                                                                                                     Description:  Test that d propagates to q 
   197.00ns INFO     cocotb.regression                         regression.py:288  in handle_result                   Test Passed: test_dff_simple
   197.00ns INFO     cocotb.regression                         regression.py:213  in tear_down                       Passed 1 tests (0 skipped)

The shared library is located at /usr/local/opt/python/Frameworks/Python.framework/Versions/3.7/Python.

dennis@Denniss-iMac 3.7 % pwd
/usr/local/opt/python/Frameworks/Python.framework/Versions/3.7
dennis@Denniss-iMac 3.7 % file Python 
Python: Mach-O 64-bit dynamically linked shared library x86_64
 
As I said before, I tried to replace the PYTHON_DYN_LIB definition is the file Makefile.pylib.Darwin, but I got the same error messages, so I assume this incorrect path is being generated somewhere else.

denniscote, 2020-06-07:
Hi Eric,

No, I don’t know how other projects do this. I can only assume they must separate the new and old examples so that the old examples remain in place for users, while the developers can create and test with new examples. The new examples would then replace the old examples at the same time the new version is released to users (i.e. when cocotb is updated on pipit). That is part of what it means to release a new version, an entire set of files that work together are made available at the same time. Copying files once in a while for a release is a fast and cheap operation.

Thanks for the work on cocotb. It looks very interesting.

ktbarrett, 2025-04-30:
I have tried this recently and didn't have any issues. Moving to `LIBPYTHON_LOC` and `PYGPI_PYTHON_BIN` seem to avoid a lot of the issues mentioned here. if there are still issues, please open a new issue.

closed 2025-04-30 by ktbarrett