cocotb/cocotb #1926: C:\cocotb\cocotb-1.3.1\examples\dff fails with Modelsim PE

mm4dd, 2020-06-11:
Hello. 

I have recently installed the 32 bit version of miniconda (Python 3.7 on Windows 10) specified in the installation section of the cocotb online docs. I also followed the instructions to install cocotb:  pip install --global-option build_ext --global-option --compiler=mingw32 cocotb .  The installation has worked fine. However, I have not been able to run any of the example tests in \cocotb-1.3.1\examples\.  The failures vary from modelsim setup problems in the dff project to python environment failures in the adder project. 

I have tried various older version of miniconda3 32 bit and various older versions of cocotb back to 1.2.0 but I have had no luck getting the example cocotb tests to run on two different windows 10 computers.

 To keep it simple I thought I would ask about getting the dff/ test working. The output from the failing dff test is shown below. I found the problem to be in the runsim.do file:  vlog -work work +define+COCOTB_SIM -sv -timescale 1ns/1us -mfcu +acc=rmb   /c/cocotb/cocotb-1.3.1/examples/dff/tests/../hdl/dff.v. The timescale directive is incorrect. I have tried fixing the issue but it leads to the simulation not being able to find the dff.v file in the hdl directory. 
It just seems like the whole environment is not running properly. If anyone has any suggestion on how I can sort things out it would be much appreciated. 

 Thanks in advance.  

(base) C:\cocotb\cocotb-1.3.1\examples\dff\tests>make SIM=modelsim TOPLEVEL_LANG=verilog
make results.xml
make[1]: Entering directory '/c/cocotb/cocotb-1.3.1/examples/dff/tests'
mkdir -p sim_build
mkdir -p /c/cocotb/cocotb-1.3.1/examples/dff/tests/build/obj/i686
mkdir -p /c/cocotb/cocotb-1.3.1/examples/dff/tests/build/libs/i686
make -C /c/Users/user1/miniconda3/lib/site-packages/cocotb/share/lib/utils SIM=modelsim
make[2]: Entering directory '/c/Users/user1/miniconda3/lib/site-packages/cocotb/share/lib/utils'
gcc -m32 -g -DDEBUG -shared -DMODELSIM -c -I/c/Users/user1/miniconda3/lib/site-packages/cocotb/share/include -IC:\Users\user1\miniconda3\include  -o /c/cocotb/cocotb-1.3.1/examples/dff/tests/build/obj/i686/cocotb_utils.o cocotb_utils.c
gcc -m32 -g -DDEBUG -shared -DMODELSIM -shared -Wl,-no-undefined -Wl,-enable-runtime-pseudo-reloc-v2 -Wl,--enable-auto-import -L/c/Users/user1/miniconda3/libs -o /c/cocotb/cocotb-1.3.1/examples/dff/tests/build/libs/i686/libcocotbutils.dll /c/cocotb/cocotb-1.3.1/examples/dff/tests/build/obj/i686/cocotb_utils.o    
make[2]: Leaving directory '/c/Users/user1/miniconda3/lib/site-packages/cocotb/share/lib/utils'
make -C /c/Users/user1/miniconda3/lib/site-packages/cocotb/share/lib/gpi_log SIM=modelsim
make[2]: Entering directory '/c/Users/user1/miniconda3/lib/site-packages/cocotb/share/lib/gpi_log'
gcc -m32 -g -DDEBUG -shared -DFILTER -DMODELSIM -c -I/c/Users/user1/miniconda3/lib/site-packages/cocotb/share/include -IC:\Users\user1\miniconda3\include  -o /c/cocotb/cocotb-1.3.1/examples/dff/tests/build/obj/i686/gpi_logging.o gpi_logging.c
gcc -m32 -g -DDEBUG -shared -DFILTER -DMODELSIM -shared -Wl,-no-undefined -Wl,-enable-runtime-pseudo-reloc-v2 -Wl,--enable-auto-import -L/c/Users/user1/miniconda3/libs -o /c/cocotb/cocotb-1.3.1/examples/dff/tests/build/libs/i686/libgpilog.dll /c/cocotb/cocotb-1.3.1/examples/dff/tests/build/obj/i686/gpi_logging.o   -lPython37
make[2]: Leaving directory '/c/Users/user1/miniconda3/lib/site-packages/cocotb/share/lib/gpi_log'
make -C /c/Users/user1/miniconda3/lib/site-packages/cocotb/share/lib/embed SIM=modelsim
make[2]: Entering directory '/c/Users/user1/miniconda3/lib/site-packages/cocotb/share/lib/embed'
gcc -m32 -g -DDEBUG -shared -DPYTHON_SO_LIB=Python37.dll -DMODELSIM -c -I/c/Users/user1/miniconda3/lib/site-packages/cocotb/share/include -IC:\Users\user1\miniconda3\include  -o /c/cocotb/cocotb-1.3.1/examples/dff/tests/build/obj/i686/gpi_embed.o gpi_embed.c
In file included from C:/Users/user1/miniconda3/lib/site-packages/cocotb/share/include/gpi.h:60:0,
                 from C:/Users/user1/miniconda3/lib/site-packages/cocotb/share/include/embed.h:33,
                 from gpi_embed.c:34:
gpi_embed.c: In function 'embed_init_python':
gpi_embed.c:167:98: warning: implicit declaration of function 'getpid' [-Wimplicit-function-declaration]
         LOG_ERROR("Waiting for %lu seconds - attach to PID %d with your debugger\n", sleep_time, getpid());
                                                                                                  ^
C:/Users/user1/miniconda3/lib/site-packages/cocotb/share/include/gpi_logging.h:55:98: note: in definition of macro 'LOG_ERROR'
 #define LOG_ERROR(...)     gpi_log("cocotb.gpi", GPIError,         __FILE__, __func__, __LINE__, __VA_ARGS__);
                                                                                                  ^
gcc -m32 -g -DDEBUG -shared -DPYTHON_SO_LIB=Python37.dll -DMODELSIM -shared -Wl,-no-undefined -Wl,-enable-runtime-pseudo-reloc-v2 -Wl,--enable-auto-import -L/c/Users/user1/miniconda3/libs -o /c/cocotb/cocotb-1.3.1/examples/dff/tests/build/libs/i686/libcocotb.dll /c/cocotb/cocotb-1.3.1/examples/dff/tests/build/obj/i686/gpi_embed.o   -lPython37 -lgpilog -lcocotbutils -L/c/cocotb/cocotb-1.3.1/examples/dff/tests/build/libs/i686
make[2]: Leaving directory '/c/Users/user1/miniconda3/lib/site-packages/cocotb/share/lib/embed'
make -C /c/Users/user1/miniconda3/lib/site-packages/cocotb/share/lib/gpi EXTRA_LIBS=-lmtipli EXTRA_LIBDIRS=-L/c/modelsim_dlx_2020.2/win32pe SIM=modelsim
make[2]: Entering directory '/c/Users/user1/miniconda3/lib/site-packages/cocotb/share/lib/gpi'
g++ -m32 -g -DDEBUG -shared -DVPI_CHECKING -DLIB_EXT=dll -DSINGLETON_HANDLES -DMODELSIM -c -I/c/Users/user1/miniconda3/lib/site-packages/cocotb/share/include -IC:\Users\user1\miniconda3\include  -o /c/cocotb/cocotb-1.3.1/examples/dff/tests/build/obj/i686/GpiCbHdl.o GpiCbHdl.cpp
g++ -m32 -g -DDEBUG -shared -DVPI_CHECKING -DLIB_EXT=dll -DSINGLETON_HANDLES -DMODELSIM -c -I/c/Users/user1/miniconda3/lib/site-packages/cocotb/share/include -IC:\Users\user1\miniconda3\include  -o /c/cocotb/cocotb-1.3.1/examples/dff/tests/build/obj/i686/GpiCommon.o GpiCommon.cpp
gcc -m32 -g -DDEBUG -shared -DMODELSIM -shared -Wl,-no-undefined -Wl,-enable-runtime-pseudo-reloc-v2 -Wl,--enable-auto-import -L/c/Users/user1/miniconda3/libs -o /c/cocotb/cocotb-1.3.1/examples/dff/tests/build/libs/i686/libgpi.dll  /c/cocotb/cocotb-1.3.1/examples/dff/tests/build/obj/i686/GpiCbHdl.o /c/cocotb/cocotb-1.3.1/examples/dff/tests/build/obj/i686/GpiCommon.o  -lcocotbutils -lgpilog -lcocotb -lstdc++ -L/c/cocotb/cocotb-1.3.1/examples/dff/tests/build/libs/i686
make[2]: Leaving directory '/c/Users/user1/miniconda3/lib/site-packages/cocotb/share/lib/gpi'
make -C /c/Users/user1/miniconda3/lib/site-packages/cocotb/share/lib/simulator SIM=modelsim
make[2]: Entering directory '/c/Users/user1/miniconda3/lib/site-packages/cocotb/share/lib/simulator'
gcc -m32 -g -DDEBUG -shared  -DMODELSIM -c -I/c/Users/user1/miniconda3/lib/site-packages/cocotb/share/include -IC:\Users\user1\miniconda3\include  -o /c/cocotb/cocotb-1.3.1/examples/dff/tests/build/obj/i686/simulatormodule.o simulatormodule.c
gcc -m32 -g -DDEBUG -shared  -DMODELSIM -shared -Wl,-no-undefined -Wl,-enable-runtime-pseudo-reloc-v2 -Wl,--enable-auto-import -L/c/Users/user1/miniconda3/libs -o /c/cocotb/cocotb-1.3.1/examples/dff/tests/build/libs/i686/libsim.dll /c/cocotb/cocotb-1.3.1/examples/dff/tests/build/obj/i686/simulatormodule.o   -lcocotbutils -lgpi -lgpilog -lPython37 -L/c/cocotb/cocotb-1.3.1/examples/dff/tests/build/libs/i686
ln -sf libsim.dll /c/cocotb/cocotb-1.3.1/examples/dff/tests/build/libs/i686/simulator.pyd
make[2]: Leaving directory '/c/Users/user1/miniconda3/lib/site-packages/cocotb/share/lib/simulator'
make -C /c/Users/user1/miniconda3/lib/site-packages/cocotb/share/lib/vpi EXTRA_LIBS=-lmtipli EXTRA_LIBDIRS=-L/c/modelsim_dlx_2020.2/win32pe SIM=modelsim
make[2]: Entering directory '/c/Users/user1/miniconda3/lib/site-packages/cocotb/share/lib/vpi'
g++ -m32 -g -DDEBUG -shared -DVPI_CHECKING -DMODELSIM -c -I/c/Users/user1/miniconda3/lib/site-packages/cocotb/share/include -IC:\Users\user1\miniconda3\include  -o /c/cocotb/cocotb-1.3.1/examples/dff/tests/build/obj/i686/VpiImpl.o VpiImpl.cpp
g++ -m32 -g -DDEBUG -shared -DVPI_CHECKING -DMODELSIM -c -I/c/Users/user1/miniconda3/lib/site-packages/cocotb/share/include -IC:\Users\user1\miniconda3\include  -o /c/cocotb/cocotb-1.3.1/examples/dff/tests/build/obj/i686/VpiCbHdl.o VpiCbHdl.cpp
[... cut ...]
# Errors: 1, Warnings: 0
# ** Error: C:/modelsim_dlx_2020.2/win32pe/vlog failed.
# Executing ONERROR command at macro ./runsim.do line 9
/c/Users/user1/miniconda3/lib/site-packages/cocotb/share/makefiles/simulators/Makefile.questa:147: recipe for target 'results.xml' failed
make[1]: *** [results.xml] Error 1
make[1]: Leaving directory '/c/cocotb/cocotb-1.3.1/examples/dff/tests'
C:\Users\user1\miniconda3\lib\site-packages\cocotb\share\makefiles/Makefile.sim:79: recipe for target 'sim' failed
make: *** [sim] Error 2

eric-wieser, 2020-06-11:
Fixed in #1795 I think.

ktbarrett, 2020-06-12:
I recommend installing the [current development version](https://docs.cocotb.org/en/latest/install_devel.html#install-devel). As Eric stated, the `getpid` issue is fixed upstream. `+acc=rmb` will also not work in 2020 version of Mentor products which has also been fixed recently (#1832 and #1908).

mm4dd, 2020-06-12:
Thanks eric-wieser and ktbarrett for the suggestions.

 I have installed the development version of cocotb and the simulation of dff has progressed a bit further. Now modelsim complains it cannot load the libcocotbvpi_modelsim.dll.  The output of the test run is below. 

(base) C:\cocotb\cocotb-1.4.0\examples\dff\tests>make SIM=modelsim TOPLEVEL_LANG=verilog GUI=1
make results.xml
make[1]: Entering directory '/c/cocotb/cocotb-1.4.0/examples/dff/tests'
set -o pipefail; MODULE=dff_cocotb TESTCASE= TOPLEVEL="work.dff" \
GPI_EXTRA= TOPLEVEL_LANG=verilog \
/c/modelsim_dlx_2020.2/win32pe/vsim -gui  -do sim_build/runsim.do  2>&1 | tee sim_build/sim.log

# Reading pref.tcl
# do sim_build/runsim.do
# Model Technology ModelSim DE vmap 2020.2 Lib Mapping Utility 2020.04 Apr 19 2020
# vmap -c 
# ** Warning: vmap will not overwrite local modelsim.ini.
# Model Technology ModelSim DE vmap 2020.2 Lib Mapping Utility 2020.04 Apr 19 2020
# vmap work sim_build/work 
# Modifying modelsim.ini
# Model Technology ModelSim DE vlog 2020.2 Compiler 2020.04 Apr 19 2020
# Start time: 07:31:07 on Jun 12,2020
# vlog -reportprogress 300 -work work "+define+COCOTB_SIM" -sv -timescale 1us/1us -mfcu "+acc" C:/cocotb/cocotb-1.4.0/examples/dff/hdl/dff.v 
# -- Compiling module dff
# 
# Top level modules:
# 	dff
# End time: 07:31:07 on Jun 12,2020, Elapsed time: 0:00:00
# Errors: 0, Warnings: 0
# vsim -onfinish exit -pli "C:/users/user1/miniconda3/lib/site-packages/cocotb/libs/libcocotbvpi_modelsim.dll" sim_build/work.dff 
# Start time: 07:31:07 on Jun 12,2020
# //  ModelSim DE 2020.2 Apr 19 2020
# //
# //  Copyright 1991-2020 Mentor Graphics Corporation
# //  All Rights Reserved.
# //
# //  ModelSim DE and its associated documentation contain trade
# //  secrets and commercial or financial information that are the property of
# //  Mentor Graphics Corporation and are privileged, confidential,
# //  and exempt from disclosure under the Freedom of Information Act,
# //  5 U.S.C. Section 552. Furthermore, this information
# //  is prohibited from disclosure under the Trade Secrets Act,
# //  18 U.S.C. Section 1905.
# //
# Loading sv_std.std
# Loading sim_build/work.dff
# Loading C:/users/user1/miniconda3/lib/site-packages/cocotb/libs/libcocotbvpi_modelsim.dll
# ** Error: (vsim-3193) Load of "C:/users/user1/miniconda3/lib/site-packages/cocotb/libs/libcocotbvpi_modelsim.dll" failed: DLL dependent library not found.
# ** Error (suppressible): (vsim-PLI-3002) Failed to load PLI object file "C:/users/user1/miniconda3/lib/site-packages/cocotb/libs/libcocotbvpi_modelsim.dll".
#    Time: 0 ps  Iteration: 0  Root: /
# Error loading design
# Error: Error loading design
#        Pausing macro execution
# MACRO ./sim_build/runsim.do PAUSED at line 10

ktbarrett, 2020-06-12:
Did you get all the [required dependencies](https://docs.cocotb.org/en/latest/install.html)?

```bash
conda install -c msys2 m2-base m2-make m2w64-toolchain libpython
```

There is a library that cocotb is depending upon that it can't find, since it can find `libcocotbvpi_modelsim.dll` I doubt it's any of the cocotb libraries that it can't find, which makes me think it's probably `libpython`.

If you did miss any of the dependencies, install those missing dependencies and then **reinstall** cocotb. Outside of that, I'm unsure.

mm4dd, 2020-06-12:
Hi ktbarrett, 

all the libraries are installed and I verified this by running the install command. 

(base) C:\Users\user1>conda install -c msys2 m2-base m2-make m2w64-toolchain libpython
Collecting package metadata (current_repodata.json): done
Solving environment: done

# All requested packages already installed.

(base) C:\Users\user1>

Thanks for the help with my issue in any event.

ktbarrett, 2020-06-12:
Someone else might be a better resource, I don't know much about running cocotb on Windows.

Windows support in cocotb has never been great, it's probably the hardest OS to support. You could look at running cocotb in Windows Subsystem for Linux?

mm4dd, 2020-06-12:
Thanks for your suggests ktbarrett. 

I am really keen to pick up cocotb for our FPGA development team. We are a small team with limited resources.  Cocotb seems to be able to do all the things that UVM (we develop our RTL with SystemVerilog) can do and it is open source.  

However, my experience in the last couple of days with miniconda3 on windows has not been good in terms of running time.  Running the cocotb on miniconda3 has been slow to slay the least and almost unusable from a simulation standpoint.  From what I have been reading the problem seems to be Anaconda and nothing to do with Cocotb.  I don't want to give up on cocotb yet as I have been reading about so many positive experiences using cocotb for doing testing.  Is there an open source distribution of Linux that anyone could recommend that would work with Modelsim? 

Thanks in advance.

ktbarrett, 2020-06-12:
We use RHEL at work with Questa, CentOS is basically a free copy of RHEL so it should be supported as well. I can't seem to find any useful public information about Linux support with ModelSim.

themperek, 2020-06-12:
So Modelsim 32-bit should work on windows.  I cross-check not so long ago, Can you list your `C:/users/user1/miniconda3/lib/site-packages/cocotb/libs` folder.

mm4dd, 2020-06-13:
Hi ktbarrett and themperek.

ktbarrett thanks for looking into the open source linux support for linux. I had a look last night as well and there seems to be many attempts to use the free version of modelsim from Intel but I did not see anyone using a paid subscription version on open source linux like CentOS.  

themperek I have reinstalled all my miniconda environment, reinstalled the cocotb development stream and rerun the test but I still get the same failure.  My libs folder contents are below: 

`(base) C:\cocotb\cocotb-1.4.0\examples\dff\tests>ls C:/users/user1/miniconda3/lib/site-packages/cocotb/libs/
libcocotb.dll              libcocotbutils.dll       libcocotbvpi_aldec.dll  libcocotbvpi_icarus.vpl    libgpi.dll
libcocotbfli_modelsim.dll  libcocotbvhpi_aldec.dll  libcocotbvpi_ghdl.dll   libcocotbvpi_modelsim.dll  libgpilog.dll`

themperek, 2020-06-13:
I have just check cocotb master + miniconda + 32bit + questa 10.7d and it works for also Modelsim Altera 10.1b works for me.
You can try MSYS2 (https://github.com/cocotb/cocotb/wiki/Tier-2-Setup-Instructions#windows-installation-using-msys2).

elgorwi, 2020-06-13:
> I have just check cocotb master + miniconda + 32bit + questa 10.7d and it works for also Modelsim Altera 10.1b works for me.
> You can try MSYS2 (https://github.com/cocotb/cocotb/wiki/Tier-2-Setup-Instructions#windows-installation-using-msys2).

For Msys2, it's maybe worth mentioning to run the tests from the mingw32 shell.

mm4dd, 2020-06-13:
Hi themperek and elgrowi.

Up to now I have used the recommended installation method which is via miniconda3.  I will try the other method mentioned using mysys2. I will let you know if I get any further.  Thanks for your assistance.

mm4dd, 2020-06-13:
I have installed MSYS2 and the related development packages needed to run cocotb. I have tried running a test on MYSYS 32 shell. Unfortunately, vsim cannot be found. Output below:
`mm@DESKTOP-MM /c/cocotb/cocotb-1.4.0/examples/dff/tests
$ make SIM=modelsim TOPLEVEL_LANG=verilog
C:/msys64/mingw32/lib/python3.8/site-packages/cocotb/share/makefiles/simulators/Makefile.questa:42: *** "Unable to locate command >vsim<".  Stop.
`

elgorwi, 2020-06-13:
Try with the mingw32 shell instead. If that doesn’t solve the path issue, there might be a setting that you have to uncomment in the shell.cmd file to inherit the path env variable.

mm4dd, 2020-06-13:
Hi elgorwi.  

I am running the MINGW32 shell as you originally suggested.  I have found the mmsys2_shell.cmd file but I am unsure what I need to do to sort out my path issue.

[msys2_shell.txt](https://github.com/cocotb/cocotb/files/4774614/msys2_shell.txt)

I have tried uncommenting the following line:   set MSYS2_PATH_TYPE=inherit
and restarted the shell but it did not help.

elgorwi, 2020-06-13:
Make sure that modelsim is installed in a location without spaces ie. not in program files. 
Then ensure that vsim is added to the path env var. 
If ```which vsim``` from a mingw32 shell does not return the expected path after removing the rem from the startup script try rebooting.

mm4dd, 2020-06-13:
The pc reboot sorted it. I can now see vsim on the mingw32 shell. 

Running the test dff test with  Modelsim PE/DE 2020.2 does not work.  There definitely seems to be some issue when running the latest version of Modelsm.  However, I reverted to an older version Modelsim PE 10.3e and the test successfully runs.  

```
$  make SIM=modelsim TOPLEVEL_LANG=verilog
make results.xml
make[1]: Entering directory '/c/cocotb/cocotb-1.4.0/examples/dff/tests'
set -o pipefail; MODULE=dff_cocotb TESTCASE= TOPLEVEL="work.dff" \
GPI_EXTRA= TOPLEVEL_LANG=verilog \
/c/modeltech_pe_10.3e/win32pe/vsim -c  -do sim_build/runsim.do  2>&1 | tee sim_build/sim.log

Reading C:/modeltech_pe_10.3e/tcl/vsim/pref.tcl

# 10.3e

# do sim_build/runsim.do
# Model Technology ModelSim PE vmap 10.3e Lib Mapping Utility 2015.02 Feb 24 2015
# vmap -modelsim_quiet -c
# ** Warning: vmap will not overwrite local modelsim.ini.
# Model Technology ModelSim PE vmap 10.3e Lib Mapping Utility 2015.02 Feb 24 2015
# vmap -modelsim_quiet work sim_build/work
# Modifying modelsim.ini
# Model Technology ModelSim PE vlog 10.3e Compiler 2015.02 Feb 24 2015
# Start time: 14:44:24 on Jun 13,2020
# vlog -work work "+define+COCOTB_SIM" -sv -timescale 1us/1us -mfcu "+acc" C:/cocotb/cocotb-1.4.0/examples/dff/hdl/dff.v
# -- Compiling module dff
#
# Top level modules:
#       dff
# End time: 14:44:24 on Jun 13,2020, Elapsed time: 0:00:00
# Errors: 0, Warnings: 0
# vsim -c -do "sim_build/runsim.do"
# Start time: 14:44:25 on Jun 13,2020
# Loading C:/msys64/mingw32/lib/python3.8/site-packages/cocotb/libs/libcocotbvpi_modelsim.dll
#      -.--ns ERROR    cocotb.gpi                         ..s/cocotb_utils.cpp:57   in utils_dyn_open                  Unable to open lib python3.8.dll.dll: The specified module could not be found.
#
#
#
#
#
#
#      -.--ns ERROR    cocotb.gpi                         ..mbed/gpi_embed.cpp:127  in embed_init_python               Failed to find Python shared library
#
#
#
#      -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:74   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
#
#      -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:105  in gpi_print_registered_impl       VPI registered
# //  ModelSim PE 10.3e Feb 24 2015
# //
# //  Copyright 1991-2015 Mentor Graphics Corporation
# //  All Rights Reserved.
# //
# //  THIS WORK CONTAINS TRADE SECRET AND PROPRIETARY INFORMATION
# //  WHICH IS THE PROPERTY OF MENTOR GRAPHICS CORPORATION OR ITS
# //  LICENSORS AND IS SUBJECT TO LICENSE TERMS.
# //  THIS DOCUMENT CONTAINS TRADE SECRETS AND COMMERCIAL OR FINANCIAL
# //  INFORMATION THAT ARE PRIVILEGED, CONFIDENTIAL, AND EXEMPT FROM
# //  DISCLOSURE UNDER THE FREEDOM OF INFORMATION ACT, 5 U.S.C. SECTION 552.
# //  FURTHERMORE, THIS INFORMATION IS PROHIBITED FROM DISCLOSURE UNDER
# //  THE TRADE SECRETS ACT, 18 U.S.C. SECTION 1905.
# //
# Loading sv_std.std
# Loading sim_build/work.dff
#      -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:244  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
#      0.00ns INFO     cocotb                                      __init__.py:189  in _initialise_testbench           Running on ModelSim PE version 10.3e 2015.02
#
#      0.00ns INFO     cocotb                                      __init__.py:195  in _initialise_testbench           Running tests with cocotb v1.4.0.dev0 from C:/msys64/mingw32/lib/python3.8/site-packages/cocotb
#
[... 34 lines cut ...]
#                                                                                                                       **                                 ERRORS : 0                                      **
#
#                                                                                                                       *************************************************************************************
#
#                                                                                                                       **                               SIM TIME : 1005000.00 NS                          **
#
#                                                                                                                       **                              REAL TIME : 0.11 S                                 **
#
#                                                                                                                       **                        SIM / REAL TIME : 9059575.49 NS/S                        **
#
#                                                                                                                       *************************************************************************************
#
#
#
# 1005000.00ns INFO     cocotb.regression                         regression.py:255  in tear_down                       Shutting down...
#
# ** Note: $finish
#    Time: 1005000001 ps  Iteration: 0  Instance: /dff
# End time: 14:44:25 on Jun 13,2020, Elapsed time: 0:00:00
# Errors: 0, Warnings: 0
 ```

**_Thanks so much to everyone for all their help!!  (elgorwi, themperek, ktbarrett, eric-wieser)_** 

I am in the process of evaluating a migration from Modelsim PE to DE. I did noticed DE has both windows and Linux support. In terms of run time does anyone have an opinion on faster run times on Windows or Linux?

ktbarrett, 2020-06-13:
@mm4dd I have also noticed issues with Questa 2019 and 2020 that are not present in 10.7c and earlier. I already put a few PRs in, but it seems I have more work to do. Could you post your failing run?

mm4dd, 2020-06-14:
@ktbarrett  Here is my failing run of the dff example.   

```
$ make SIM=modelsim TOPLEVEL_LANG=verilog
make results.xml
make[1]: Entering directory '/c/cocotb/cocotb-1.4.0/examples/dff/tests'
set -o pipefail; MODULE=dff_cocotb TESTCASE= TOPLEVEL="work.dff" \
GPI_EXTRA= TOPLEVEL_LANG=verilog \
/c/modelsim_dlx_2020.2/win32pe/vsim -c  -do sim_build/runsim.do  2>&1 | tee sim_build/sim.log
Reading pref.tcl

# 2020.2

# do sim_build/runsim.do
# Model Technology ModelSim DE vmap 2020.2 Lib Mapping Utility 2020.04 Apr 19 2020
# vmap -c
# ** Warning: vmap will not overwrite local modelsim.ini.
# Model Technology ModelSim DE vmap 2020.2 Lib Mapping Utility 2020.04 Apr 19 2020
# vmap work sim_build/work
# Modifying modelsim.ini
# Model Technology ModelSim DE vlog 2020.2 Compiler 2020.04 Apr 19 2020
# Start time: 10:09:52 on Jun 14,2020
# vlog -work work "+define+COCOTB_SIM" -sv -timescale 1us/1us -mfcu "+acc" C:/cocotb/cocotb-1.4.0/examples/dff/hdl/dff.v
# -- Compiling module dff
#
# Top level modules:
#       dff
# End time: 10:09:52 on Jun 14,2020, Elapsed time: 0:00:00
# Errors: 0, Warnings: 0

# vsim -onfinish exit -pli "C:/msys64/mingw32/lib/python3.8/site-packages/cocotb/libs/libcocotbvpi_modelsim.dll" sim_build/work.dff
# Start time: 10:09:52 on Jun 14,2020
# //  ModelSim DE 2020.2 Apr 19 2020
# //
# //  Copyright 1991-2020 Mentor Graphics Corporation
# //  All Rights Reserved.
# //
# //  ModelSim DE and its associated documentation contain trade
# //  secrets and commercial or financial information that are the property of
# //  Mentor Graphics Corporation and are privileged, confidential,
# //  and exempt from disclosure under the Freedom of Information Act,
# //  5 U.S.C. Section 552. Furthermore, this information
# //  is prohibited from disclosure under the Trade Secrets Act,
# //  18 U.S.C. Section 1905.
# //
# Loading sv_std.std
# Loading sim_build/work.dff
# Loading C:/msys64/mingw32/lib/python3.8/site-packages/cocotb/libs/libcocotbvpi_modelsim.dll
# ** Error: (vsim-3193) Load of "C:/msys64/mingw32/lib/python3.8/site-packages/cocotb/libs/libcocotbvpi_modelsim.dll" failed: DLL dependent library not found.
# ** Error (suppressible): (vsim-PLI-3002) Failed to load PLI object file "C:/msys64/mingw32/lib/python3.8/site-packages/cocotb/libs/libcocotbvpi_modelsim.dll".
#    Time: 0 ps  Iteration: 0  Root: /
# Error loading design
Error loading design
# End time: 10:17:51 on Jun 14,2020, Elapsed time: 0:07:59
# Errors: 2, Warnings: 0
make[1]: *** [C:/msys64/mingw32/lib/python3.8/site-packages/cocotb/share/makefiles/simulators/Makefile.questa:133: results.xml] Error 12
make[1]: Leaving directory '/c/cocotb/cocotb-1.4.0/examples/dff/tests'
make: *** [C:/msys64/mingw32/lib/python3.8/site-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2

```
The miniconda3 environment was a no go for me on windows but the msys2 environment seems to work well. 

Do you want me to close my issue since I have gotten cocotb working with an older version of Modelsim or leave it open for the moment?

closed 2020-06-16 by mm4dd