cocotb/cocotb #1927: Compilation Error with 32 bit python and Modelsim-altera

RuiqiW, 2020-06-11:
Hello

I have installed 32-bit version miniconda (python 3.7 on Linux), and cocotb with the 32-bit conda. However, when I ran "make" for the cocotb (1.3.1) examples, the code still compiled with 64-bit gcc library. 
I have installed ubuntu i386 library (32-bit). I tried to add "-m32" flag in the cocotb/share/makefiles/Makefile.sim file, but the compilation still failed with the error messages as below.
I am wondering if there is a way to change the default gcc library for compilation. Thanks in advance.

`/usr/bin/ld: skipping incompatible /home/ruiqi/cocotb-1.3.1/adder/tests/build/libs/x86_64/libcocotbutils.so when searching for -lcocotbutils
/usr/bin/ld: cannot find -lcocotbutils
collect2: error: ld returned 1 exit status
/home/ruiqi/Applications/miniconda3/lib/python3.7/site-packages/cocotb/share/makefiles/Makefile.rules:63: recipe for target '/home/ruiqi/cocotb-1.3.1/adder/tests/build/libs/x86_64/libcocotb.so' failed
make[2]: *** [/home/ruiqi/cocotb-1.3.1/adder/tests/build/libs/x86_64/libcocotb.so] Error 1
make[2]: Leaving directory '/home/ruiqi/Applications/miniconda3/lib/python3.7/site-packages/cocotb/share/lib/embed'
/home/ruiqi/Applications/miniconda3/lib/python3.7/site-packages/cocotb/share/lib/Makefile:46: recipe for target '/home/ruiqi/cocotb-1.3.1/adder/tests/build/libs/x86_64/libcocotb.so' failed
make[1]: *** [/home/ruiqi/cocotb-1.3.1/adder/tests/build/libs/x86_64/libcocotb.so] Error 2
make[1]: Leaving directory '/home/ruiqi/cocotb-1.3.1/adder/tests'
/home/ruiqi/Applications/miniconda3/lib/python3.7/site-packages/cocotb/share/makefiles/Makefile.sim:79: recipe for target 'sim' failed
make: *** [sim] Error 2
`

ktbarrett, 2020-06-12:
> I tried to add "-m32" flag in the cocotb/share/makefiles/Makefile.sim file

To what in Makefile.sim? Could you post the entire log file? You can capture the output with `tee`:

```bash
make SIM=icarus | tee cmd.log
```

Then you can upload the `cmd.log` file by selecting the lower bar in the reply window that starts "Attach files dragging ..."

ktbarrett, 2020-06-12:
I'll also recommend you just [install the current development version](https://docs.cocotb.org/en/latest/install_devel.html#install-devel), the installation process is a lot different and this really shouldn't be an issue for you.

RuiqiW, 2020-06-12:
> To what in Makefile.sim? Could you post the entire log file? You can capture the output with `tee`:
Sorry, I tried adding "-m32" after GXX_ARGS (line 92) in Makefile.inc, and I got the ouput:
[cmd.log](https://github.com/cocotb/cocotb/files/4771604/cmd.log)

ktbarrett, 2020-06-12:
Are you changing that line in the installed makefiles or the local makefiles? And in your project are you using the installed makefiles or the local makefiles? You need to modify the set of makefiles you project is using. Again, I'll plug the [dev version](https://docs.cocotb.org/en/latest/install_devel.html#install-devel) which should have none of these issues.

themperek, 2020-06-12:
I suggest using development version and follow instructions here: https://github.com/cocotb/cocotb/wiki/Tier-2-Setup-Instructions#32-bit-python

closed 2020-11-24 by github-actions[bot]