cocotb/cocotb #1930: Make coverage deterministic

ktbarrett, 2020-06-13:
Every PR is showing slight fluctuations in coverage, likely due to random test data. I find this irritating. We should try to design tests that cover all branches of the thing we are testing. Fixing the seed would also work, but I think is a bad idea.

imphil, 2020-06-22:
> Fixing the seed would also work, but I think that's a bad idea.

Deterministic tests are important for CI runs, and fixing the seed is a absolutely valid way of doing that.

ktbarrett, 2020-06-23:
If that's the case, we probably should test with multiple fixed seeds.

ktbarrett, 2020-06-23:
That might mask errors if we also don't write our tests for 100% coverage.

ktbarrett, 2021-02-16:
It seems there were many issues with coverage collection that are now fixed (#2397).

closed 2021-02-16 by ktbarrett