cocotb/cocotb #1933: issue with importing cocotb

kirknodeng, 2020-06-16:
Hi, all

Since 2017 or 2018, I used cocotb v1.0 version. That time cocotb installation was simple. 
Just copy and define evironment variables.. 

These day, I want to update cocotb v1.4 dev0. It's little changed.. cocotb can update with pip and go into the "site-packages".. 

In my case, I can't access our system.. So, I can't install and access GNU tools.. (It's connected with only intranet)

This is my server information.. 
Linux <SERVERNAME> 2.6.32-696.el6.x86_64 #1 SMP Tue Feb 21 00:53:17 EST 2017 x86_64 x86_64 x86_64 GNU/Linux

When I install cocotb, after then I execute "cocotb-config". I show message below. 

Traceback (most recent call last):
  File "cocotb-config", line 11, in <module>
    load_entry_point('cocotb==1.4.0.dev0', 'console_scripts', 'cocotb-config')()
  File "/xxx/Python-3.6.4/lib/python3.6/site-packages/pkg_resources/__init__.py", line 565, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/xxx/Python-3.6.4/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2631, in load_entry_point
    return ep.load()
  File "/xxx/Python-3.6.4/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2291, in load
    return self.resolve()
  File "/xxx/Python-3.6.4/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2297, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/xxx/Python-3.6.4/lib/python3.6/site-packages/cocotb/__init__.py", line 43, in <module>
    import cocotb.handle
  File "/xxx/Python-3.6.4/lib/python3.6/site-packages/cocotb/handle.py", line 36, in <module>
    from cocotb import simulator
ImportError: /usr/lib64/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /xxx/Python-3.6.4/lib/python3.6/site-packages/cocotb/libs/libgpi.so)

As I mentioned above, GLIBCXX_3.4.21 can't install in our system.. (a lot of dependency issues.. )

How can I use this version of cocotb? Is it possible to use with environment variables define? (like cocotb v1.0) 

Thank you,

ktbarrett, 2020-06-16:
cocotb no longer supports old toolchains; the C++ codebase is now taking advantage of C++11 features.

If you could install cocotb, you likely have a C++11 compiler and standard library. When you run python, an older libstdc++ is being loaded instead of the one you compiled the cocotb libraries with. This seems to be more of an an issue with your environment. Could you find a newer version of libstdc++ on your system (presumably the one that you compiled cocotb with) and preload it?

```bash
export LD_PRELOAD=/path/to/newer/libstdc++.so
```

I've had problem similar to this running Questa where it puts it's packaged out-of-date versions of GCC in front of the newer system ones.

kirknodeng, 2020-06-16:
@ktbarrett  

I try to load using LD_PRELOAD. 
It works!

Thank you,

closed 2020-06-16 by kirknodeng