cocotb/cocotb #1934: How to deal with VHDL records

Botnic, 2020-06-18:
Hi

I have some interface which I access with the BusDriver class (class Axi4StreamMaster(BusDriver) from https://github.com/corna/cocotbext-axi4stream @corna )

This works find. But now I would like to wrap my interface into two records:
```

 type Axi4s_M2S is record              -- master to slave direction
    Tvalid : std_logic;
    Tdata  : std_logic_vector;
    Tstrb  : std_logic_vector;
    Tkeep  : std_logic_vector;
    Tlast  : std_logic;
    Tid    : std_logic_vector;
    Tdest  : std_logic_vector;
    Tuser  : std_logic_vector;
  end record Axi4s_M2S;

  type Axi4s_S2M is record              -- slave to master direction
    Tready : std_logic;
  end record Axi4s_S2M;`
```

I was able to access the signals by simply writing:
`dut.myInterface_M2S.Tvalid`

Now I could rewrite the complete driver to match this syntax. But this would break compatibility and would also not allow to use different drivers. I'm sure there is a better option.
But what would be the best option to combine this with the driver class?

My idea was to make a renamed copy of the signal in the dut, where the driver can connect to. For example:
`dut.myInterface_M2S_Tvalid = dut.myInterface_M2S.Tvalid`

I did not figure out what the right syntax would be to make such a connection. 
Thank you for your help.

ktbarrett, 2020-06-18:
> My idea was to make a renamed copy of the signal in the dut, where the driver can connect to. For example:
dut.myInterface_M2S_Tvalid = dut.myInterface_M2S.Tvalid

That is not possible. 

There is currently no support in `BusDriver` or `AXIStreamMaster` for this at the moment. 

Here: https://github.com/cocotb/cocotb/blob/cdfee3694474a0e4daa4d73d85c17f3abe483fc6/cocotb/drivers/__init__.py#L247-L250

We see the `Bus` object being created in `BusDriver`, which would ultimately end up being used by the `AXI4StreamMaster`. `Bus` can take a dictionary of names to signals, which `BusDriver` is passing in as `self._signals`. I can't seem to see where this attribute is set? But if you make changes to `BusDriver` and `AXI4StreamDriver` so you can pass a bundle of signals in, you should be able to support that.

Botnic, 2020-06-19:
Hi @ktbarrett 
Thank you for your response. I made an external wrapper to unpack the Records and will start with that. Once I have more experience, I may try to write an overloaded function.

closed 2020-11-24 by github-actions[bot]