cocotb/cocotb #1938: make SIM=verilator issue

Divyanirankari, 2020-06-20:
while running cocotb/examples/adder/tests getting verilator error:
```
(py36) divya@divya:~/cocotb/examples/adder/tests$ make SIM=verilator
make results.xml
make[1]: Entering directory '/home/divya/cocotb/examples/adder/tests'
verilator -cc --exe -Mdir sim_build -DCOCOTB_SIM=1 --top-module adder --vpi --public-flat-rw --prefix Vtop -o adder -LDFLAGS "-Wl,-rpath,/home/divya/cocotb/cocotb/libs -L/home/divya/cocotb/cocotb/libs -lcocotbvpi_verilator -lgpi -lcocotb -lgpilog -lcocotbutils"  /home/divya/cocotb/examples/adder/tests/../hdl/adder.v /home/divya/cocotb/cocotb/share/lib/verilator/verilator.cpp
%Error: Invalid Option: --public-flat-rw
%Error: Command Failed /opt/verilator-4/bin/verilator_bin -cc --exe -Mdir sim_build '-DCOCOTB_SIM=1' --top-module adder --vpi --public-flat-rw --prefix Vtop -o adder -LDFLAGS '-Wl,-rpath,/home/divya/cocotb/cocotb/libs -L/home/divya/cocotb/cocotb/libs -lcocotbvpi_verilator -lgpi -lcocotb -lgpilog -lcocotbutils' /home/divya/cocotb/examples/adder/tests/../hdl/adder.v /home/divya/cocotb/cocotb/share/lib/verilator/verilator.cpp
/home/divya/cocotb/cocotb/share/makefiles/simulators/Makefile.verilator:42: recipe for target 'sim_build/Vtop.mk' failed
make[1]: *** [sim_build/Vtop.mk] Error 10
make[1]: Leaving directory '/home/divya/cocotb/examples/adder/tests'
/home/divya/cocotb/cocotb/share/makefiles/Makefile.inc:39: recipe for target 'sim' failed
make: *** [sim] Error 2
```

closed 2020-06-20 by Divyanirankari

Amithkagga, 2021-03-04:
How did you solve this issue

Divyanirankari, 2021-03-04:
@Amithkagga Its a verilator issue,I installed the latest version of verilator.