cocotb/cocotb #1943: Hard-coded RPATH in libraries

mtikekar, 2020-06-23:
When I run `pip install cocotb==1.4.0rc1` in one conda environment, it builds and installs a wheel and saves the wheel in a cache. Now, when I run the same command in a new conda environment, it reuses the wheel. The problem is that the `.so` files have a hard-coded RPATH to the first environment, so when that env is deleted, cocotb breaks in the other env.

From what I can tell, conda solves this problem for its packages: https://docs.conda.io/projects/conda-build/en/latest/resources/make-relocatable.html

themperek, 2020-06-24:
What OS is this? What is the error?

mtikekar, 2020-06-24:
This was on Linux. The error message was something like libpython3.7m.so not found followed by the simulator (xcelium) crashing.

eric-wieser, 2020-06-24:
Can you give an example of the hard-coded rpath?

themperek, 2020-06-24:
Is this?
https://github.com/cocotb/cocotb/blob/c2acc9f4608660271f7035a178eb0cc9273ce065/cocotb_build_libs.py#L217

Can we maybe use `PYTHONHOME` to load this?
Block making the wheel package?

mtikekar, 2020-06-24:
> Can you give an example of the hard-coded rpath?

The libcocotb.so file has rpath that looks something like: `/home/mtikekar/.conda/envs/envname/lib:$ORIGIN`.

> Can we maybe use `PYTHONHOME` to load this?

I built a conda package for cocotb to see what they do. I ran:

```
conda skeleton pypi cocotb
conda build cocotb
```

It basically runs pip and then goes through each executable and replaces any hard-coded rpaths with relative paths using `$ORIGIN`. For libcocotb.so in the conda package, rpath is `$ORIGIN/../../../..:$ORIGIN`.

eric-wieser, 2020-06-24:
@themperek's point about `PYTHON_SO_LIB` seems salient too - our binary contains a string constant that presumably has a bad path as well.

themperek, 2020-06-24:
So I have this:
```
building 'cocotb/libs/libcocotb' extension
    creating build/temp.linux-x86_64-3.7/cocotb/share/lib/embed
    gcc -pthread -B /home/themperek/miniconda/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -DPYTHON_SO_LIB=libpython3.7m.so -Icocotb/share/include -I/home/themperek/miniconda/include/python3.7m -c cocotb/share/lib/embed/gpi_embed.cpp -o build/temp.linux-x86_64-3.7/cocotb/share/lib/embed/gpi_embed.o -std=c++11 -Wall -Wextra -Wcast-qual -Wwrite-strings -Wconversion -Wnon-virtual-dtor -Woverloaded-virtual -D__STDC_FORMAT_MACROS
    g++ -pthread -shared -B /home/themperek/miniconda/compiler_compat -L/home/themperek/miniconda/lib -Wl,-rpath=/home/themperek/miniconda/lib -Wl,--no-as-needed -Wl,--sysroot=/ build/temp.linux-x86_64-3.7/cocotb/share/lib/embed/gpi_embed.o -Lbuild/lib.linux-x86_64-3.7/cocotb/libs -Lbuild/lib.linux-x86_64-3.7/cocotb/libs -lpython3.7m -lgpilog -lcocotbutils -o build/lib.linux-x86_64-3.7/cocotb/libs/libcocotb.so -Wl,-rpath,$ORIGIN
```
We have ` -Wl,-rpath=/home/themperek/miniconda/lib`  and `-lpython3.7m` .

I probably do not get something but why the position of `libpython` would change?

mtikekar, 2020-06-24:
Because I am installing in multiple conda environments. pip caches the wheel when I install in the first environment. So if I delete that environment, the wheel is broken when pip reuses it in the next environment.

eric-wieser, 2020-06-24:
Does each conda environment have its own copy of `libpython`?

mtikekar, 2020-06-24:
Yes. They use symlinks to avoid keeping multiple copies of the same `libpython` on the disk, but we can basically treat each conda environment as a standalone python distribution.

themperek, 2020-06-24:
So some solution would be to add `$ORIGIN/../../../..` to `-rpath`? Not sure if this is great.

eric-wieser, 2020-06-24:
Yeah, I'm not sure we can make any guarantees about the relative paths between our libraries and libpython.

eric-wieser, 2020-06-24:
I assume only `libcocotb` has the problem?

themperek, 2020-06-24:
One can do `conda clean` but ...

mtikekar, 2020-06-24:
> I assume only `libcocotb` has the problem?

All the libraries get that rpath as they are all built by setuptools/distutils which is where the rpath flag is coming from. But only `libcocotb` is really affected as it has a `-libpython`

> One can do `conda clean` but ...

Might not help. pip cache is typically in ~/.cache/pip.

themperek, 2020-06-24:
One can also try to set `LD_LIBRARY_PATH`/'PATH' that we try to avoid ...

eric-wieser, 2020-06-24:
> All the libraries get that rpath as they are all built by setuptools/distutils which is where the rpath flag is coming from. 

If this is the case, then presumably this affects all python packages with extension modules, not just ours.

themperek, 2020-06-24:
So the problem is when you create an environment install cocotb than install in another environment and remove the first one?

themperek, 2020-06-24:
And if we suggest `pip cache purge` before install?

themperek, 2020-06-24:
Maybe we can add to `setup.cfg`:

```
[build_ext]
force=1
```

eric-wieser, 2020-06-24:
I think before we do anything here we need to understand whether this problem is ours or conda's.

mtikekar, 2020-06-24:
Yes, all extension modules built with setuptools/distutils will have this rpath. The difference is that they are being loaded by `python` which has already loaded `libpython`. So these extension modules don't really care for having the right rpath. 

You can check by building some simple extension modules like https://github.com/starnight/python-c-extension. This is the compile command it runs:

```
gcc -pthread -shared -B /home/mtikekar/.conda/envs/foo/compiler_compat -L/home/mtikekar/.conda/envs/foo/lib \
-Wl,-rpath=/home/mtikekar/.conda/envs/foo/lib -Wl,--no-as-needed -Wl,--sysroot=/ build/temp.linux-x86_64-3.7/bind.o \
build/temp.linux-x86_64-3.7/libmypy.o -o build/lib.linux-x86_64-3.7/helloworld.cpython-37m-x86_64-linux-gnu.so
```

You can see the `-Wl,-rpath` but no `-lpython`.

For cocotb, `libcocotb` is loaded by the simulator which then has to load `libpython`. That is source of the problem here.

ktbarrett, 2020-06-26:
> All the libraries get that rpath as they are all built by setuptools/distutils which is where the rpath flag is coming from.

So setuptools/distutils is making crappy assumptions and we suffer. If we moved the build over to build_clib I'm guessing (hoping) that this wouldn't be an issue, would it? Or maybe a proper and well documented build system...

eric-wieser, 2020-06-26:
> So setuptools/distutils is making assumptions and we suffer.

From a different perspective: users make assumptions that conda is compatible with setuptools, but the reality is they are not always. Unfortunately, it's partly our fault that users assume that, because our recommended install process on windows is via conda!

mtikekar, 2020-06-26:
Another option is to distribute conda packages via conda-forge. I haven't done it before, but the process seems simple - https://conda-forge.org/#add_recipe. Their CI service will build Windows, Linux and macOS packages. And they have a [package](https://anaconda.org/conda-forge/iverilog) for Icarus Verilog, so testing the package should be easy.

If this is of interest, I'd be happy to take a shot at it.

ktbarrett, 2020-06-26:
@eric-wieser 
> users make assumptions that conda is compatible with setuptools

I assumed that for a while. It's somewhat deceiving because it does work 98% of the time.

> An another one: conda assumes that the link to libpython.so doesn't matter, and extension modules will always be loaded by a process already running python.

I feel like I've run into that recently somewhere else as well. Conda doesn't seem to take that into consideration. 

Still I think that setuptool's is the poor behavior. I assume it was added to be a bit of convenience for people embedding Python interpreters rather than making extension modules. You don't want to depend on the user's environment being correct, so you assume that you can point to the libpython used to build the package. That's fine until you are confronted the current issue, or worse, binary distribution (maybe bdist handles this? I doubt it.).

ktbarrett, 2020-06-26:
Call me a heretic if you like, but we can install libpython with cocotb and use a relative RPATH 😝.

ktbarrett, 2020-06-27:
#### This isn't just an issue with conda, it affects pip because the built package is stored in pip's wheel cache.

Confirmed with the following:

```shell
> conda create -n test python=3.7
> conda activate test
> pip install cocotb==1.4.0rc1
> conda deactivate
> conda create -n test2 python=3.7  # must be the same version of python to invoke the error
> conda activate test2
> pip install cocotb==1.4.0rc1  # must be same cocotb version
> conda env remove -n test
> make
```

Results:
```
make results.xml
make[1]: Entering directory '/home/kaleb/dev/cocotb/tests/test_cases/test_cocotb'
MODULE=" test_deprecated, test_doctests, test_synchronization_primitives, test_concurrency_primitives, test_tests, test_generator_coroutines, test_timing_triggers, test_scheduler, test_clock, test_edge_triggers, test_async_coroutines, test_handle, test_logging, " TESTCASE= TOPLEVEL=sample_module TOPLEVEL_LANG=verilog \
        /usr/bin/vvp -M /home/kaleb/miniconda3/envs/py37/lib/python3.7/site-packages/cocotb/libs -m libcocotbvpi_icarus   sim_build/sim.vvp 
libcocotbvpi_icarus:`/home/kaleb/miniconda3/envs/py37/lib/python3.7/site-packages/cocotb/libs/libcocotbvpi_icarus.vpl' failed to open using dlopen() because:
    libpython3.7m.so.1.0: cannot open shared object file: No such file or directory.
VCD info: dumpfile waveform.vcd opened for output.
ERROR: results.xml was not written by the simulation!
make[1]: *** [/home/kaleb/miniconda3/envs/py37/lib/python3.7/site-packages/cocotb/share/makefiles/simulators/Makefile.icarus:69: results.xml] Error 1
make[1]: Leaving directory '/home/kaleb/dev/cocotb/tests/test_cases/test_cocotb'
make: *** [/home/kaleb/miniconda3/envs/py37/lib/python3.7/site-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2
```

eric-wieser, 2020-06-27:
Pip's cache presumably assumes that one cache path serves at most one python installation (eg, location of libpython).

Conda violates this assumption.

ktbarrett, 2020-06-27:
Binary packaging would also violate this assumption. We can't be sure where the libpython on the end user's system is. And we shouldn't ship packages with meaningless RPATHs. RPATH is *not* the right answer for sourcing libpython into the library load path.

IMO libpython should be treated as a system library in this context (and really every context). Overriding system libraries is done with `LD_LIBRARY_PATH`. If that's too distasteful we could always ship libpython with our binary packages, like most applications do.

ktbarrett, 2020-06-27:
I opened a [discussion on pypa's discourse](https://discuss.python.org/t/packaging-interpreter-embedding-frameworks/4577) to maybe get some more experienced eyes on the problem.

eric-wieser, 2020-06-27:
@mtikekar, does the issue persist with python 3.8?

ktbarrett, 2020-06-27:
Another possible solutions is:
* never support binary distribution with pip
* only use conda for binary distribution
* recommend using the conda package and not pip if you are going to use conda to manager your environments

I guess that's what @mtikekar was recommending.

ktbarrett, 2020-06-27:
@eric-wieser Tested with Python 3.8, the RPATH issue is still present.

themperek, 2020-06-27:
We can try to set LD_LIBRARY_PATH at runtime?

eric-wieser, 2020-06-27:
> @eric-wieser Tested with Python 3.8, the RPATH issue is still present.
> 
> I spoke too soon, the RPATH is the same, but it's not failing... Ahh, `ldd` to the rescue, it's loading my system Python 3.8 library, it would still otherwise be broken.

Is your system Darwin or Linux? I thought the conclusion was that distutils does not add an rpath in 3.8?

ktbarrett, 2020-06-27:
@eric-wieser 
Debian Sid. I think it's self-inflicted, we are using `library_dir` and paths to user libraries are added to the RPATH by distutils (I think I was reading that code correctly).

@themperek 
I feel like that is the only reasonable choice. Interestingly, see this [answer](https://stackoverflow.com/a/24115039/6614127). If this is going to work we need to stop linking libcocotb against libpython and instead go with the dynamic loading solution already present in gpi_embed.

mtikekar, 2020-06-28:
@eric-wieser This problem persists on python3.8 even after I remove python lib dir from the link flags in https://github.com/cocotb/cocotb/blob/8de3acedcb4bb1e5ca3726ec75f034c1e7fe7712/cocotb_build_libs.py#L222

I fairly certain that conda's python build is configured to add the rpath flag.

ktbarrett, 2020-06-28:
I just tested @mtikekar's hypothesis and I'm fairly certain that yes it's how conda has configured their distutils that always adds the link. Trying with my system Python 3.8 install the RPATH is just `$ORIGIN`. This is probably a bug that needs to be fed up to conda.

ktbarrett, 2020-06-28:
From the discourse discussion:

> Paul Moore
> > It sounds to me that Python’s packaging tools simply aren’t the correct solution for you.
>
> > Of course, it’s quite possible that you can get the packaging toolchain to work, but it wouldn’t be a supported configuration.

> Steve Dower
> > Most of the Python ecosystem depends on you using a “normal” distribution. For embedded scenerios, you’re unfortunately on your own.
>
> > In the worst case, you could execute a script with Python to locate its libpython, but you won’t necessarily get the same result if you then load it directly. So I’d really suggest keeping full control over libpython by including it, or have no control by only spawning a process and use inter-process comms.

Some advice, but a resounding "That's weird, you're on your own." And perhaps we shouldn't be compiling the libraries as extensions because they don't fit the definition of an extension.

mtikekar, 2020-06-28:
When you build a conda package, which is as simple as `conda skeleton pypi cocotb; conda build cocotb`, it creates a relocatable package. It does so by replacing the hardcoded rpath with `$ORIGIN/../../../..`. The caveat is that conda packages are always installed in their environment - there is no equivalent of `pip install --editable`. So, the relative path of libpython to `libcocotb.so` is guaranteed to be fixed. 

One option is to emulate conda's rpath logic in cocotb_build_libs.py but turn it off for editable builds.

Also, agree that compiling the libraries as extensions is a little hacky. As I understand, it was done to avoid having to use make/cmake. The alternatives are build_clib and setuptools-dso. There's more discussion on #1387 and #1739

ktbarrett, 2020-07-02:
> It does so by replacing the hardcoded rpath with $ORIGIN/../../../..

That relative path is only guaranteed to work with conda envs. It's probably best for us to just find ways to remove the RPATH to libpython altogether.

> One option is to emulate conda's rpath logic in cocotb_build_libs.py but turn it off for editable builds

Are you saying we could fixup the RPATH on install like conda does? We could accomplish that with setuptools using a [post-install hook](https://stackoverflow.com/questions/20288711/post-install-script-with-python-setuptools) calling patchelf, but 1. that sounds like more work, 2. we now have an *installation* prerequisite that isn't available on most systems.

> As I understand, it was done to avoid having to use make/cmake. The alternatives are build_clib and setuptools-dso.

I'm not quite sure why this is so controversial. CMake is better documented and far more powerful of a build system than setuptools. We can use scikit-build to co-ordinate the two, which also contains functions for easy description of Python extension modules in CMake. As for the issue of additional build system requirements: CMake and Ninja are distributed as package on Pypi. And that of course is only an issue until we start distributing binaries. One additional argument for CMake is that if/when we move the bindings to pybind11, pybind11 provides CMake utilities that are much easier to use than doing the same with setuptools.

eric-wieser, 2020-07-02:
`build_clib` is not a viable option, that only builds static libraries but we need shared libraries.

mtikekar, 2020-07-03:
> Are you saying we could fixup the RPATH on install like conda does? We could accomplish that with setuptools using a post-install hook calling patchelf, but 1. that sounds like more work, 2. we now have an installation prerequisite that isn't available on most systems.

We could modify the rpath in cocotb_build_libs.py itself so that patchelf is not needed. I first thought it would work beyond conda too (except editable installs), but on Ubuntu libpython is under `/usr/lib/x86_64-linux-gnu` while cocotb goes in `/usr/local/lib`, so it basically wouldn't work for any case (`pip install -e`, `pip install --user`, `pip install`).

Setting LD_LIBRARY_PATH at runtime can be problematic too. Xcelium needs LD_LIBRARY_PATH for its own libraries and I can see that several libraries are common between it and my conda environment. If the cocotb Makefile were to modify  LD_LIBRARY_PATH, either Xcelium would load conda libraries or python would load Xcelium libraries. Neither is a good idea.

I also don't see how using CMake would fix this problem.

If it is clear that this problem affects conda only, we should just distribute a conda package for cocotb.

themperek, 2020-07-04:
>If it is clear that this problem affects conda only, we should just distribute a conda package for cocotb.

Can we replace `rpath` with `$ORIGIN/../../../..:$ORIGIN` only in case of conda environment?

PS. I do not see how changing the build system would make a difference here. This will not solve the problem of how everything is linked together as far as I can see?

mtikekar, 2020-07-04:
Conda does not replace the build system; it sits on top of pip/setuptools or whatever we choose (e.g cmake). To make a conda package for cocotb, we would create a meta.yml file with the following info:

1. source path (e.g. https://pypi.io/packages/.../cocotb-1.4.tar.gz)
2. requirements for build, install and test
3. build command (e.g `pip install`)

When you run `conda build`, it does the following things:

1. Create a temporary conda env with the build deps
2. Copy the sources into that env
3. Run the build command
4. Look at what changed in the env from before and after step 3 and copy them to a separate folder
5. For all libraries, replace any absolute rpaths with relative rpaths

As the Python devs pointed out, cocotb is not entirely a python extension. So for distributing binaries, a conda package is really a much better fit than pip wheels. I am biased, but I would recommend making conda the default way to use cocotb on all platforms and not just Windows.

ktbarrett, 2020-07-04:
I only mentioned CMake because it shouldn't be lumped with Make. There are no technical reasons we couldn't use it; while with Make, there is: Windows support. It would be an alternative to setuptools-dso to solve the the issue of our pure C libraries being linked against libpython because we use `distutils.Extension` to build them. Which to choose is a matter of preference, but there already is a setuptools-dso solution, so perhaps this point is moot. CMake might be a better solution in the future, depending on a few things, so we should keep it in mind.

I was hoping with some additional discussion we could figure out a general solution to binary redistribution that would work with our current pip/pypi method of distribution, but I don't think there is one. Source distribution will continue to work fine, but we should figure out some way of preventing wheel re-use, so users don't end up with issues. Assuming they will read the documentation and use `--no-cache-dir` is not a great solution. It would be nice if we could turn off wheel creation in setup.py, so we force a rebuild. That would be the actual solution to the original problem.

@mtikekar is correct, conda binary packages seem to be the only reasonable way to do binary redistribution given our current situation, pip simply doesn't have the feature set.

ktbarrett, 2020-07-04:
I did think of a pretty hacky work around. Instead of using patchelf at post-install, at post-install we write to a file the locations to search for libpython based upon the installed environment (or we can locate the actual path or require the use to supply it somehow). Before the embedded interpreter is set, we read the aforementioned file and temporarily add it to the LD_LIBRARY_PATH before runtime loading libpython (Perhaps there is some better way to add a path to the linker search path at runtime?). This would require us to solve the issue of linking against libpython in the C libraries and gpi_embed.

This should allow us to support binary redistribution with pip/pypi. Please shoot this full of hole, it's terrible, but possibly effective.

ktbarrett, 2020-12-12:
No longer an issue with #2268 in. After it is merged confirm close.

ktbarrett, 2021-01-05:
While this is still an issue when building cocotb in a conda environment, it is no longer an issue in general. The changes in #2268 defeat the effect of RPATH and LD_LIBRARY_PATH when loading libpython.

closed 2021-01-05 by ktbarrett