cocotb/cocotb #1945: PYTHONHOME issue?

juliusbaxter, 2020-06-25:
I had cocotb running in our work environment only last week or so, but something has started to cause the following issue.

I'm on CentOS 7 and the `SIM=xcelium`

The `simple_dff` example appears to die the same way, so here's the output from that:

```
        module worklib.dff:sv
                errors: 0, warnings: 0
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:74   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
Fatal Python error: Py_Initialize: Unable to get the locale encoding
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00007fa44e62ee00 (most recent call first):
ImportError: No module named site
xmelab: *F,INTERR: INTERNAL EXCEPTION
```
Some output from `cocotb-config`:
```
$ cocotb-config --version
1.4.0rc1
$ cocotb-config --python-bin
/usr/bin/python3
$ cocotb-config --prefix
/usr/local/lib64/python3.6/site-packages
```
Running the following (as per `Makefile.inc`'s way of calculating `PYTHONHOME`):
```
$ /usr/bin/python3 -c 'from distutils.sysconfig import get_config_var; print(get_config_var("prefix"))'
/usr
```

An interesting observation (I think anyway?) is if I set `PYTHONHOME` to `/usr/local` under which the `site-packages/cocotb` is really located (as per the `cocotb-config` output), I see the same crash.
```
$ PYTHONHOME=/usr /usr/bin/python3
Python 3.6.8 (default, Apr  2 2020, 13:34:55)
[GCC 4.8.5 20150623 (Red Hat 4.8.5-39)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>>

$ PYTHONHOME=/usr/local /usr/bin/python3
Fatal Python error: Py_Initialize: Unable to get the locale encoding
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00007f4ce0135740 (most recent call first):
Aborted
```

Any ideas? I'm quite certain this worked only a week ago, I just updated to the latest cocotb, but I'm not ruling out something changing on my system, too.

Thanks for any pointers, and great job with cocotb so far all!!

mtikekar, 2020-06-25:
I am on the same system (CentOS 7 and xcelium) and had the same issue show up this week. One difference is that I use conda environments instead of the system python. For me, the problem went away after I set an environment variable named `VIRTUAL_ENV` to the path to my conda environment. I'm not a 100% certain that was what actually fixed it, but you can try setting `VIRTUAL_ENV` to `/usr`.

I am on cocotb 1.4.0rc1.

imphil, 2020-06-25:
Odd. There weren't that many commits in the last week, and on a quick look I can't see anythng that looks super suspicious to break the installation (https://github.com/cocotb/cocotb/commits/master).

Could you try to go back to an older version to get a regression window? (Or even bisect it?) In parallel I'll think a bit more about potential causes ...

juliusbaxter, 2020-06-25:
Thanks for the idea. Unfortunately `VIRTUAL_ENV` did nothing for me here.

@imphil what's the easiest way to go back an earlier version if I did the `pip install https://github.com/cocotb/cocotb/archive/master.zip` method of installing coco?

imphil, 2020-06-25:
The probably easiest option for a bisect is to clone the cocotb repository, and then install with pip from the repository.

```
git clone https://github.com/cocotb/cocotb
cd cocotb
pip install .
```

Then you can either use `git bisect` or just manually go back to a commit a couple weeks back. Re-run `pip install` after you've checked out a different commit.

themperek, 2020-06-25:
One thing to watch out is maybe that Xcelium seems to come with its own Python 3.6.

themperek, 2020-06-25:
Can you check what `/usr/bin/python3 -c 'from distutils.sysconfig import get_config_var; print(get_config_var("exec_prefix"))'` gives?

themperek, 2020-06-25:
It should default to `/usr/local` https://docs.python.org/3/using/cmdline.html?highlight=pythonhome#envvar-PYTHONHOME

themperek, 2020-06-25:
`PYTHONHOME` was necessary on windows on linux it was working without but ...

juliusbaxter, 2020-06-25:
> /usr/bin/python3 -c 'from distutils.sysconfig import get_config_var; print(get_config_var("exec_prefix"))'

It gives me `/usr` as I mentioned in the issue.

If I manually set `PYTHONHOME=/usr/local` then it crashes as reported.

That link you pointed to is interesting, it suggests I can set `PYTHONHOME=prefix:exec_prefix`, however any python invocation then doesn't get very far.

Trying with `PYTHONHOME=/usr/local:/usr`:

```
$ PYTHONHOME=/usr/local:/usr cocotb-config --help
Fatal Python error: Py_Initialize: Unable to get the locale encoding
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00007fd561adc740 (most recent call first):
Aborted
```
and the other way around
```
$ PYTHONHOME=/usr:/usr/local cocotb-config --help
Traceback (most recent call last):
  File "/usr/local/bin/cocotb-config", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python3.6/site-packages/pkg_resources/__init__.py", line 26, in <module>
    import zipfile
  File "/usr/lib64/python3.6/zipfile.py", line 14, in <module>
    import struct
  File "/usr/lib64/python3.6/struct.py", line 13, in <module>
    from _struct import *
ModuleNotFoundError: No module named '_struct'
```

juliusbaxter, 2020-06-25:
Hmm bisecting all the way back to 
```
* 3b47f724  (HEAD) 2020-05-24 (5 weeks ago) Kaleb Barrett Change LOG_CRITICAL to LOG_ERROR in C code called by cocotb
```

Didn't show any working versions.

BTW, to bisect properly I think I had to `pip3 uninstall cocotb` each time before installing again, else I got this message:
```
$ sudo pip3 install .
Processing /projects/digital/work/jules/cocotb
  Requirement already satisfied (use --upgrade to upgrade): cocotb==1.4.0.dev0 from file:///projects/digital/work/jules/cocotb in /usr/local/lib64/python3.6/site-packages
```

I suspect something on my system may have changed. I'll have a bit more a dig on this end.

themperek, 2020-06-26:
I have just check on centos 7 `sudo /usr/bin/pip3 install https://github.com/cocotb/cocotb/archive/master.zip` and it works for me.

themperek, 2020-06-26:
If I do `docker run -it centos:7 bash` and:
```
yum install make gcc gcc-c++ libstdc++-devel python3 python3-devel python3-pip wget unzip
yum install https://download-ib01.fedoraproject.org/pub/epel/7/x86_64/Packages/i/iverilog-10_2-2.el7.x86_64.rpm
/usr/bin/pip3 install https://github.com/cocotb/cocotb/archive/master.zip
unzip master.zip 
cd cocotb-master/examples/simple_dff/
make
```
It also works  for me.

thasti, 2020-07-31:
Is it possible this is a Xcelium version issue? I do see this problem (using CentOS 7) for Xcelium 19.09.012, while with Xcelium 18.09.017 cocotb works as expected.

I'm running in a virtual environment and during startup, cocotb in both cases confirms using the venv-python-interpreter.

```
       -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:97   in set_program_name_in_venv        Using Python virtual environment interpreter at /.../venv/bin/python
Fatal Python error: Py_Initialize: Unable to get the locale encoding
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00007f2bf8d92e00 (most recent call first):
ImportError: No module named site
xmelab: *F,INTERR: INTERNAL EXCEPTION
```

ktbarrett, 2020-08-19:
I agree with @themperek this is likely an issue of Xcelium shipping it's own Python installation and modifying `LD_LIBRARY_PATH` or some other environment variable during startup to point to it. The mixed environment is not happy.

@juliusbaxter Did you ever resolve this issue?

juliusbaxter, 2020-09-16:
Had another bit of a dig at this today. I can confirm what @thasti sees in that the later versions of Xcelium exhibit this crash. I don't have the 18 Xcellium version, but I do have an older version (152) of Incisive which works fine.

I too tried with a virtual environment and saw the same issue. I did notice that message from gpi_embed changing mentioning it noticed I was in a virtual environment.

I have no `LD_LIBRARY_PATH` set.

I have no idea how to debug this further short of reaching out to Cadence to try and understand if `xsim` fiddles with the Python path internally or something.

Perhaps I could get whatever Python is running just before it crashes to print `sys.path`? Would that help understand what it's doing? Any other suggestions to figure out what's going on are appreciated!

ktbarrett, 2020-09-16:
When libpython is loaded, if PYTHONHOME is not set, it will deduce the value of PYTHONHOME based on the location of libpython. So there are two possibilities: the wrong libpython is being loaded, or PYTHONHOME is being set to something unexpected.

To find out exactly what is going on, you need to see which libpython is loaded using `LD_DEBUG=libs make SIM=xsim ...`. Then maybe you can add a log statement to the C++ code to print out the value of PYTHONHOME?

thasti, 2020-09-16:
I would not be surprised if Cadence fiddles with the library paths in different places. I've seen them do some terrible things in the wrapper scripts they distribute with their simulators.  Also be aware they tend to strip/override any LD_PRELOAD/LD_DEBUG variables provided by the user in those wrappers.

I may also be digging into this if lack of simulator access by the devs is what's holding this back, so thanks for the suggestions @ktbarrett.

ktbarrett, 2020-09-16:
>  they tend to strip/override any LD_PRELOAD/LD_DEBUG variables provided by the user

But... why?

A potential solution is that if the simulator ships with a Python installation, it might be a good idea to install cocotb into that installation rather than a system one.

cmarqu, 2020-09-16:
FWIW, I'm not seeing this problem on any combination of Xcelium 19.09.011, 20.03.002 / RHEL6, RHEL7. I'm not using a virtual environment.

ktbarrett, 2020-09-16:
@cmarqu Anaconda could be preventing the problem. Conda's distutils is customized so all `Extension`s have the path to the environment's libpython hardcoded in the RPATH (#1943), which would defeat the issue of picking up the wrong libpython.

juliusbaxter, 2020-10-21:
It indeed did turn out to be an `RPATH` problem in `xrun` and `xmelab`. They bundle a Python3 install with the Xcelium install, the `RPATH` is set to their local `lib64` area, and when running with `LD_DEBUG=libs` it was clear it was always picking up `libpython3.6m.so.1.0` from its local install and not the system one.

I've managed to make cocotb 1.4 work against Xcelium 1909002 by moving those Python 3 libraries out of the Xcelium local lib64 directory. I don't know what the side-effect of this will be on Xcelium, I suspect the system's Python3 install will be adequate for whatever it requires.

themperek, 2020-10-21:
Some kind of "workaround" is to run a different (conda?) python version.

ktbarrett, 2020-10-21:
@juliusbaxter Glad to hear you've tracked that down and have a workaround. I have an idea of how to build a workaround into cocotb, but it will take some time to implement. And of course it only works until the simulators start depending on and loading libpython before we get a say... I stand by my previous suggestion of installing cocotb into the bundled Python installation, if possible. We might need to talk to simulator vendors to nip this bad behavior in the bud.

cmarqu, 2020-11-25:
I just came across the following irun/xrun switch (or rather, remembered it now) which may be worth a try: `-iusldno`.
The documentation is not quite clear whether this has an effect on non-AMS (i.e., "normal") runs, but it is about *not* prefixing `` `xmroot`/tools/lib`` to `LD_LIBRARY_PATH`.

themperek, 2020-12-30:
@juliusbaxter Can you check with the latest master?

juliusbaxter, 2021-03-17:
> @juliusbaxter Can you check with the latest master?

Sure - you mean see if #2268 fixes this? I can do that...

I received a question on how I made this work initially.

Basically I removed the symlink from the Xcelium `tools.lnx86/lib/64bit` for the `libpython*.so.*` files, in my case for my version of Xcelium it was something like
```
cd $XCELIUM_DIR/tools.lnx86/lib/64bit
unlink libpython3.6m.so
unlink libpython3.6m.so.1.0
unlink libpython3.so
```

ktbarrett, 2021-03-17:
I'm guessing it instead picks up the system libpython? I'm guessing in your case they are the same version or Python?

juliusbaxter, 2021-03-21:
> I'm guessing it instead picks up the system libpython? And they are the same version or Python?

Yes, and not sure. Close enough I suppose. Worth checking before doing this, though.

thasti, 2021-04-07:
I got back to testing this issue and can confirm it is still present on recent releases after https://github.com/cocotb/cocotb/pull/2268 was merged.

```
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:110  in set_program_name_in_venv        Using Python virtual environment interpreter at /.../venv/bin/python
Fatal Python error: Py_Initialize: Unable to get the locale encoding
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00007feab3461a80 (most recent call first):
ImportError: No module named site
xmelab: *F,INTERR: INTERNAL EXCEPTION
-----------------------------------------------------------------
```

ktbarrett, 2021-04-07:
@thasti Any user had the issue and was able to work around it #2474. We also had @cmarqu investigate and showed the change does work. I think it remains an environment issue on your end. The user from the other issue was able to resolve his issue by not using virtualenv. I guess it has problems with libpython's location and inferring the base install PYTHONHOME instead of the virtualenv's since virtualenv doesn't handle libpython correctly? That's a wild guess.

Could you run `cocotb-config --libpython`? Is that location reported in the environment where cocotb is installed?

thasti, 2021-04-14:
Thanks for the pointers @ktbarrett. I was able to sort out my environment issues and don't see this problem anymore.

ktbarrett, 2021-04-14:
@thasti What was your solution? Did you just stop using virtualenv?

thasti, 2021-04-14:
No, I'm still using venv (and venv was not actually part of the problem). For unrelated reasons, I had to make libpython3.6m.so available in the environment via LD_LIBRARY_PATH, which worked for the build process, but was not properly picked up by `cocotb-config --libpython` and Xcelium 20 (which probably strips those paths now, and it ended up picking up a different libpython.
I now have a proper system-wide libpython3.6m.so installed in the expected library search paths and cocotb works fine.

ktbarrett, 2021-04-14:
`cocotb-config --libpython` is a wrapper around `find_libpython` which is vendored into the project. Could you open an issue [upstream in `find_libpython`](https://github.com/ktbarrett/find_libpython) with details about your environment so I can determine why it failed?

thasti, 2021-04-14:
Yes, I will attempt to recreate the failure mode. To make it clear: Things started working after _removing_ the hacks that injected libpython via LD_LIBRARY_PATH. These used to work for older versions of Xcelium, but stopped working with Xcelium 20.

Since I was recently able to get the python development packages installed on the systems in question, I do not need to perform these hacks anymore and cocotb works as-is, even in a venv.

closed 2021-05-22 by github-actions[bot]

gretzteam, 2022-01-10:
This started to happen to me when I upgraded my virtual env from python 3.8.6 to 3.9.1. 
Interestingly this only happens on my work environment - on my home machine I have no problem with all the same versions.
Xcelium 20.09.019

xrun starts to compile the verilog, and then gives this error:

```
_in_venv        Using Python virtual environment interpreter at /designs/ip/.direnv/python-3.9.1/bin/python
INFO cocotb: Fatal Python error: init_import_site: Failed to import the site module
INFO cocotb: Python runtime state: initialized
INFO cocotb: Traceback (most recent call last):
INFO cocotb:   File "/usr/local/pkg/python/3.9.1/lib/python3.9/os.py", line 46, in _get_exports_list
INFO cocotb:     return list(module.__all__)
INFO cocotb: AttributeError: module 'posix' has no attribute '__all__'
```

Other potential useful info:
```
cocotb-config --libpython
/usr/local/pkg/Python/3.9.1/lib/libpython3.9.so.1.0
```

I tried adding the /usr/local/pkg/python/3.9.1/lib/ to LD_LIBRARY_PATH, as well as the virtual env lib path, essentially all combinations I could think with no success.
If I simply revert to python 3.8.1, it works fine.

Any idea?

ktbarrett, 2022-01-10:
Well you are loading a Python 3.9 venv and it is pointing to a Python 3.9 stdlib. The only thing I could see wrong is that the interpreter is loading from the venv and the library from the global namespace. Could you try just removing the `export PYTHONHOME = ...` line from the makefiles.

gretzteam, 2022-01-11:
I use cocotb-test but I commented the line setting up PYTHONHOME:
```
self.env["PYTHONHOME"] = get_config_var("prefix")
```
and no change...

xiaoling-yi, 2023-09-12:
I also encounter the same problem. I use conda environment. After I re-install a conda environment and re-install the verilator using the same conda environment as I run the testbench. This issue gets away.

imphil, 2023-09-12:
@xiaoling-yi Please open a new issue with more details about your environment, the issue here has been already closed.

ktbarrett, 2025-05-19:
@0BAB1 This has been fixed on master.

0BAB1, 2025-05-20:
Re-installing cocotb did not work but reinstalling verilator did ! Problem solved. Thanks